dresden-elektronik/deconz-rest-plugin #4249: Aqara Motor B1 does not report anymore

wvuyk, 2021-01-21:
## Describe the bug
Related to #3508. 

When this issue was originally closed all was working well. But something has changed during the last few versions.
If deCONZ is started sending commands do not get a response on the device anymore. When the Analog Output cluster is read once after starting deCONZ REST API commands start to get response from the Aqara motor. Mainly `"lift"=nn` commands are sent.
But the REST API does not return any updates of the commands back:

After sending a `"lift"=0`:

[image]

After sending a  `"lift"=100`:

[image]

Note that only "lastseen" is changed here. So it looks like reporting is not active anymore.

## Steps to reproduce the behavior

## Expected behavior

Reporting back in REST API
## Screenshots

[image]

## Environment

 - Host system: PC 
 - Running method: Windows 
 - Firmware version: 26650700
 - deCONZ version: 2.09.01
 - Device: ConBee II
 - Do you use an USB extension cable:  no -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? none

## deCONZ Logs

## Additional context

ebaauw, 2021-01-21:
Did you try and power cycle the B1?

wvuyk, 2021-01-22:
I think he has done this. The use is holding 5 curtain motors and all are showing the same results. I have forwarded your suggestion to him just now so you can try it still. Will report back when I have an answer

wvuyk, 2021-01-22:
@ebaauw  

No, taking out the battery and reinserting does not change anything, still a first read by hand on Analog output is needed here to have the device accepting commands. Anything we can try to understand what might go wrong?

ebaauw, 2021-01-22:
Is this only after starting deCONZ, or after not interacting with the B1 for a while?  This doesn't make sense to me, as the B1 is reachable, since they can read the _Analog Output_ cluster.

We need to figure out if its the communication between the coordinator and the B1 or the REST API plugin.  After starting deCONZ, can they write the _Present Value_ attribute in the GUI to open/close the curtains, or do they need to read the value first?

What commands are they sending?  Do they have deCONZ logging or Zigbee sniffing to see what Zigbee commands the REST API Plugin makes of these?

wvuyk, 2021-01-22:
I can collect the commands send and webhook returns. He is running windows so tracing as far I know is not going to work out. Would love it if that would someday be enabled as my plugin has many Windows users Same for sniffing I am afraid.

Not executing commands is only after starting deCONZ longer no interaction is not a problem as far we checked. Once communication has been starting by the read, comands are accepted, but no updates are send on the current status as shown in the above JSON samples. Except for `lastseen` nothing is updated.

Commands send by my plugin are normally only `"lift"=nn` commands. I am not using `open` commands here.

wvuyk, 2021-01-23:
@ebaauw 

User has checked on the Present Value:

Stop - start deConz
Write a value to Present Value, the motor responds correct
Control the motor by JowiHue works now also without a read

But no updates on current position on the lift..

ebaauw, 2021-01-23:
So it would seem that the API plugin doesn’t fully restore the resource until it receives a message from the device.  Maybe a side effect of the `/lights` resources now (finally) being available on startup?

As for the updates: does the API update the state when reading the attribute in the GUI?  If so, it’s probably a missing binding and/or attribute report configuration.  If not, there’s something broken.  Does the API update `battery` and `lastupdated` of the ZHABattery resource?

wvuyk, 2021-01-23:
Attributes are changing in the GUI according to the user. Will check on the battery values 

If this would be a side effect, would it not be the same for an IKEA blind? I have one here and it is functioning fine, after a restart of deCONZ commands are executed fine through the plugin.

SwoopX, 2021-01-23:
Seems like #4139 is related to that. Interestingly, after a recent restart, the plug came up like this

[image]

ebaauw, 2021-01-23:
> Attributes are changing in the GUI according to the user.

Then the reporting by the B1 to the coordinator is working.  Definitely a REST API plugin issue.

> note the name

Could mean that `name` wasn't written to the database (could happen if you restart deCONZ too soon after pairing), or that the resource wasn't properly restored.  This seems consistent with having to trigger a message from the device to the gateway, so the API plugin initialises the resource in full.

> would it not be the same for an IKEA blind

Both my mains-powered `lumi.curtain` and IKEA FYRTUR work fine in v2.9.0 and v2.9.1.

The _Analog Output_ and _Analog Input_ clusters use a `float` data type for _Present Value_.  Historically this has been challenging for deCONZ (both core and REST API plugin).  The standard _Window Covering_ cluster uses integer data types.

SwoopX, 2021-01-23:
> Could mean that name wasn't written to the database (could happen if you restart deCONZ too soon after pairing), or that the resource wasn't properly restored. This seems consistent with having to trigger a message from the device to the gateway, so the API plugin initialises the resource in full.

No clue what might have happened here. I just checked the DB entries and it seems as if all resources were in state `deleted` since 5 days. However, I could swear that the device name turned to its NWK just yesterday... Regardless, I'll keep an eye on that (but that's a different topic).

Anyway, the reference made still shows similarities.

wvuyk, 2021-02-24:
The issue is still in need of a solution? what can we do to get this solved?
How can we deliver a trace through using Windows? Why is this not made possible?

Current status: 

When deCONZ is started, a read is needed of Present Value before any command is accepted. From then on commands are accepted, but reporting is not doen, so my plugin does not get the status updates.

The user has a Zigbee sniffer, ordered a China Aqara hub 2 and discovered he can do initial settings with the Xiaomi bridge. For one he found how the curtain can be controlled by hand (in id oxF0000, bit 0x20000. Bit is read only, but the aqara hub seems to have tricks to go around this.... Not sure if this is useful to anyone.

I am sure if sniffing helps for this issue, I can ask him?

wvuyk, 2021-03-19:
*bump* is there anyone who can help here?

wvuyk, 2021-03-29:
@Mimiix  or @manup 
How can I get some help in this? If deCONZ is started the aqara curtain motor B1 does not report through the REST-API. Once a read of any item in basic cluster is done, the motor functions normally and reports as it should.

We are willing to try to get any information to you, but as the user is runnnig on Windows, this is a difficult road. Is there any alternate soltion to get support here?

Mimiix, 2021-03-29:
I have no clue where to start. @SwoopX  or @ebaauw any comments?

Krocko, 2021-04-20:
I have the same problem. The motor don‘t report the current state. Are there any news, or a workaround?

wvuyk, 2021-04-29:
Why is this difficult? Anybody help us on how to get this issue clearer for any devs?

Krocko, 2021-04-30:
@wvuyk is the state report working for you? For me not.

wvuyk, 2021-04-30:
When deCONZ is (re)started, nothing happens, commands are not executed, state reports are not send. After a read of the basic cluster reports start working as well as command. As for the reports the `state\lift `parameter is not updated. `state\lastseen` and `state\open` are updated last time I checked.

Mimiix, 2021-04-30:
Posted it in the devs channel in discord. Hope one can reply :)

manup, 2021-04-30:
Hmm hard to tell without having the device for testing, I put it on the order list.
I can imagine this is similar to the IKEA KADRILJ issue https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4818

wvuyk, 2021-05-01:
The user tested the 2.11.2 beta, but no differences - restart deCONZ does need an extra read on the basic cluster before it accepts commands and starts reporting. I guess the IKEA issue is not related then?

Krocko, 2021-05-01:
For me it is working, if I press the button on the B1. But state report don't work all the time.

Krocko, 2021-05-10:
@manup are there any news?

Krocko, 2021-06-01:
Bump.
@manup are there any news?

wvuyk, 2021-06-01:
Waiting here too, was under the impression there would be a motor in its way in?

Mimiix, 2021-06-01:
Pinged Manuel

manup, 2021-06-01:
Hi everybody, we haven't received the motor yet :/

wvuyk, 2021-06-01:
@manup thanks, hoping this will be resolved.

The stale mark is not removed from this issue? Is the bot being bad with us?

oktay89, 2021-06-18:
Is this issue related to my problem with openhabian 3.1M5? I am not able to control anymore the b1 curtain motor. 
I can see it's online, but when I send command "Down" this log shows and on "Up" I have no error/warning, but it's also not working.
`2021-06-18 15:11:12.477 [WARN ] [ernal.handler.DeconzBaseThingHandler] - Sending command DOWN to channel deconz:windowcovering:e283a68dbd:04cf8cdf3c74b5e901:position failed: 400 - [{"error":{"address":"/lights/4/state","description":"parameter, ontime, not available","type":6}}]`

A few weeks ago it was closing/opening, but could not stop or set percentage to open just a bit - 40%. 
First the curtains stopped closing (Down), now just both directions are not working.

Mimiix, 2021-06-18:
@wvuyk @SwoopX Stale is removed 1 day after it was added

Krocko, 2021-07-14:
Bump. 
@manup have you received the motor?

Krocko, 2021-08-09:
Hi there. Any answer?

Krocko, 2021-09-02:
@manup @Mimiix is this issue ignored?

Mimiix, 2021-09-02:
@Krocko not by me. Can't speak for Manup.

Krocko, 2021-09-24:
Really?

Mimiix, 2021-09-24:
Well, yes, it would auto close in a week.

Krocko, 2021-09-24:
Why? This problem is not solved and no more response from the devs in months.

Mimiix, 2021-09-24:
Because the bot keeps everyone active. You just got the notified of the issue and so am i , so now i can (again) message Manuel about it.

I'm happy to put a label on it that doesn't auto close (and not notify anyone), but than it probably gets forgotten more easily.

I've (once again) messaged Manuel.

mvalsgaard, 2021-10-12:
Any news on this?

Krocko, 2021-11-06:
Really great! 👏🏻

Mimiix, 2021-11-06:
I really can't do more than ping @manup everytime. No need to become sarcastic @Krocko.

Krocko, 2021-11-06:
Months ago he said he is waiting for delivery. Since then no more reaction.

Mimiix, 2021-11-06:
I am not saying I'm not agreeing, because I do. But as I said, I can't force him to look at it.

manup, 2021-11-07:
Hi everyone, sorry I missed to update here. Unfortunately the motor controller shipped was not the Aqara B1 but the "X Gear".
Looking forward, the issue gets added to the v2.14.x milestone. If possible it would be great if somebody could send us the device for testing, or join for live session in Discord to create a DDF, after the v2.13 stable release.

ilya-kozyrev, 2021-11-29:
> Hi everyone, sorry I missed to update here. Unfortunately the motor controller shipped was not the Aqara B1 but the "X Gear".
> Looking forward, the issue gets added to the v2.14.x milestone. If [...]

Hi, big thanks for your contribution folks, I'll happy to help with life session, just ping me in this issue.

Krocko, 2021-12-20:
@manup i am now on the latest stable v2.13.4. If you need someone for a live session, please ping me.

manup, 2021-12-22:
Hi, thanks we can schedule a live session after Xmas, 27.12 or later would work for me

Krocko, 2021-12-22:
No problem.

ebaauw, 2021-12-29:
I don't have the B1, only the original mains-powered motor and now the roller shade driver E1.

Afaik, the B1 should report its current position through the _Present Value_ attribute in the _Analog Output_ cluster.   The mains-powered motor uses the _Windows Covering_ cluster, but also the _Analog Output_.  I hacked the API plugin to ignore the _Windows Covering_ reports, relying only on the _Analog Output_, and that worked. The E1 also only reports current position through _Analog Output_, and that works as well.  Concluding, I think the API plugin logic is sound.  That would mean that the issue is with the (configuration of the) B1.
Could you please check in the GUI whether the _Present Value_ attribute in the _Analog Output_ cluster is updated when moving the curtains?  What about when reading the attribute?

Does the Lumi specific cluster (0xFCC0) carry u8 attribute 0x0009, _Device Operation Mode_?  Could you please check its value (make sure to read it using the _Attribute Editor_ popup window.  If it's any other value than 1, could you write 1 to it, and see if _Present Value_ in the _Analog Output_ starts reporting after that?  That magic seems to be needed for the E1.

Can you please confirm whether the API plugin exposes a ZHABattery for the B1?  Looking at the code, it's supposed to, but I don't think it's working.  Should be fixed in my latest PR.

Can someone please list the Xiaomi special attribute report from the deCONZ log?  See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5330 for what I need (this is from the E1).  I hope to find out how the battery is exposed, and whether it also exposes charging state, like the E1.

ebaauw, 2021-12-30:
I think I might have found the issue.  Symptoms:
- The B1 can no longer be controlled from the API, but can still be controlled from the GUI.  The API actually returns a gateway busy error, but I suppose not all clients check nor report that;
- The B1 has another parent router than the coordinator.

I'll include a fix for this issue in my PR.

The issue is that `writeAttribute()` refuses to write to sleepy devices (_Receiver on When Idle_ is false) unless it's seen a message from the device within the last 3 seconds.  This is under the assumption that the device is asleep, and we dont want to fill the queues with messages that cannot be delivered.  When the device has the coordinator as parent, the poll requests count as incoming messages.  However, the gateway doesn't see the poll requests to another parent router.  Saw a similar issue quite a while back, when trying to read the _Time_ cluster attributes.  Will provide the same workaround: whitelist the _Analog Output_, _Multistate Output_, and _Lumi Specific_ clusters to skip the check.

Mimiix, 2021-12-30:
@ebaauw could this also apply for the fyrtur issues people see when the fyrtus has another device than the coordinator as a parent?

ebaauw, 2021-12-30:
No, the FYRTUR and KADRILJ use _Window Covering_ cluster commands, rather than writing attributes in the _Analog Output_, _Multistate Output_ and _Lumi Specific_ clusters.  Are they even seeing the same symptoms: device still contollable by the GUI, but not by the API (unless you read it first from the GUI)?

Mimiix, 2021-12-30:
I'm not sure on that. The fyrtur seemed to have issues with controlling them over the api when they were connected with a router instead of the coordinator. I was never able to reproduce this.

ebaauw, 2021-12-31:
> Will provide the same workaround: whitelist the Analog Output, Multistate Output, and Lumi Specific clusters to skip the check.

Not too happy about that; the _Output_ clusters are probably only used by Xiaomi window covering devices, but the _Lumi Specific_ cluster is more common.  Changed the workaround: now bluntly calling `rx()` before calling `writeAttributes()`.

closed 2022-01-30 by github-actions[bot]