dresden-elektronik/deconz-rest-plugin #4251: New firmware for Elko Super TR making it not available in REST API and Home Assistant

henrikrox, 2021-01-21:
## Device

- Product name: Elko Super TR
- Manufacturer: Elko
- Model identifier: Super TR
- Device type : 
  - Thermostat

## Screenshots

[image]
[image]

This should have already been added trough
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2088

However it looks like a i have a new firmware, since there are groups/scenes etc, also in thermostat i have a lot more attributes compared to the original issue #2088

Find the thermostat in deconz, but rest api returns nothing under sensors.

henrikrox, 2021-01-21:
I also notice a lot of 

`21:53:54:690 No thermostat sensor found for 0x680AE2FFFE6963BF, endpoint: 0x00000001
21:53:54:690 ZCL attribute report 0x680AE2FFFE6963BF for cluster: 0x0201, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:54:15:303 Current channel 11
21:54:15:311 Device TTL 1134 s flags: 0x7`

in deconz logs. 

However looking from thermostat attributes, it at least sees the correct temperature (air in this case)
[image]

SwoopX, 2021-01-23:
Hm, apparently, they changed the RF-Chip. We now see a different MAC address as well as manufacturer code, resulting in a required check to fail. Should be just one line to add.

closed 2021-02-15 by manup

oywino, 2021-03-09:
@henrikrox did you update the firmware in your "old" thermostat, or is this a the brand new version from ELKO that they advertise as ZigBee 3.0 compatible?

andersop91, 2021-03-27:
@oywino seems like the new hardware. I've bought one of this (https://elkosmart.elko.no/produkter/smarttermostat-super), ZigBee 3.0 compatible. It looks just the same in deCONZ at least.

jornholm, 2021-06-22:
@andersop91 did you get your device to work? I have 2 of the same version and they are not showing up in the api.

andersop91, 2021-06-29:
@jornholm Yes, it showed up, but weren't seamless. I changed to ZHA instead, where also the PIR was supported.