dresden-elektronik/deconz-rest-plugin #4255: Device name in Deconz GUI changes after restart

easybeat, 2021-01-22:
## Describe the bug
I have 3 devices which if I name them in Phoscon App to the correct name, they change also in Deconz GUI the name. But after I restart Deconz GUI, they change the name back in Deconz GUI but in Phoscon App they still show with the correct name.

## Steps to reproduce the behavior
For only 3 device on my system this happens:
1. Rename the device in Phoscon APP
2. Check name in Deconz GUI (Result: Name has changed)
3. Stop Deconz GUI
4. Start Deconz GUI
5. Check name in Phoscon (Result: Name is still as changed)
6. Check name in Deconz GUI (Result: Name has changed back)

## Expected behavior
Names of devices in Phoscon App and Deconz GUI should always be the same

## Screenshots

[image]
[image]

After restart of deconz GUI:
[image]

## Environment

 - Host system: NUC with Proxmox
 - Running method: Ubuntu  Virtual Machine in Proxmox
 - Firmware version: 26680700
 - deCONZ version: 2.08.00 / 8.12.2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? yes If so: Aeotec Z-Wave Stick

## deCONZ Logs
can't log as it happens at startup

## Additional context

Mimiix, 2021-01-23:
After the restart, it seems it's not connected (grayed out).

easybeat, 2021-01-26:
> After the restart, it seems it's not connected (grayed out).

But it is connected. It is a light and works without any problems.

[image]

Smanar, 2021-01-26:
Lol, ha yes, and there is something called "Schlafzimmer OF" somewhere ? Deconz can't imagine this name.

It s the older name ?

easybeat, 2021-01-26:
> Lol, ha yes, and there is something called "Schlafzimmer OG" somewhere ? Deconz can't imagine this name.
> 
> It s the older name ?

Sorry, don't get you mean...anyway the name in Phoscon is still "Esszimmer Decke". Everything else is wrong.

Smanar, 2021-01-26:
But at restart, the device is called "Radiator 01 Schlazfimmer OG 001" in phoscon, it can invent this name, where it s from ?

easybeat, 2021-01-26:
Ah now I understand. Yes this device exists and is like the name says a radiator TRV:

[image]

But it has a completly different MAC-Adress/ID

And as you can see the device is still named in Phoscon with the correct name:
[image]

Smanar, 2021-01-26:
Lol, so to resume, after a restart you have a device that take the name from an other device in deconz, but all names are fine in phoscon ?

Funny ^^, but as the name are good in phoscon, I think it s more a problem in deconz core, no ?

easybeat, 2021-01-27:
Yes I think it is a deconz problem...

easybeat, 2021-01-27:
@Smanar are you going to change this issue to BUG?

Mimiix, 2021-01-27:
@easybeat Smanar can't do that. I'll ask Manuel to check this issue first, as i cant replicate it.

easybeat, 2021-01-27:
> @easybeat Smanar can't do that. I'll ask Manuel to check this issue first, as i cant replicate it.

Ok...and I opened this as a bug report....

kyletpugh, 2021-01-27:
I have this problem, I ended up closing the GUI and stopping the service. I then edited zll.db with SQLite and the names stayed. I also had duplicates of every button. I removed all the duplicates and renamed everything properly, when I restarted everything came back up as it was supposed to be named. So it appears that something on GUI or service shutdown is writing out wrong values as when I changed zll.db while running it was changed back after restart. I am unsure if these duplicate buttons (none appeared duplicated in GUI or phoscon) were causing issues.

Smanar, 2021-01-27:
Yep, don't worry, we can change the label to "bug" instead of "user question", but asking to deconz dev first.

manup, 2021-01-29:
There are some issues with names for API resources and inconsistencies when storing them in the database.
Also when joining a device which is already paired, existing names can be modified and revert to the default name.

This is under investigation and should be fixed in the upcoming releases, but likely after the next stable since. A few places need to be reviewed and refactored.

easybeat, 2021-02-24:
Any news on this bug?

Mimiix, 2021-02-24:
Made a internal ticket.

kyletpugh, 2021-02-24:
It appear to be fixed, although I'm not sure. I've removed one of the devices that was never properly named, restarted, re-added the light, renamed the light, restarted again, the light had the same name. So that seems fixed. However, I could not remove the device from Phoscon, I had to use the GUI to right click the node and delete. Otherwise, the device just stopped working but was not grayed in Phoscon and all the neighbor links in the GUI turned red. Took several hours for the neighbor links to turn green again after the successful removal and replacement.

easybeat, 2021-02-24:
> It appear to be fixed, although I'm not sure. I've removed one of the devices that was never properly named, restarted, re-added the light, renamed the light, restarted again, the light had the same [...]

OK. Which Firmware are you running?

kyletpugh, 2021-02-24:
> OK. Which Firmware are you running?

Oh, I wish I looked. I just upgraded after I wrote that. I am using the beta channel packages for Ubuntu 20.04. Version is now 2.10.01 / 12/23/2020 / 26660700. It looks fixed in this version as I just restarted and everything is green to start. One of my lights has a terrible connection now though, which makes no sense. It is 68/0 even though it's about eight feet from the Conbee II and it's being routed through another device that is eight get in the other direction.

easybeat, 2021-02-26:
OK, I did the update to 2.10.1 and the problem still exists. 

@kyletpugh I think the different is that you did delete the device. I don't want to delete and re-pair it.

kyletpugh, 2021-02-26:
@easybeat I actually edited the database, that's the only way I could get it to work because it appears the problem is the device us added multiple times. It didn't work with removing the device for me either, I had to edit the database and remove the duplicates, then rename the remaining ones correctly. Then I have working names again, I removed and readded a few devices on 2.10.01 and it seems to work correctly now.

kyletpugh, 2021-03-20:
Pretty sure this is caused by unescaped strings being used to build SQL queries. I started from scratch and made sure no nodes, devices, or entities had any non alphabet characters. Don't seem to be having any of the issues I was having, I do not think that the queries are succeeding if they are not strictly alphanumeric.

darrylo, 2021-04-04:
I think I've only seen this with names with an apostrophe (forward single-quote character).  Names like "John's Lamp" seem to not work.  So, unescaped SQL commands is a good possibility.

easybeat, 2021-08-24:
Hi

I just checked on latest stable 2.12.6 and the problem still exists.

1. Change the name in Phoscon APP
2. Name changes in Deconz GUI
3. restart deconz GUI
4. Name is wrong again in deconz GUI but OK in Phoscon App

Thanks

Beat

Smanar, 2021-08-24:
>Name changes in Deconz GUI

You can't change the name is the GUI. you need to use the API, or use third app like Phoscon.

manup, 2022-10-07:
Problem was due the ordering of the sub resources, this was fixed with DDF capabilities in [v2.13.1-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.13.1-beta).

closed 2022-10-07 by manup