dresden-elektronik/deconz-rest-plugin #4256: Namron 4 in 1 Zigbee Rotary Switch

corvy, 2021-01-22:
## Device

- Product name: Namron 4 in 1 Zigbee Rotary Switch
- Manufacturer: NAMRON AS
- Model identifier: 4512726
- Device type : Please remove all unrelated device types. 
  - Switch

## Screenshots

[image]

[image]

  If available add screenshots of other clusters.

Power config: 
[image]

I have successfully got this rotary dimmer into the Web API page and added lights to a group. I can successfully dim up or down, and then suddenly the group gets deleted and goes back to zero. See this sequence of screenshots.

First, no lights in group. 
[image]

Add these lights:
[image]

Confirming lights in group:
[image]

After dimming up or down, no lights in group anymore:
[image]

Should we capture some logs to understand what is going on here?

Note that this dimmer has 4 modes. Monochrome dimming, CCT, RGB and RGB+CCT. Push on/off and turn to dim. Doubleclick to change colors. I have tried to play with chaning the mode but it does not seem to work in any config. 

This is the manual for the device:
https://www.elektroimportoren.no/docs/lib/4512726-Brukerveiledning-5.pdf

Image of the device: 
[image]

corvy, 2021-01-22:
I did some log captures: 

> 13:41:13:051 group 25345 member 30
13:41:13:051 group 25345 member 31
13:41:13:052 group 25345 member 32
13:41:13:052 group 25345 member 33
13:41:17:935 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 054
13:41:17:937 reanimate group 44033 for sensor 4512726 26
13:41:17:938 delete old group 25345 of sensor 4512726 26
13:41:17:938 attach group 44033 to sensor 26
13:41:17:946 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 054
13:41:17:949 delete old group 44033 of sensor 4512726 26
13:41:17:949 reanimate group 25345 for sensor 4512726 26
13:41:17:949 attach group 25345 to sensor 26
13:41:18:239 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 05D
13:41:18:241 reanimate group 44033 for sensor 4512726 26
13:41:18:241 delete old group 25345 of sensor 4512726 26
13:41:18:242 attach group 44033 to sensor 26
13:41:18:247 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 05D
13:41:18:249 delete old group 44033 of sensor 4512726 26
13:41:18:250 reanimate group 25345 for sensor 4512726 26
13:41:18:250 attach group 25345 to sensor 26
13:41:18:578 Remove from group response for light 30. Status: 0x8B, capacity: 255
13:41:18:822 Remove from group response for light 31. Status: 0x8B, capacity: 255
13:41:19:052 Remove from group response for light 32. Status: 0x8B, capacity: 5
13:41:19:324 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 051
13:41:19:326 reanimate group 44033 for sensor 4512726 26
13:41:19:326 delete old group 25345 of sensor 4512726 26
13:41:19:327 attach group 44033 to sensor 26
13:41:19:333 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 051
13:41:19:335 delete old group 44033 of sensor 4512726 26
13:41:19:336 reanimate group 25345 for sensor 4512726 26
13:41:19:336 attach group 25345 to sensor 26
13:41:19:407 Remove from group response for light 33. Status: 0x8B, capacity: 1
13:41:19:567 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 048
13:41:19:570 reanimate group 44033 for sensor 4512726 26
13:41:19:570 delete old group 25345 of sensor 4512726 26
13:41:19:571 attach group 44033 to sensor 26
13:41:19:572 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 048
13:41:19:574 delete old group 44033 of sensor 4512726 26
13:41:19:575 reanimate group 25345 for sensor 4512726 26
13:41:19:576 attach group 25345 to sensor 26
13:41:19:725 Current channel 25
13:41:19:734 Device TTL 3512 s flags: 0x7
13:41:20:649 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 05A
13:41:20:731 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 05A
13:41:20:733 reanimate group 44033 for sensor 4512726 26
13:41:20:734 delete old group 25345 of sensor 4512726 26
13:41:20:734 attach group 44033 to sensor 26
13:41:20:962 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 063
13:41:20:975 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 051
13:41:20:977 delete old group 44033 of sensor 4512726 26
13:41:20:977 reanimate group 25345 for sensor 4512726 26
13:41:20:978 attach group 25345 to sensor 26

During this log I basically added the lights to the group with the remote, then tried to dim up/down. And the remote then lost the lights in the group.

SwoopX, 2021-01-23:
You're lucky here as I make the code generic enough also to cover your device ;)

Regardless, we see 2 open topics here:
1. Your device seems to require a different button map than all the other devices. That is comparably easy to create and most likely doesn't require any code amendments
2. What's a bit odd is that the device seems to be bound to 2 groups and each button press is broadcasted to those. That's probably the reason why the groups jump as seen in the logs
3. Just checked the manual, and the device does support color control cluster, that requires some code changes. Meanwhile, we can manually make sure this is considered

corvy, 2021-01-24:
Thanks @SwoopX, after alot of trial and error I think I managed to get it to work. Now it is stable on group 25345, hex 0x6301. I manually bound the colorcontrol and now I can change the colors as-well. Very cool :) 

So far it seems like it is a bit hit and miss with the colorcode setting, and the CC to WW setting. When I just use it for dimming the light brightness it is a lot more stable. I will use the dimmer and capture logs to see if I can get more control over how it works. 

One thing, could it be that the color cluster was not bound because the dimmer was not put in the RGB+CCT mode before pairing it with the hub?

Also, if there is something special you would need me to capture please tell me. I run deconz in the home assistant docker container.

corvy, 2021-01-26:
I have tried the switch a bit, and apart from this middinh bing for colorcontrol I cannot see any issues. Here is a log where I dim up and down, change to color control, change the colors around and then dim agin. 

 
```
14:54:20:947 0x00178801040D757B error APSDE-DATA.confirm: 0xE9 on task
14:54:21:059 ZCL attribute report 0xF0D1B8000012A05E for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:54:23:219 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0AE
14:54:23:362 ZCL attribute report 0xF0D1B80000128955 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:54:23:366 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
14:54:23:370 ZCL attribute report 0xF0D1B8000012A05E for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:54:24:175 ZCL attribute report 0x7CB03EAA00AC5125 for cluster: 0x0006, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
14:54:24:193 ZCL attribute report 0x7CB03EAA00AC5125 for cluster: 0x0008, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
14:54:24:216 ZCL attribute report 0x7CB03EAA00AC5125 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
14:54:24:268 0x0017880108E91A07 error APSDE-DATA.confirm: 0xE9 on task
14:54:24:273 ZCL attribute report 0x7CB03EAA00AC5125 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
14:54:24:666 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0C2
14:54:24:756 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
14:54:24:890 ZCL attribute report 0xF0D1B80000128D90 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:54:24:895 ZCL attribute report 0xF0D1B8000012A05E for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:54:25:004 ZCL attribute report 0xF0D1B80000128D90 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:54:25:022 ZCL attribute report 0xF0D1B80000128955 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:54:25:735 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0D6
14:54:25:809 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
14:54:26:548 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0FD
14:54:26:673 ZCL attribute report 0xF0D1B8000012A05E for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:54:26:690 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
14:54:26:954 ZCL attribute report 0xF0D1B80000128D90 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:54:27:325 ZCL attribute report 0xF0D1B80000128D90 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:54:27:338 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 011
14:54:27:482 ZCL attribute report 0xF0D1B80000128955 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:54:27:486 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
14:54:27:883 0x001788010352A706 error APSDE-DATA.confirm: 0xE9 on task
14:54:28:504 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0FD
14:54:28:596 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
14:54:28:793 ZCL attribute report 0xF0D1B8000012A05E for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:54:29:471 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0D6
14:54:29:562 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
14:54:30:560 ZCL attribute report 0xF0D1B80000128D90 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:54:30:576 Current channel 25
14:54:30:586 Device TTL 3413 s flags: 0x7
14:54:30:598 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0E9
14:54:30:664 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
14:54:32:389 ZCL attribute report 0xF0D1B80000128955 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:54:32:405 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0FD
14:54:32:508 0x001788010669C00F error APSDE-DATA.confirm: 0xE9 on task
14:54:32:925 ZCL attribute report 0xF0D1B8000012A05E for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:54:32:931 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
[... cut ...]
14:54:57:122 ZCL attribute report 0x0017880104B4B9DB for cluster: 0x0400, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
14:54:58:217 ZCL attribute report 0xF0D1B8000012A05E for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:55:02:053 ZCL attribute report 0x0017880104B547D2 for cluster: 0x0406, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
14:55:04:405 verified group capacity: 21 and group count: 3 of LightNode 0x0017880106696d02
14:55:04:405 0x0017880106696d02 found group 0x0007
14:55:04:406 0x0017880106696d02 found group 0xFFF0
14:55:04:406 0x0017880106696d02 found group 0x000E
```

SwoopX, 2021-01-26:
Thanks, but I also need to know what is what. I can't know which messages were generated upon which action ;)

What about on/off (single/double press, hold)?

corvy, 2021-01-26:
Ok, i Will create a map of actions Vs functions.

corvy, 2021-02-01:
Hello @SwoopX, sorry for the delay! Here is my breakdown. 

At start the dimmer is in dimming function. So I start here.

Push on:
`Button 1002 On 4512726`

Push off:
`Button 2002 Off 4512726``

Turn right:
` No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 09A`

Turn left:
` No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 077`

Double click (to change to color control: No log event. 
After this I turn the wheel again. 

Turn right: 
`No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: 0x00 payload[0]: 080`

Turn left:
`No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: 0x00 payload[0]: 070`

It seems to me that the color control RGB+WW/CW control is about 32 steps. Not absolutely sure. 

Here is a log after turning to the right, cycling through the colors. 
```
12:36:45:262 ZCL attribute report 0xF0D1B80000128D90 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:36:45:278 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
12:36:45:282 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: 0x00 payload[0]: 000
12:36:45:353 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
12:36:45:371 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
12:36:46:744 ZCL attribute report 0xF0D1B80000128955 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:36:46:764 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: 0x00 payload[0]: 010
12:36:46:778 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
12:36:46:860 ZCL attribute report 0xF0D1B8000012A05E for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:36:46:877 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
12:36:48:282 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: 0x00 payload[0]: 020
12:36:48:304 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
12:36:48:354 ZCL attribute report 0xF0D1B8000012A05E for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:36:48:361 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
12:36:49:301 ZCL attribute report 0xF0D1B80000128D90 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:36:49:322 ZCL attribute report 0xF0D1B80000128955 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:36:50:080 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: 0x00 payload[0]: 030
12:36:50:122 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
12:36:50:164 ZCL attribute report 0xF0D1B8000012A05E for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:36:50:182 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
12:36:51:483 ZCL attribute report 0xF0D1B80000128D90 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:36:54:942 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: 0x00 payload[0]: 040
12:36:54:981 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
12:36:55:008 ZCL attribute report 0xF0D1B8000012A05E for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:36:55:032 ZCL attribute report 0xF0D1B80000128D90 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:36:55:038 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
12:36:55:111 ZCL attribute report 0x0017880104B4B9DB for cluster: 0x0406, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
12:36:55:116 ZCL attribute report 0xF0D1B80000128955 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:36:58:003 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: 0x00 payload[0]: 050
12:36:58:120 ZCL attribute report 0xF0D1B80000128D90 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:36:58:141 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
12:36:59:562 ZCL attribute report 0xF0D1B80000128955 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:36:59:576 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: 0x00 payload[0]: 060
12:36:59:672 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
12:36:59:745 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
12:37:00:116 ZCL attribute report 0xF0D1B8000012A05E for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
[... cut ...]
12:37:21:253 ZCL attribute report 0xF0D1B8000012A05E for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
```

Also it seems that the "ZCL attribute report 0xF0D1B80000128D90 for cluster: 0x0300"  and similar comes right after each command form the switch/dimmer. Is this related somehow to the dimmer itself? 

From my  experience I find that the dimmer works very well and quite stable. Especially on the lightlevel dimming. 

What more do you need? Shall I conduct more tests and log captures?

SwoopX, 2021-02-01:
So that would be the button map then, based on your input. Please note the last entry, as the command is send, but not "documented".

```
"4512726Map": {
            "vendor": "Namron",
            "doc": "Namron 4 in 1 Zigbee Rotary Switch",
            "modelids": ["4512726"],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_TO_LEVEL_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Dim (with on/off)"],
                [1, "0x01", "COLOR_CONTROL", "MOVE_TO_HUE", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Dycle colors"],
                [1, "0x01", "COLOR_CONTROL", "MOVE_TO_COLOR_TEMPERATURE", "0", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "?????"]
            ]
        },
```

corvy, 2021-02-05:
Thanks for this. Bet because I am not used to how to add this, should I include it in my deconz container? Or would the keymap be part of a new release.  Also, lastly, would the last entry become a problem, or am I supposed to do something about it? 

Sorry for only posting stupid questions :D :D

SwoopX, 2021-02-07:
Those are no stupid questions ;)

Well, you cann add that map yourself, if you can find the file `button_maps.json` in your container, but this is something I cannot help you with.
When you add it, I'd recommend to make a backup of the original file just in case anything goes wrong and you can restore it or give it another go. You need to be careful not to break the structure here.

I don't think the last entry would be any problem at all. However, I'd perfer to have it tested and complete before including it officially and that's where some testing comes into play.

corvy, 2021-02-09:
Thanks 👍 Happy you took the time again. 

I added it simply this by installing nano (apt update && apt install nano) and then editing the file like this (appending to the end):

[image]

I started deconz again by running ./init and it actually worked. 

This is the log:

```
13:18:32:092 [INFO] - Found file containing button maps. Parsing data...
13:18:32:093 [WARNING] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
13:18:32:094 [INFO] - Button map item #3, field #4 for '4512726Map' was not found in object 'commands' for cluster 'COLOR_CONTROL'.
[13:18:32] INFO: Starting Nginx...
13:18:32:097 [INFO] - Button maps loaded.
```

Tried to capture some logs from operating the switch and something has changed: 
```
13:20:58:017 verified group capacity: 21 and group count: 3 of LightNode 0x001788010669c00f
13:20:58:017 0x001788010669c00f found group 0x0007
13:20:58:017 0x001788010669c00f found group 0xFFF0
13:20:58:017 0x001788010669c00f found group 0x000E
13:20:58:083 ZCL attribute report 0xF0D1B80000128D90 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:20:58:743 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0008, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
13:20:58:837 verified group capacity: 19 and group count: 5 of LightNode 0x001788010352a6ab
13:20:58:838 0x001788010352a6ab found group 0x000A
13:20:58:838 0x001788010352a6ab found group 0xFFF0
13:20:58:838 0x001788010352a6ab found group 0x62CC
13:20:58:838 0x001788010352a6ab found group 0x0401
13:20:58:838 0x001788010352a6ab found group 0x0402
13:20:59:409 ZCL attribute report 0xF0D1B80000128955 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:20:59:558 verified group capacity: 21 and group count: 3 of LightNode 0x001788010669b829
13:20:59:558 0x001788010669b829 found group 0x0007
13:20:59:559 0x001788010669b829 found group 0xFFF0
13:20:59:559 0x001788010669b829 found group 0x000E
13:21:00:268 verified group capacity: 255 and group count: 4 of LightNode 0x000b57fffe369428
13:21:00:268 0x000b57fffe369428 found group 0x62CC
13:21:00:269 0x000b57fffe369428 found group 0x0401
13:21:00:269 0x000b57fffe369428 found group 0xFFF0
13:21:00:269 0x000b57fffe369428 found group 0x0404
13:21:00:785 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0x000B57FFFE369428 (wait reading or unsupported)
13:21:00:880 ZCL configure reporting rsp seq: 75 0x000B57FFFE369428 for ep: 0x01 cluster: 0x0006 attr: 0x0000 status: 0x00
13:21:01:092 verified group capacity: 21 and group count: 3 of LightNode 0x0017880106687e7f
13:21:01:092 0x0017880106687e7f found group 0x0007
13:21:01:092 0x0017880106687e7f found group 0xFFF0
13:21:01:092 0x0017880106687e7f found group 0x000E
13:21:01:616 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 059
13:21:01:709 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0008, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
13:21:01:791 verified group capacity: 19 and group count: 5 of LightNode 0x001788010352ec08
13:21:01:792 0x001788010352ec08 found group 0x000A
13:21:01:792 0x001788010352ec08 found group 0xFFF0
13:21:01:792 0x001788010352ec08 found group 0x62CC
13:21:01:792 0x001788010352ec08 found group 0x0401
13:21:01:792 0x001788010352ec08 found group 0x0402
13:21:01:862 ZCL configure reporting rsp seq: 78 0x000B57FFFE369428 for ep: 0x01 cluster: 0x0000 attr: 0x4000 status: 0x00
13:21:01:959 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 069
13:21:02:012 ZCL attribute report 0xF0D1B80000128D90 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:21:02:017 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0008, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
13:21:02:021 ZCL attribute report 0xF0D1B8000012A05E for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:21:02:243 [INFO] - No button handler for: 4512726 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 082
13:21:02:281 ZCL attribute report 0xF0D1B80000128955 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:21:02:435 ZCL attribute report 0x0017880104B547D2 for cluster: 0x0406, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
13:21:02:528 verified group capacity: 19 and group count: 5 of LightNode 0x001788010352e2e0
13:21:02:528 0x001788010352e2e0 found group 0x000A
13:21:02:529 0x001788010352e2e0 found group 0xFFF0
13:21:02:529 0x001788010352e2e0 found group 0x62CC
13:21:02:529 0x001788010352e2e0 found group 0x0401
13:21:02:529 0x001788010352e2e0 found group 0x0402
[... 124 lines cut ...]
13:21:16:916 Unbind response failed with status 0x88 for 0x086bd7fffe5ce34d ep: 0x01 cluster: 0x2000
13:21:17:267 [INFO] - Button 5002 ????? 4512726
13:21:17:360 ZCL attribute report 0x7CB03EAA00AEFDF1 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
13:21:17:472 verified group capacity: 20 and group count: 5 of LightNode 0x00178801040d7867
13:21:17:472 0x00178801040d7867 found group 0xFFF0
13:21:17:473 0x00178801040d7867 found group 0x000A
13:21:17:473 0x00178801040d7867 found group 0x62CC
13:21:17:473 0x00178801040d7867 found group 0x0401
13:21:17:473 0x00178801040d7867 found group 0x0402
13:21:17:575 [INFO] - Button 5002 ????? 4512726
[... cut ...]
13:21:18:232 0x00178801090583f7 found group 0xFFF0
13:21:18:232 0x00178801090583f7 found group 0x62CC
13:21:18:994 verified group capacity: 23 and group count: 2 of LightNode 0x0017880108e91a07
13:21:18:994 0x0017880108e91a07 found group 0xFFF0
13:21:18:994 0x0017880108e91a07 found group 0x0004
```

During this quick test I dimmed up, then down, then double-clicked and then changed the colors a buit. Last I reverted to dimming a little before I captured, what I believe is aprox the logs at the time of "action".

SwoopX, 2021-02-14:
Hm, I gotta check back with the other devs on the MOVE_TO_LEVEL_WITH_ON_OFF command/behavior. Presumably, this should/could be mapped to a HOLD action, not sure if that would have any relevance though

I also do not fully understand what's going on with the color control. However, lights and such is not really my turf.

Presumably, the button map isn't that important anyway here.

closed 2021-03-15 by github-actions[bot]