dresden-elektronik/deconz-rest-plugin #4257: Device name

barlock, 2021-01-22:
## Device

- Product name: Yoolax Blinds
- Manufacturer: Yookee
- Model identifier: D10110
- Device type : Please remove all unrelated device types. 
  - Light
  - other: Cover

## Screenshots

[image]

barlock, 2021-01-22:
Added some more screenshots. This is a battery powered one. 

I don't know how to compile and test. I was looking for some kind of contributing docs but couldn't find any. Have a link? 

I do have a full linux machine, I was planning on doing it in Docker, but if I can pull out a linux box if I need to.

Any good examples for adding support for something like this? From glancing at other PRs it looks like I need to touch `bindings.cpp`?

barlock, 2021-02-24:
For anyone else who finds this, as a node dev, I had an easier time migrating to zigbee2mqtt and adding support there. Closing this as I don't need it anymore.

closed 2021-02-24 by barlock

Schaussi, 2021-03-23:
Hello,
are you planning support for Yoolax blinds in the near future?
As by myself I think I cannot help here much (was not able to get manufacturer or model ID within deConz GUI).
But maybe some other users have found out more, e.g. how the pairing process exactly works.
I use Yoolax Zebra Duo blinds, for the moment I will use a Wemos D1 Mini with a relay card and ESPHome to simulate keypresses on the remote.
Best regards

Mimiix, 2021-03-23:
We need screenshots for this to work :)

Schaussi, 2021-03-23:
I upgraded to the latest deConz software and firmware and finally got a bit further.

The pairing process:
- Blinds must be powered on (default; otherwise by pressing motor button for 1s)
- Pairing mode: 4x press motor button (you hear 4 beeps in sum), then press 1x for 3s (motor will jog)
- Phoscon App (web): Trying to search for switches, then for lights (no success here, but device shows up in deConz GUI)

Screenshots:
[image]
[image]

Please tell me if you need more data.
Thank you

Schaussi, 2021-04-06:
Would you mind re-opening this issue?
I have a spare ConBee II and could test and compile C code, if you could give me some hints which files or data needs to be modified.
Thank you!

Smanar, 2021-04-06:
Hello, have you a linux machine, with real OS ?
If yes

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch yoolax https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry,it can freeze during compilation, my pi 3 do, so remove "-j2"

I haven't tested the code on my side, so you can have compilation error.

Schaussi, 2021-04-07:
Thank you, this worked so far.
I started "adding a light" in the web app and the yoolax blind appeared there as a light.
(A brightness bar appears in addition in the test group, which has no effect yet)

New screenshot:
[image]

(Tab "cluster info" has the same content like before)

Please tell me if you maybe need other data.

Smanar, 2021-04-07:
The integration is perhaps fine.
But you can't test it using phoscon, it don't support yet covering.

You don't have third app ? Else I can explain how to use the API ?

Schaussi, 2021-04-07:
Great! I did not expect it would be already working within Home Assistant, thank you.

(For other testing users reading this: If using 2 Conbee II with 1 HA instance, you have to activate "authenticate app" _on the zigbee host you usually use_, not on the one created for testing. Otherwise you get an HA error "could not find API key". Phoscon web app on your usual host should find both Conbee sticks, even if one is remote)

So basically, I can move up and down the blinds with HA, I am happy with it.

If you would like to go into further detail, what I experienced with the first HA tests within the web interface:
- Position to 100 -> moves _up_
- Down button -> moves down (OK)
- Up button -> Always disabled
- Position from 100 to 97 -> moves up to 0 (does not stop)
- Position from 97 to 100 -> moves _up_ (I guess it does not get a current position value, not knowing if this is even possible)

It would be great if you could consider this in a future version.
Thank you!

Smanar, 2021-04-07:
The correct json need to be

```
shutter closed
on= true
bri = 255
open = false
lift = 100%
```

And no, it work for other device, so the problem is probably from my code.
I probably need to reverse lift/bri. Can you show how are your values ?

Schaussi, 2021-04-07:
These are the values I get from HA:
[image]
(The current blind (hardware) position seems to have no effect on this, the "current position" value is always 100, and state is always "closed").
In case you need other values, please give me information where to get it from.

Smanar, 2021-04-07:
If you take a look a look in logs I think you can find the part I m missing.
I realy need to compare on/open with bri/lift for a full closed or a full open, in the same request.

Schaussi, 2021-04-07:
I tried to get some information from the deCONZ logs:

Setting position to 5% in HA (which will move the closed blind up by 5%):

> 22:25:27:625 HTTP API PUT /api/F4CF203485/lights/2/state - 10.1.0.4
> 22:25:27:627 Text Data: 	{"lift": 95}
> 22:25:27:628 ApiMode: 0
> 22:25:27:630 [{"success":{"/lights/2/state/lift":95}}]
> 22:25:28:760 Current channel 25
> 22:25:28:799 CTRL got nwk update id 0
> 22:25:28:836 Device TTL 4019 s flags: 0x7
> 

Pressing Down button in HA:

> 22:25:35:245 [{"success":{"/lights/2/state/open":false}}]

Setting position to 100% in HA (which will move the closed blind fully up):

> 22:34:27:334 HTTP API PUT /api/F4CF203485/lights/2/state - 10.1.0.4
> 22:34:27:340 Text Data: 	{"lift": 0}
> 22:34:27:341 ApiMode: 0
> 22:34:27:342 [{"success":{"/lights/2/state/lift":0}}]

Smanar, 2021-04-08:
If you can't access to HA deconz log, there is 2 other way
- Use a websocket client (I m using "Browser websocket client' on firefox and connect it to deconz websocket `ws://192.168.1.1:8088` for me
- -Take a look direclty in the api `http://IP:PORT/api/KEY/lights` IP and PORt are the same than for phoscon, and KEY is a deconz api key.

Schaussi, 2021-04-08:
Results I got via second way:

Blind is on bottom (closed):

> {"1":{"etag":"2ed6e2cffb641f76c70bcff9e72be53e","hascolor":false,"lastannounced":null,"lastseen":"2021-04-08T17:04Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:05:41:72-01"},"2":{"etag":"cce80700e551e6c9b31fd3e113507a8c","hascolor":false,"lastannounced":"2021-04-07T22:59:29Z","lastseen":"2021-04-08T17:03Z","manufacturername":"yooksmart","modelid":"D10110","name":"Window covering device 2","state":{"bri":10,"lift":0,"on":true,"open":false,"reachable":true},"swversion":"20201120","type":"Window covering device","uniqueid":"84:fd:27:ff:fe:6a:63:1d-01"}}

Blind is going upwards:

> {"1":{"etag":"dd9ae87df63ad8583fc94425e330190e","hascolor":false,"lastannounced":null,"lastseen":"2021-04-08T17:07Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:05:41:72-01"},"2":{"etag":"eaf44da9684f0200cab74cee8b7e26df","hascolor":false,"lastannounced":"2021-04-07T22:59:29Z","lastseen":"2021-04-08T17:07Z","manufacturername":"yooksmart","modelid":"D10110","name":"Window covering device 2","state":{"bri":10,"lift":0,"on":true,"open":false,"reachable":true},"swversion":"20201120","type":"Window covering device","uniqueid":"84:fd:27:ff:fe:6a:63:1d-01"}}

Blind is on top (opened):

> {"1":{"etag":"7633212cecfddc6072415e96094f9996","hascolor":false,"lastannounced":null,"lastseen":"2021-04-08T17:08Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:05:41:72-01"},"2":{"etag":"4355993c39209337b76cb4acec7de673","hascolor":false,"lastannounced":"2021-04-07T22:59:29Z","lastseen":"2021-04-08T17:08Z","manufacturername":"yooksmart","modelid":"D10110","name":"Window covering device 2","state":{"bri":10,"lift":0,"on":true,"open":false,"reachable":true},"swversion":"20201120","type":"Window covering device","uniqueid":"84:fd:27:ff:fe:6a:63:1d-01"}}

Blind on position 93:

> {"1":{"etag":"b337268f37ca8f38f33eadb35f2799ae","hascolor":false,"lastannounced":null,"lastseen":"2021-04-08T17:10Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:05:41:72-01"},"2":{"etag":"300126a7ba7563e2ee6bdec911eea604","hascolor":false,"lastannounced":"2021-04-07T22:59:29Z","lastseen":"2021-04-08T17:10Z","manufacturername":"yooksmart","modelid":"D10110","name":"Window covering device 2","state":{"bri":10,"lift":0,"on":true,"open":false,"reachable":true},"swversion":"20201120","type":"Window covering device","uniqueid":"84:fd:27:ff:fe:6a:63:1d-01"}}

Hope it is useful for you, they seem to be quite the same.

Smanar, 2021-04-09:
>they seem to be quite the same.

Exactly ^^
The bind/report is not working.

Have updated the code if you can try it ?
```

cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Schaussi, 2021-04-09:
Thank you, it looks better now:

Blind is on bottom (closed):

> {"1":{"etag":"3d612f3ae7c948f7025dfcaba9dfcea4","hascolor":false,"lastannounced":null,"lastseen":"2021-04-09T17:38Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:05:41:72-01"},"2":{"etag":"6efefcf942b2c1b10d306915bebff9c8","hascolor":false,"lastannounced":"2021-04-09T17:32:43Z","lastseen":"2021-04-09T17:37Z","manufacturername":"yooksmart","modelid":"D10110","name":"Window covering device 2","state":{"bri":0,"lift":0,"on":false,"open":true,"reachable":true},"swversion":"20201120","type":"Window covering device","uniqueid":"84:fd:27:ff:fe:6a:63:1d-01"}}

Blind is going upwards:

> {"1":{"etag":"bef2cd9a101079acc4a97ae3820ced9f","hascolor":false,"lastannounced":null,"lastseen":"2021-04-09T17:40Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:05:41:72-01"},"2":{"etag":"b71448be5a8b4867674c2c6a7327b04e","hascolor":false,"lastannounced":"2021-04-09T17:32:43Z","lastseen":"2021-04-09T17:39Z","manufacturername":"yooksmart","modelid":"D10110","name":"Window covering device 2","state":{"bri":76,"lift":30,"on":true,"open":true,"reachable":true},"swversion":"20201120","type":"Window covering device","uniqueid":"84:fd:27:ff:fe:6a:63:1d-01"}}

Blind is on top (open):

> {"1":{"etag":"bef2cd9a101079acc4a97ae3820ced9f","hascolor":false,"lastannounced":null,"lastseen":"2021-04-09T17:40Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:05:41:72-01"},"2":{"etag":"20741beefcf14a604375dc9e56584cd1","hascolor":false,"lastannounced":"2021-04-09T17:32:43Z","lastseen":"2021-04-09T17:40Z","manufacturername":"yooksmart","modelid":"D10110","name":"Window covering device 2","state":{"bri":254,"lift":100,"on":true,"open":false,"reachable":true},"swversion":"20201120","type":"Window covering device","uniqueid":"84:fd:27:ff:fe:6a:63:1d-01"}}

Blind on position 93:

> {"1":{"etag":"6f17e71913f480e985f72c762f420439","hascolor":false,"lastannounced":null,"lastseen":"2021-04-09T17:42Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:05:41:72-01"},"2":{"etag":"04732df5a226e964ea30e1745ba8dd59","hascolor":false,"lastannounced":"2021-04-09T17:32:43Z","lastseen":"2021-04-09T17:42Z","manufacturername":"yooksmart","modelid":"D10110","name":"Window covering device 2","state":{"bri":236,"lift":93,"on":true,"open":true,"reachable":true},"swversion":"20201120","type":"Window covering device","uniqueid":"84:fd:27:ff:fe:6a:63:1d-01"}}

From what I experienced, I guess the "lift" value might be reversed.
- Up button works
- Down button works
- Stop button works
- Position bar imho shows the correct value (when going down, the position value increases (goes to the right))
- - When setting the bar to e.g. 93, it goes down only 7 %, then shows a json value of 93

Some screenshots:

[image]
(Blind is closed, but Up button disabled)

[image]
(Blind is open, but Down button disabled)

[image]
(Setting position to 93 -> blind moves down by 7 %, then position bar also jumps to value 7)

Smanar, 2021-04-09:
We realy need a setting in the api to reverse covering ...
Can you try the new code again ? Don't need to re-include the device.

Schaussi, 2021-04-10:
Thank you for the code.
The buttons work fine (up/down/stop), the positioning bar is reversed now (100 = top/open within HA).
Though, when setting the value via the position bar (e.g. from 100 to 93), the blind moves down very far (I guess to HA position 7). When I set e.g. a value 25 at the position bar, it moves down to HA value 75).
Sorry if it's confusing :)

This is the current json data when it is on top/open:

> {"1":{"etag":"10641ba1b60221f892b856bcf4af7832","hascolor":false,"lastannounced":null,"lastseen":"2021-04-10T06:28Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:05:41:72-01"},"2":{"etag":"ba1f73a7b1367b954b47298ebaba2a23","hascolor":false,"lastannounced":"2021-04-10T06:22:46Z","lastseen":"2021-04-10T06:29Z","manufacturername":"yooksmart","modelid":"D10110","name":"Window covering device 2","state":{"bri":0,"lift":0,"on":false,"open":true,"reachable":true},"swversion":"20201120","type":"Window covering device","uniqueid":"84:fd:27:ff:fe:6a:63:1d-01"}}

HA position bar value 75:

> {"1":{"etag":"c23e99622795c08be9adb79a23e9ea9b","hascolor":false,"lastannounced":null,"lastseen":"2021-04-10T06:35Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:05:41:72-01"},"2":{"etag":"5755f7a84fdddff43c527d75666030e6","hascolor":false,"lastannounced":"2021-04-10T06:22:46Z","lastseen":"2021-04-10T06:36Z","manufacturername":"yooksmart","modelid":"D10110","name":"Window covering device 2","state":{"bri":63,"lift":25,"on":true,"open":true,"reachable":true},"swversion":"20201120","type":"Window covering device","uniqueid":"84:fd:27:ff:fe:6a:63:1d-01"}}

Smanar, 2021-04-10:
>This is the current json data when it is on top/open
>"bri":0,"lift":0,"on":false,"open":true

Fine for it, it s percent closure.

For the second one IDK if it s HA that have modified the value or the code. I say that because

> Setting position to 5% in HA (which will move the closed blind up by 5%):
> 
>     22:25:27:625 HTTP API PUT /api/F4CF203485/lights/2/state - 10.1.0.4
>     22:25:27:627 Text Data: {"lift": 95}

You are setting 5% in HA but sending 95 to the device, so it s normal the device don't go at same position.
Are you sure not haven't a reverse setting in HA ?

Schaussi, 2021-04-10:
I ran some further tests, I think you are right.
Everything works, but HA's "current position" value is inverted for unknown reason.
For my purposes it does not matter, or if needed I could use a custom template.
However it's a pity there is no simple invert option in HA's cover.
Thank you for your help, is there a chance your code will be included in an official release soon?

Smanar, 2021-04-10:
>However it's a pity there is no simple invert option in HA's cover.

For me it s strange, generaly it s the reverse problem ^^, we need a reverse and HA don't so it, strange it reverse it naturaly.

I will make the PR now, This code is light, I think it will be in next stable, but can't promise.

jcarusone1650, 2021-06-04:
Looks like in Homebridge/Hue the position is reversed (ie 5% Open is actually 95% Open)

Smanar, 2021-06-05:
Can be from the mounting mode ?
Can you show the json for a complete position ?
Have you tried the command to reverse the device ?

jcarusone1650, 2021-06-05:
This is the current json data when it is on top/open:

`18:21:34:460 {"1":{"etag":"17d3b8b9478997a83a486ee7fe9a9290","hascolor":false,"lastannounced":null,"lastseen":"2021-06-05T18:21Z","manufacturername":"dresden elektronik","modelid":"RaspBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26690700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:06:4e:4a-01"},"2":{"etag":"10c6f5ba1a1a5db1c4352cd60fcecca1","hascolor":false,"lastannounced":"2021-06-04T20:29:03Z","lastseen":"2021-06-05T18:21Z","manufacturername":"yooksmart","modelid":"D10110","name":"Window covering device 2","state":{"bri":0,"lift":0,"on":false,"open":true,"reachable":true},"swversion":"20201120","type":"Window covering device","uniqueid":"60:a4:23:ff:fe:04:04:58-01"}}`

This is the current json data when it is 90% Open (Almost Full Open):

`18:21:49:462 {"1":{"etag":"17d3b8b9478997a83a486ee7fe9a9290","hascolor":false,"lastannounced":null,"lastseen":"2021-06-05T18:21Z","manufacturername":"dresden elektronik","modelid":"RaspBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26690700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:06:4e:4a-01"},"2":{"etag":"4990747cba746316521682e2f3ea6005","hascolor":false,"lastannounced":"2021-06-04T20:29:03Z","lastseen":"2021-06-05T18:21Z","manufacturername":"yooksmart","modelid":"D10110","name":"Window covering device 2","state":{"bri":27,"lift":11,"on":true,"open":true,"reachable":true},"swversion":"20201120","type":"Window covering device","uniqueid":"60:a4:23:ff:fe:04:04:58-01"}}`

This is the current json data when it is 10% Open (Almost Full Closed):

`18:51:04:460 {"1":{"etag":"28229d30b036a87804eb22b71102f641","hascolor":false,"lastannounced":null,"lastseen":"2021-06-05T18:50Z","manufacturername":"dresden elektronik","modelid":"RaspBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26690700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:06:4e:4a-01"},"2":{"etag":"6ae273da34d3c41ea421a8b357eafaa2","hascolor":false,"lastannounced":"2021-06-04T20:29:03Z","lastseen":"2021-06-05T18:50Z","manufacturername":"yooksmart","modelid":"D10110","name":"Window covering device 2","state":{"bri":231,"lift":91,"on":true,"open":true,"reachable":true},"swversion":"20201120","type":"Window covering device","uniqueid":"60:a4:23:ff:fe:04:04:58-01"}}`

This is the current json data when it is Fully Closed:

`18:51:09:459 {"1":{"etag":"28229d30b036a87804eb22b71102f641","hascolor":false,"lastannounced":null,"lastseen":"2021-06-05T18:50Z","manufacturername":"dresden elektronik","modelid":"RaspBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26690700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:06:4e:4a-01"},"2":{"etag":"daa4a1a0fa56a5bcefef0e0a1a115903","hascolor":false,"lastannounced":"2021-06-04T20:29:03Z","lastseen":"2021-06-05T18:50Z","manufacturername":"yooksmart","modelid":"D10110","name":"Window covering device 2","state":{"bri":248,"lift":98,"on":true,"open":true,"reachable":true},"swversion":"20201120","type":"Window covering device","uniqueid":"60:a4:23:ff:fe:04:04:58-01"}}`

Smanar, 2021-06-06:
If I take the last logs (you have take logs during move ?), it seem fine, you have for exemple "open":false  for fully closed

The standards is

> 
> **shutter closed**
> on= true
> bri = 255
> open = false
> lift = 100%

But If I remember, I have already see this issue with alexa, for exemple https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4071#issuecomment-762172074

There is a command to reverse the side, on the GUI, on the covering cluster, but for you all seem fine.

jcarusone1650, 2021-06-08:
The part I am struggling with is if you look at the command being sent in the JSON above, when I set "lift=0", the blind is fully closing and reports lift=100, BRI=254.  Alternatively, if I send a "lift=100" the blind fully opens and returns a "lift=0,BRI=0".

I have tried reversing the motor and while the LIFT/BRI are then correct, sending open = TRUE  closes the blind, and open = FALSE opens the blind. There is a mismatch between the LIFT/BRI values and the open=TRUE/FALSE operation.

Smanar, 2021-06-08:
Ha, I haven't see that.
You are right ...
Do you have a linux machine to test code modification ?

@barlock and @Schaussi you don't the issue on your side ?

jcarusone1650, 2021-06-08:
Yes, I do have a Linux box and happy to test.

Smanar, 2021-06-08:
So wiating for more return you can try that, you have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch yoolax_reverse https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

I have keep the code intact, I just reverse the value you are using in the request, and only for lift/bri.

Remember the standard for test

> **shutter closed**
> on= true
> bri = 255
> open = false
> lift = 100%

jcarusone1650, 2021-06-09:
When running make I get the following error, any suggestions?

`de_web_plugin.cpp: In member function ‘void DeRestPluginPrivate::checkSensorButtonEvent(Sensor*, const deCONZ::ApsDataIndication&, const deCONZ::ZclFrame&)’:
de_web_plugin.cpp:4452:34: error: use of ‘buttonMap’ before deduction of ‘auto’
     for (const auto &buttonMap : buttonMap->buttons)
                                  ^
Makefile.Release:761: recipe for target 'release/de_web_plugin.o' failed
make[1]: *** [release/de_web_plugin.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/root/deconz-rest-plugin'
Makefile:38: recipe for target 'release' failed
make: *** [release] Error 2`

Smanar, 2021-06-09:
Yep, the dev package was updated some days ago, so try to update all and use the command (in the same folder)

`make clean`

To clean tempory files

jcarusone1650, 2021-06-09:
New Linux box so pulled the repo for the first time on it yesterday. Also running a clean does not seem to get past that error

Smanar, 2021-06-09:
Strange, You have updated the OS too ?

jcarusone1650, 2021-06-09:
Yep fully updated

Smanar, 2021-06-09:
And your dev package version ?

`deconz-dev is already the newest version (2.12.00-raspbian-stretch-beta).`

jcarusone1650, 2021-06-09:
deconz-dev is already the newest version (2.12.00-ubuntu-xenial-beta).

Smanar, 2021-06-09:
Ok, .... , no idea, will ask to other devs.

jcarusone1650, 2021-06-12:
I was able to get this to compile on the previous version of Debian. It does correct the reversing issues

Smanar, 2021-06-12:
No one have idea for your procedent issue ...

Are you sure all is fine ? Check with  

> **shutter closed**
> on= true
> bri = 255
> open = false
> lift = 100%

And pls test with open and lift.

As I don't have return from others user I will make the PR if all is fine for you.

jcarusone1650, 2021-06-14:
Confirmed and tested two additional shades to confirm it was not specific to the first unit I installed. The patch does fix the reversal in all states.

Smanar, 2021-06-14:
Thx ^^ , PR done.

oripaamoni, 2022-05-18:
I am having the same issue described above using a Pi4 and conbee 2 using the zigbee integration that was autodetected in the UI. Do I need to load this patch separately? Or is there a config I just need to change somewhere. I am pretty new to HA, Switching over from homebrige and trying to ditch homekit all together.

Smanar, 2022-05-19:
This patch is included in deconz since more than 1 year.

If you have only the return reversed, can be solved with using a DDF
If request are reversed too, there is a code WIP, but I don't found users to test it.

oripaamoni, 2022-05-19:
> This patch is included in deconz since more than 1 year.
> 
> If you have only the return reversed, can be solved with using a DDF If request are reversed too, there is a code WIP, but I don't found users to test it.

Thank you for the reply but I was able to solve my problem by disabling quirks. It seems like the latest yooksmark motor firmware corrected the issue and doesn’t require the use of quirks anymore. Is there a way to only disable quirks for one device rather than all of them?

Smanar, 2022-05-19:
Hu ?
"quirks" ? you are using ZHA ? not deconz ?

oripaamoni, 2022-05-19:
> 

Correct

Smanar, 2022-05-19:
Ok, so you are not in the good place ^^