dresden-elektronik/deconz-rest-plugin #4260: [DEVICE REQUEST] SilverCrest doorbell

Fredo70, 2021-01-23:
## Device

- Product name: Smart Home SilverCrest Kabellose Türklingel (model N° on the back of the device HG06668)
- Manufacturer: _TZ1800_ladpngdx
- Model identifier: TS0211
- Device type : Please remove all unrelated device types. 
  - Other: doorbell

## Screenshots

[image]
[image]

ebaauw, 2021-01-23:
Please list the _IAS Zone_ cluster.  Make sure to read the attributes, pressing the device (or possibly its reset button) to wake it.

Fredo70, 2021-01-23:
[image]

Fredo70, 2021-01-23:
[image]

Mario-75, 2021-01-26:
I have the same doorbell but it does not show up at all on my ConBee 2.
I have Rpi4 with Openhabian ("Raspbian GNU/Linux 10 (buster)"). I installed the latest firmware (26660700) and used beta driver (2.09.01). The gateway seems work since it finds my new switches (SilverCrest Zwischenstecker) and I can control them with openHAB.
I did another installation on my Win10 computer with GUI, moved the ConBee2 stick to the machine, but same situation. Doorbell does not show up when pressing the connect button (3x within 2s).
I tried the loging (/usr/bin/deCONZ --dbg-info=2 > dbg-info2.txt) but I don’t really know what to look for.

Mimiix, 2021-01-26:
@Mario-75 That's why support is requested: You can't use it yet. 

You need to wait for a dev to include it :)

commit 6f5ce728 (2021-02-15): Add support for Lidl/Silvercrest door bell (TS0211)

Fixes #4260
closed 2021-02-15 by manup