dresden-elektronik/deconz-rest-plugin #4263: Mi Magic Cube MFKZQ01LM firing only rotation

OshriDan, 2021-01-23:
## Describe the bug

  Mi Magic Cube MFKZQ01LM only firing rotation cw and ccw (logged via homeassistant) the rest of the events are not firing.

## Steps to reproduce the behavior

1. shake cube
2. check logs
3. no shake event was fired

## Expected behavior

Cube shake event fires event with gesture 1

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26390500
 - deCONZ version: 2.09.01
 - Device: RaspBee I

## deCONZ Logs
none

## Additional context
I cannot attach this switch to the light group as well, I could before but now it doesn't she it in the selection, the switch is available and reporting angle correctly.

Smanar, 2021-01-23:
Hello, this device need to have 2 sensor created, one ending by 000C and one other by 0012.
If you miss one of them, pls try to re include it (without deleting old entry)

turbouk2312, 2021-01-25:
I had to try a few times before both sensor types were added. Try adding it again, but shake the cube during the searching process

OshriDan, 2021-01-26:
Ok, so after re adding it and shaking now the cube fires all events, thank you all very much for the help :)

closed 2021-01-26 by OshriDan