dresden-elektronik/deconz-rest-plugin #4269: Envilar ZGR904-S remote (same as iCasa remote)

spiff42, 2021-01-24:
## Device

- Product name: ZGR904-S Zigbee Fjernbetjening (Touchlink)
- Manufacturer: ENVILAR
- Model identifier: ZGR904-S
- Device type : Remote

The remote is a clone of iCasa remote ICZB-RM11S (see  https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2648#issuecomment-760462453). I actually bought it hoping it would work because it looked exactly like the iCasa remote, which was listed on the supported devices page.

It was bought here:
https://www.plusled.dk/zgr904-s-zigbee-fjernbetjening-touchlink.html

The PDF manual (in Danish and English):
https://www.plusled.dk/amfilerating/file/download/file_id/79/

## Screenshots

Just like the iCasa remote, the device has 4 identical endpoints:
[image]

The basic cluster of the first endpoint (all look to be the same):
[image]

[image]

I can provide screenshots of other clusters if needed, but since this remote appears to be another clone of the iCasa remote (like the Sunrichter Remote #3125), I am assuming this is sufficient.

Mimiix, 2021-01-24:
@spiff42 please add all clusters. We can't assume anything unfortunately.

SwoopX, 2021-01-24:
The screenshots given suffice.

I'll check if we can just treat if like the other remotes referred to.

spiff42, 2021-03-01:
Thanks @SwoopX for picking this up again. Please let me know if I can be of any assistance.

closed 2021-03-03 by manup