dresden-elektronik/deconz-rest-plugin #4273: Danfoss icon master controller with zigbee modul

Nicolaj1710, 2021-01-24:
## Device

- Product name: Danfoss icon master controller with danfoss zigbee modul
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: Danfoss
- Device type : Please remove all unrelated device types. 
  - Thermostat
   

[image]

donnib, 2021-01-27:
@Nicolaj1710 interesting, i was just looking to see if there was support for this. Do you have this connected to your water based floor heating?

Nicolaj1710, 2021-01-27:
Yes I have, and can control everything in deconz, but missing in HA

donnib, 2021-01-27:
Can you control the telestats as well ?

Nicolaj1710, 2021-01-27:
Yes I can, and with schedule and home and away

mihaicph, 2021-01-27:
@Nicolaj1710 so in the essence you could get a way without the gateway or do you need it for anything ?

Nicolaj1710, 2021-01-27:
Yes you don't need it, but I can't get it in HA before deconz rest API
support it

Nicolaj1710, 2021-01-27:
[Zigbee API Danfoss Icon Radio Module - Developer - 20200902.xlsx](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5883742/Zigbee.API.Danfoss.Icon.Radio.Module.-.Developer.-.20200902.xlsx)

hanskroner, 2021-02-08:
Unfortunately, my Danfoss Icon Master Controller is "too old" to work with the Zigbee module and I'll need to source a replacement. I haven't been able to test the below, but it looks like adding basic support for the telestats in the REST API should be as simple as this:

```diff
diff --git a/bindings.cpp b/bindings.cpp
index 939307b..a933141 100644
--- a/bindings.cpp
+++ b/bindings.cpp
@@ -1364,6 +1364,42 @@ bool DeRestPluginPrivate::sendConfigureReportingRequest(BindingTask &bt)
             return sendConfigureReportingRequest(bt, {rq, rq2, rq3}) || // Use OR because of manuf. specific attributes
                    sendConfigureReportingRequest(bt, {rq4, rq5});
         }
+        else if ( (sensor && sensor->modelId() == QLatin1String("0x8020")) || // Danfoss RT24V Display thermostat
+                  (sensor && sensor->modelId() == QLatin1String("0x8021")) || // Danfoss RT24V Display thermostat with floor sensor
+                  (sensor && sensor->modelId() == QLatin1String("0x8030")) || // Danfoss RTbattery Display thermostat
+                  (sensor && sensor->modelId() == QLatin1String("0x8031")) || // Danfoss RTbattery Display thermostat with infrared
+                  (sensor && sensor->modelId() == QLatin1String("0x8034")) || // Danfoss RTbattery Dial thermostat
+                  (sensor && sensor->modelId() == QLatin1String("0x8035")) )  // Danfoss RTbattery Dial thermostat with infrared
+        {
+            rq.dataType = deCONZ::Zcl16BitInt;
+            rq.attributeId = 0x0000;       // local temperature
+            rq.minInterval = 60;
+            rq.maxInterval = 3600;
+            rq.reportableChange16bit = 50;
+
+            ConfigureReportingRequest rq2;
+            rq2.dataType = deCONZ::Zcl8BitBitMap;
+            rq2.attributeId = 0x0002;        // Occupancy
+            rq2.minInterval = 60;
+            rq2.maxInterval = 43200;
+            rq2.reportableChange8bit = 1;
+
+            ConfigureReportingRequest rq3;
+            rq3.dataType = deCONZ::Zcl16BitInt;
+            rq3.attributeId = 0x0012;        // Occupied heating setpoint
+            rq3.minInterval = 1;
+            rq3.maxInterval = 43200;
+            rq3.reportableChange16bit = 1;
+
+            ConfigureReportingRequest rq4;
+            rq4.dataType = deCONZ::Zcl16BitInt;
+            rq4.attributeId = 0x0014;        // Unoccupied heating setpoint
+            rq4.minInterval = 1;
+            rq4.maxInterval = 43200;
+            rq4.reportableChange16bit = 1;
+
+            return sendConfigureReportingRequest(bt, {rq, rq2, rq3, rq4});
+        }
         else if (sensor && sensor->modelId() == QLatin1String("902010/32")) // Bitron thermostat
         {
             rq.dataType = deCONZ::Zcl16BitInt;
@@ -1595,7 +1631,13 @@ bool DeRestPluginPrivate::sendConfigureReportingRequest(BindingTask &bt)
             rq.reportableChange8bit = 4; // value used by Hue bridge
         }
         else if ( (sensor && sensor->modelId() == QLatin1String("eTRV0100")) || // Danfoss Ally
-                  (sensor && sensor->modelId() == QLatin1String("TRV001") ))
+                  (sensor && sensor->modelId() == QLatin1String("TRV001")) ||   // Hive TRV
+                  (sensor && sensor->modelId() == QLatin1String("0x8020")) ||   // Danfoss RT24V Display thermostat
+                  (sensor && sensor->modelId() == QLatin1String("0x8021")) ||   // Danfoss RT24V Display thermostat with floor sensor
+                  (sensor && sensor->modelId() == QLatin1String("0x8030")) ||   // Danfoss RTbattery Display thermostat
+                  (sensor && sensor->modelId() == QLatin1String("0x8031")) ||   // Danfoss RTbattery Display thermostat with infrared
+                  (sensor && sensor->modelId() == QLatin1String("0x8034")) ||   // Danfoss RTbattery Dial thermostat
+                  (sensor && sensor->modelId() == QLatin1String("0x8035")) )    // Danfoss RTbattery Dial thermostat with infrared
         {
             rq.minInterval = 3600;         // Vendor defaults
             rq.maxInterval = 43200;        // Vendor defaults
@@ -2797,6 +2839,13 @@ bool DeRestPluginPrivate::checkSensorBindingsForAttributeReporting(Sensor *senso
         sensor->modelId() == QLatin1String("DS01") ||
         // Danfoss
         sensor->modelId() == QLatin1String("eTRV0100") ||
+        sensor->modelId() == QLatin1String("TRV001") ||
[... 13 lines cut ...]
@@ -404,7 +404,7 @@ static const SupportedDevice supportedDevices[] = {
     { VENDOR_ALERTME, "SLR2", computimeMacPrefix }, // Hive   Heating Receiver 2 channel
     { VENDOR_ALERTME, "SLR2b", computimeMacPrefix }, // Hive   Heating Receiver 2 channel second version
     { VENDOR_ALERTME, "SLT2", computimeMacPrefix }, // Hive thermostat
-    { VENDOR_DANFOSS, "TRV001", silabs2MacPrefix }, // Hive thermostat (From Danfoos)
+    { VENDOR_DANFOSS, "TRV001", silabs2MacPrefix }, // Hive thermostat (From Danfoss)
     { VENDOR_SUNRICHER, "45127", silabs2MacPrefix }, // Namron 1/2/4-ch remote controller
     { VENDOR_SENGLED_OPTOELEC, "E13-", zhejiangMacPrefix }, // Sengled PAR38 Bulbs
     { VENDOR_SENGLED_OPTOELEC, "E1D-", zhejiangMacPrefix }, // Sengled contact sensor
@@ -429,6 +429,12 @@ static const SupportedDevice supportedDevices[] = {
     { VENDOR_NONE, "TH01", tiMacPrefix }, // Sonoff SNZB-02
     { VENDOR_NONE, "DS01", tiMacPrefix }, // Sonoff SNZB-04
     { VENDOR_DANFOSS, "eTRV0100", silabs2MacPrefix }, // Danfoss Ally thermostat
[... cut ...]
+    { VENDOR_DANFOSS, "0x8034", silabs6MacPrefix }, // Danfoss RTbattery Dial thermostat
+    { VENDOR_DANFOSS, "0x8035", silabs6MacPrefix }, // Danfoss RTbattery Dial thermostat with infrared
     { VENDOR_LDS, "ZBT-CCTSwitch-D0001", silabs2MacPrefix }, // Leedarson remote control
     { VENDOR_KWIKSET, "SMARTCODE_CONVERT_GEN1", zenMacPrefix }, // Kwikset 914 ZigBee smart lock
     { VENDOR_EMBER, "TS1001", silabs5MacPrefix }, // LIDL Livarno Lux Remote Control HG06323
```

The telestats support additional functionality, but hopefully this is a starting point.

hanskroner, 2021-02-15:
I opened a Pull Request that implements basic support for the Zigbee module for Danfoss Icon masters. Like I mention in the PR, it's a little tricky to join the device, but can be done with a bit of patience. As I also mention in the PR, the created Sensors don't show up in Phoscon, but work just fine over the REST API.

The deCONZ Home Assistant integration doesn't yet handle the separate battery levels for each in-room thermostat, so I created an [issue](https://github.com/Kane610/deconz/issues/95) in that project to hopefully get that implemented. Everything else seems to work fine, but I'd appreciate if any of you can build and test my attempt.

closed 2021-02-22 by manup

WhistleMaster, 2021-02-23:
Well, that's great news ! So basically, you have to join the Zigbee module and all the thermostat in Deconz ? If all the devices are already configured but not yet in Deconz, how do you start the join process ? Thanks !

hanskroner, 2021-02-24:
The in-room thermostats are not Zigbee. They communicate to the Icon Master via what Danfoss calls the "Radio Module." The "Zigbee Module" acts as a bridge, exposing each of the in-room thermostats to a Zigbee network. I'm coming up on a two weeks of working with mine, and the only annoyance I've found is that changing a thermostat's setpoint via Zigbee doesn't update the setpoint in the actual thermostat - the Icon Master registers the change and acts on it, but checking the setpoint physically at the in-room thermostat shows the "old" value. Hopefully this is just an oversight Danfoss can fix later via a firmware update (the Icon Master's firmware can be updated via the Zigbee module).

I joined my device by starting the process in Phoscon. You only join the Zigbee module, not the individual thermostats. Check the notes I made in the pull request linked above and let me know if you succeed in joining your device.

jameswoudenberg, 2021-03-29:
> The in-room thermostats are not Zigbee. They communicate to the Icon Master via what Danfoss calls the "Radio Module" - which is a separate piece of hardware from the "Zigbee Module." The "Zigbee Module" acts as a bridge, exposing all of the in-room thermostats to a Zigbee network. I'm coming up on two weeks of working with mine, and the only annoyance I've found is that changing a thermostat's setpoint via Zigbee doesn't update the setpoint in the actual thermostat - the Icon Master registers the change and acts on it, but checking the setpoint physically at the in-room thermostat shows the "old" value. Hopefully this is just an oversight Danfoss can fix later via a firmware update (the Icon Master's firmware can be updated via the Zigbee module).
> 
> I joined my device by starting the process in Phoscon. You only join the Zigbee module, not the individual thermostats. Check the notes I made in the pull request linked above and let me know if you succeed in joining your device.

 Could you maybe colaborate how to add the Radio Module in Phoscon? I'm new to the whole deCONZ set-up and installed this specially for my Danfoss icon installation. I have a App mdule and the Radio module to manage the thermostats. 

I tried to add devices -> Sensor -> Vendor Other.  Only i can't really notice that the seeming round button on the radio module is really a button. Documentation on the danfoss website is very poor on this. Can you help?

hanskroner, 2021-03-30:
You're either using the wrong terminology, or you have the wrong module. What Danfoss calls "App Module" is Wi-Fi, and what Danfoss calls "Radio Module" is a Sub-GHz radio - neither are Zigbee, and neither can be joined in deCONZ. Make sure that module you're using is this one (check the part number): https://store.danfoss.com/en/Climate-Solutions-for-heating/Hydronic-floor-heating/Room-Controls/Floor-Heating-Controls%2C-Danfoss-Icon%2C-Zigbee-module%2C-5-0-V%2C-Attachable/p/088U1130

If you do have the Zigbee module, Danfoss' installation instructions are easy to follow: https://assets.danfoss.com/documents/36851/AN32293719605201-010102.pdf

brnstd-dev, 2021-12-27:
How to link the thermostats? I see the zigbee module in deconz, but can't find any information on how to proceed.

PR #4423 (2021-02-22): Add basic support for Danfoss Icon Zigbee Module
Resolves #4273 

The device is a bit tricky to join. It's basically a bridge exposing each In-Room Thermostat in a Danfoss Icon floor heating controller as an endpoint. It takes a while for all the simple descriptors and clusters to be read (the Danfoss literature suggests up to 3 minutes), so the process sometimes fails in Phoscon. I've had success by starting the "Add Sensor" process in Phoscon, and then manually reading the "Node Descriptor" and "Simple Descriptors" in the GUI, once a node for it shows up, followed by manually reading the Bas [... cut]