dresden-elektronik/deconz-rest-plugin #4276: IKEA Trådfri Shortcut button

enjikaka, 2021-01-26:
## Device

- Product name: IKEA Trådfri Genvägsknapp
- Manufacturer: IKEA of Sweden
- Model identifier: TRADFRI SHORTCUT Button
- Device type : Please remove all unrelated device types. 
  - Switch
  - Remote?
[image]
[image]

## Screenshots

ebaauw, 2021-01-26:
Support was already added in https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3975.

closed 2021-01-26 by ebaauw

enjikaka, 2021-01-26:
Ah ok, the hassio addon is very outdated then. Hoping for an update to that soon.