dresden-elektronik/deconz-rest-plugin #4280: possibility to perform actions via http get request

helbgd, 2021-01-27:
## Feature request type

A Lot of IOT Devices support only HTTP get requests (simple url) to trigger state changes.
Please add support for state change actions via HTTP get requests in your api and not only via http put requests.

## Description

In my example I want to use shelly i3 URL Action to toggle a LED Bulb unfortunately no http put is supported there, it only supports http get requests

## Considered alternatives

use a wrapper for example like the one in the url to execute a curl command: https://gist.github.com/senko/4491981

## Additional context

please support url's like this via your api:

as HTTP Get request
http://xxx:80/api/<apikey>/lights/<id>/state?action=on

MadMonkey87, 2021-01-27:
Why not use a proxy or an api gateway? Performing action using get requests seems very unproper and also might not b a god way to go, for example if you need to provide a lot and nested parameters

manup, 2021-01-29:
I'd vote against this, the API is already a complex beast, supporting GET beside PUT would be a huge pile of work and hard to maintain. Therefore a wrapper script seems to be the best shot.

ebaauw, 2021-01-29:
Also against.  It doesn’t seem very RESTful to use GET for updates.  https://en.wikipedia.org/wiki/Representational_state_transfer

Pe-MaKer, 2022-01-02:
> Why not use a proxy or an api gateway? Performing action using get requests seems very unproper and also might not b a god way to go, for example if you need to provide a lot and nested parameters

What kind of gateway could we use here? 
I'm running deconz on a dedicated Pi Zero W and it would be great to have the gateway on the same hardware to minimize the points of failure. I don't think the workload of the gateway would would be too heavy. 
So is there any gateway script or service that can be run on a rasbeeGW?

Pe-MaKer, 2022-03-10:
I am still very interested in a solution.

I have my entire house full of Shellys behind every wall switch and in the fuse box.

I want the Shellys to be able to control my lights directly via deCONZ instead of cutting off their power and tearing a hole in my Zigbee mesh.

Home Assistant as middleware is welcome to take over automation of comfort functions. But simple tasks such as on, off, set the scene should work at all times. I restart Home Assistant several times a week. deCONZ is only restarted with every update.
That is a huge difference with regard to the WAF.

Unfortunately, the engaged Shelly community can't help here. The only option would be an alternative firmware such as ESPHOME, but this would become a maintenance nightmare over time and the many devices.

The API gateway mentioned by @MadMonkey87 should ideally run on the RaspiGW.
Who can build something like this?

Mimiix, 2022-03-10:
Isn't there a way to do a POST request?