dresden-elektronik/deconz-rest-plugin #4288: TUYA curtain motor

NMGAndy, 2021-01-28:
## Device

- Product name: TUYA CURTAIN MOTOR
- Manufacturer: _TZE200_nogaemzt
- Model identifier: TS0601
- Device type :  
  - Other: Window Covering

## Screenshots

[image]
[image]

Smanar, 2021-01-28:
Hello, easy to add, but are you able to compile the code for test ? You just need a linux but with full OS ?

NMGAndy, 2021-01-29:
> Hello, easy to add #4223 , but are you able to compile the code for test ? You just need a linux but with full OS ?

I use it with docker on synology nas. Have no linux machine.

Smanar, 2021-01-29:
Ok, so will be a surpise, but your device is router, so I don't think we can have problem.

NMGAndy, 2021-02-02:
> Hello, easy to add #4223 , but are you able to compile the code for test ? You just need a linux but with full OS ?

Compiled the code and tested it with my device. Now I see open/close buttons in Home Assistant. They work.
The questions is how can I calibrate the motor now?

Smanar, 2021-02-02:
Good question, there is no command for that on cluster 0xEF00 yet, what is written on the manual ? need to use the app, a button, a remote ?

closed 2021-02-15 by manup

NMGAndy, 2021-02-17:
> Good question, there is no command for that on cluster 0xEF00 yet, what is written on the manual ? need to use the app, a button, a remote ?

I found how to calibrate the motor. It calibrates automatically when its shaft is blocked. I currently have no curtain rails so I blocked motor shaft manually to calibrate it (opened/closed positions).

But there is a problem. When I click OPEN button in Home Assistant, CLOSE button becomes disabled (not OPEN).
So If I open curtain I cannot close it. Or if I close curtain I cannot open it - CLOSE button still active, OPEN button disabled.

Smanar, 2021-02-17:
>But there is a problem. When I click OPEN button in Home Assistant, CLOSE button becomes disabled (not OPEN).
So If I open curtain I cannot close it. Or if I close curtain I cannot open it - CLOSE button still active, OPEN button disabled.

It mean the state is reversed.
Are you using the normal version or the one with modification ?

NMGAndy, 2021-02-18:
> It mean the state is reversed.
> Are you using the normal version or the one with modification ?

Version of what?

Smanar, 2021-02-18:
Deconz version ^^. Because the PR was validated on the last.
Are are you still on the version you have recompiled ?

And if you can share the device json ? I just need bri/on/lift/open.

Smanar, 2021-02-21:
Need more information @NMGAndy ?

NMGAndy, 2021-04-06:
> Need more information @NMGAndy ?

Sorry for long reply.

I'm currently on latest firmware with same problem.
 "alert": "none",
            "bri": 254,
            "lift": 100,
            "on": false,
            "open": true,
            "reachable": true

Smanar, 2021-04-06:
Ok, so there is a problem yes, On/open are the reverse of bri/lift.
Wich one deconz version are you using, the firmware is not important.
Have you tried the "reverse" command ? was repaired on last deconz version, but as you have a reverse only in 50% of command, not sure it can solve your issue.

NMGAndy, 2021-04-07:
> Wich one deconz version are you using, the firmware is not important.
> Have you tried the "reverse" command ? was repaired on last deconz version, but as you have a reverse only in 50% of command, not sure it can solve your issue.

Version 2.10.04
Reverse command - where I can find it? 
I am using Home Assistant. There are only 2 buttons UP and DOWN and position slider to manipulate the motor in HA.

NMGAndy, 2021-04-07:
Tried to use the motor with Node Red instead of Home Assistant.
In Node Red I can use Switch command (TRUE - closes curtains, FALSE - opens).
Also I can use Brightness - 0 - closes curtains, 255 - opens.
Is it right behaviour?

Smanar, 2021-04-07:
The reverse command is repair only in 2.10.05 (or when you compile the code yourself)

In Node Red I can use Switch command (TRUE - closes curtains, FALSE - opens).
Also I can use Brightness - 0 - closes curtains, 255 - opens.

It s right for Switch command but not for Brigtness, I have made a change on the code, can you try it ?

```
sudo apt install deconz-dev
git clone --branch tuya_cov_4 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

You will have a "native" mode with command in the good order (if it work), don't need to re-include the device, just restart.

BTW, I have see your closed issue for another curtain motor, you have 2 motors with 2 differents manufacture name ?

NMGAndy, 2021-04-08:
> It s right for Switch command but not for Brigtness, I have made a change on the code, can you try it ?

Sure! I'll try it. Thank you!

> BTW, I have see your closed issue for another curtain motor, you have 2 motors with 2 differents manufacture name ?

Yes. I have both motors. No differences in clusters. Just manufacture name.

Mimiix, 2021-04-08:
@Smanar I'd like to have a full issue on it with screenshots. Tuya is way to unpredictable to assume it's the same.  If you integrate it, please stick to #4723

NMGAndy, 2021-04-08:
> sudo apt install deconz-dev
> git clone --branch tuya_cov_4 https://github.com/Smanar/deconz-rest-plugin.git
> cd deconz-rest-plugin
> qmake && make -j2
> sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins

Just tried this version. Everything is ok now with _TZE200_nogaemzt motor.
But [_TZE200_zpzndjez](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4723) is reversed.

Smanar, 2021-04-09:
For the _TZE200_zpzndjez, it s not a new device, so if I reverse it, I will broke the installation of users that already use it.
So pls wait for some return first https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4521

NMGAndy, 2021-04-10:
> For the _TZE200_zpzndjez, it s not a new device, so if I reverse it, I will broke the installation of users that already use it.
> So pls wait for some return first #4521
> 
> Edit:
> BTW, the PR was in a not published version the (v2.10.5)
> So yes, I can reverse it without problem, but I prefer wait a least 2/3 days for another return.

Ok. Thank you!

Smanar, 2021-04-14:
@NMGAndy I have no answer from other user, so have reversed it too

Can you make tries again ? Same branch

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

NMGAndy, 2021-04-15:
> @NMGAndy I have no answer from other user, so have reversed it too
> Can you make tries again ? Same branch

@Smanar Both motors works properly now. Thank you!

Smanar, 2021-04-15:
Ok, thx, making the PR.

shlomki, 2021-11-18:
Having some trouble with this curtain motor, it's only recognized as a Smart Plug.
I'm running the latest version of deconz on beta (2.13.02 / 11/10/2021). I've tried to do inclusion twice without deleting, but it stays as smart plug. I also tried to delete and include again - still smart plug.
There was only one time that it was successful in becoming a covering device, but the up/down buttons in HA didn't work, so I decided to delete and try again, since then it's not being recognized as covering device.

Any ideas? @NMGAndy is it still working correctly for you? Which version of deconz do you have?

shlomki, 2021-11-18:
I tried to turn on debug logs now and do inclusion again, all of the sudden it's being recognized correctly as a window cover device. Some voodoo in here maybe :)

Smanar, 2021-11-18:
nope, it s a classic deconz issue.
It 's for that tuya device need 2 inclusions, to make short deconz have problem to reconize them (they are zigbee compliant, but not realy respect it, for exemple the firmware realy show him as plug).
Will be solved in some month with DDF file.