dresden-elektronik/deconz-rest-plugin #4291: Deconz would not start

WernerSch, 2021-01-28:
## Describe the question or issue you are having

i install deconz on a Raspberry Pi where iobroker runs on it. When i use the command "systemctl status deconz" i get always this

● deconz.service - deCONZ: ZigBee gateway -- REST API
Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
Active: activating (auto-restart) (Result: exit-code) since Thu 2021-01-28 08:47:52 CET; 2s ago
Process: 2555 ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80 (code=exited, status=217/USER)
Main PID: 2555 (code=exited, status=217/USER)

Jan 28 08:47:52 raspberrypi systemd[1]: deconz.service: Failed with result 'exit-code'.

## Screenshots

## Environment

Its a Raspberry Pi with
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

Jan 28 19:55:25 raspberrypi systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.
Jan 28 19:55:25 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Jan 28 19:55:25 raspberrypi systemd[1]: deconz-init.service: Start request repeated too quickly.
Jan 28 19:55:25 raspberrypi systemd[1]: deconz-init.service: Failed with result 'start-limit-hit'.
Jan 28 19:55:25 raspberrypi systemd[1]: Failed to start deCONZ: ZigBee gateway -- Initialisation.
Jan 28 19:55:25 raspberrypi systemd[22656]: deconz.service: Failed to determine user credentials: No such process
Jan 28 19:55:25 raspberrypi systemd[22656]: deconz.service: Failed at step USER spawning /usr/bin/deCONZ: No such process
Jan 28 19:55:25 raspberrypi systemd[1]: deconz.service: Main process exited, code=exited, status=217/USER
Jan 28 19:55:25 raspberrypi systemd[1]: deconz.service: Failed with result 'exit-code'.
Jan 28 19:55:25 raspberrypi systemd[1]: deconz-init.service: Start request repeated too quickly.
Jan 28 19:55:25 raspberrypi systemd[1]: deconz-init.service: Failed with result 'start-limit-hit'.
Jan 28 19:55:25 raspberrypi systemd[1]: Failed to start deCONZ: ZigBee gateway -- Initialisation.
Jan 28 19:55:55 raspberrypi systemd[1]: deconz.service: Service RestartSec=30s expired, scheduling restart.
Jan 28 19:55:55 raspberrypi systemd[1]: deconz.service: Scheduled restart job, restart counter is at 1337.
Jan 28 19:55:55 raspberrypi systemd[1]: deconz-init.service: Start request repeated too quickly.
Jan 28 19:55:55 raspberrypi systemd[1]: deconz-init.service: Failed with result 'start-limit-hit'.
Jan 28 19:55:55 raspberrypi systemd[1]: Failed to start deCONZ: ZigBee gateway -- Initialisation.
Jan 28 19:55:55 raspberrypi systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.
Jan 28 19:55:55 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Jan 28 19:55:55 raspberrypi systemd[22705]: deconz.service: Failed to determine user credentials: No such process
Jan 28 19:55:55 raspberrypi systemd[22705]: deconz.service: Failed at step USER spawning /usr/bin/deCONZ: No such process
Jan 28 19:55:55 raspberrypi systemd[1]: deconz.service: Main process exited, code=exited, status=217/USER
Jan 28 19:55:55 raspberrypi systemd[1]: deconz.service: Failed with result 'exit-code'.
Jan 28 19:55:55 raspberrypi systemd[1]: deconz-init.service: Start request repeated too quickly.
Jan 28 19:55:55 raspberrypi systemd[1]: deconz-init.service: Failed with result 'start-limit-hit'.
Jan 28 19:55:55 raspberrypi systemd[1]: Failed to start deCONZ: ZigBee gateway -- Initialisation.
Jan 28 19:56:25 raspberrypi systemd[1]: deconz.service: Service RestartSec=30s expired, scheduling restart.
Jan 28 19:56:25 raspberrypi systemd[1]: deconz.service: Scheduled restart job, restart counter is at 1338.
Jan 28 19:56:25 raspberrypi systemd[1]: deconz-init.service: Start request repeated too quickly.
Jan 28 19:56:25 raspberrypi systemd[1]: deconz-init.service: Failed with result 'start-limit-hit'.
Jan 28 19:56:25 raspberrypi systemd[1]: Failed to start deCONZ: ZigBee gateway -- Initialisation.
Jan 28 19:56:25 raspberrypi systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.
Jan 28 19:56:25 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
[... 27 lines cut ...]
Jan 28 19:57:26 raspberrypi systemd[1]: Failed to start deCONZ: ZigBee gateway -- Initialisation.
Jan 28 19:57:26 raspberrypi systemd[22884]: deconz.service: Failed to determine user credentials: No such process
Jan 28 19:57:26 raspberrypi systemd[22884]: deconz.service: Failed at step USER spawning /usr/bin/deCONZ: No such process
Jan 28 19:57:26 raspberrypi systemd[1]: deconz.service: Main process exited, code=exited, status=217/USER
Jan 28 19:57:26 raspberrypi systemd[1]: deconz.service: Failed with result 'exit-code'.
Jan 28 19:57:26 raspberrypi systemd[1]: deconz-init.service: Start request repeated too quickly.
Jan 28 19:57:26 raspberrypi systemd[1]: deconz-init.service: Failed with result 'start-limit-hit'.
Jan 28 19:57:26 raspberrypi systemd[1]: Failed to start deCONZ: ZigBee gateway -- Initialisation.
Jan 28 19:57:36 raspberrypi systemd[1]: Reloading.
Jan 28 19:57:45 raspberrypi systemd[1]: Starting deCONZ: ZigBee gateway -- Initialisation...
Jan 28 19:57:45 raspberrypi systemd[1]: deconz-init.service: Succeeded.
Jan 28 19:57:45 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- Initialisation.
Jan 28 19:57:56 raspberrypi systemd[1]: deconz.service: Service RestartSec=30s expired, scheduling restart.
Jan 28 19:57:56 raspberrypi systemd[1]: deconz.service: Scheduled restart job, restart counter is at 1341.
Jan 28 19:57:56 raspberrypi systemd[1]: Starting deCONZ: ZigBee gateway -- Initialisation...
Jan 28 19:57:56 raspberrypi systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.
Jan 28 19:57:56 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Jan 28 19:57:56 raspberrypi systemd[22982]: deconz.service: Failed to determine user credentials: No such process
Jan 28 19:57:56 raspberrypi systemd[22982]: deconz.service: Failed at step USER spawning /usr/bin/deCONZ: No such process
Jan 28 19:57:56 raspberrypi systemd[1]: deconz.service: Main process exited, code=exited, status=217/USER
Jan 28 19:57:56 raspberrypi systemd[1]: deconz.service: Failed with result 'exit-code'.
Jan 28 19:57:56 raspberrypi systemd[1]: deconz-init.service: Succeeded.
Jan 28 19:57:56 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- Initialisation.

## Additional context

Smanar, 2021-01-28:
Are you using docker or classic OS ?
Try before 
```
$sudo systemctl stop deconz
$sudo systemctl stop deconz-gui
```

WernerSch, 2021-01-28:
Hello

i have no desktop OS.
i follow this  ( https://phoscon.de/en/conbee2/install#raspbian ) 
But i forget a important point. On the raspberry Pi  iobroker is running

Disable the gui doesnt help.

best regards
Werner

Smanar, 2021-01-29:
Ok so try to run deconz using debug mode > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags

```
$sudo systemctl stop deconz
$sudo systemctl stop deconz-gui
$/usr/bin/deCONZ -platform minimal --dbg-info=2 --dbg-error=2 
```

WernerSch, 2021-01-29:
Hello 
in the Attachnemt there is the debug info from putty terminal. 
But i see that there is no folder deconz in /usr/bin

best reagrds
[debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5893892/debug.txt)

Smanar, 2021-01-29:
For me it s look like a communication problem beetween the app and the key.

Can you check
`ls -l /dev/ttyACM*`
`crw-rw---- 1 root dialout 166, 0 27. Jun 11:39 /dev/ttyACM0`

wich one installation procedure have you followed ? You have not skipped
`sudo gpasswd -a $USER dialout`

WernerSch, 2021-01-29:
Hello

i checked it and it looks the same.

No i have not skipped the installation. 

on a helpfull page there are two steps which have be to do
can this be the problem?

1 | sudo nano /boot/config.txt
1 | enable_uart=1

1 | sudo nano /boot/cmdline.txt
delete console=serial0

best regards

Smanar, 2021-01-30:
This concern only Raspbee, not conbee.

Can you try to close all deconz instances again, and use
`ps ax | grep deCONZ`

It will display if you have another one somewhere

Do you have other USB plug ? can you make a try without them ?

You can make a test on another machine (a PC ?) to be sure it s not from the conbee ?

WernerSch, 2021-01-30:
Hello

i try to close all deconz but there is a password needed. The password from iobroker does not work. Then i get this

$ ps ax | grep deCONZ
  380 ?        Ss     0:56 /bin/bash /usr/bin/deCONZ-WIFI2.sh
  384 ?        Ss     0:49 /bin/bash /usr/bin/deCONZ-update2.sh
17225 pts/0    S+     0:00 grep --color=auto deCONZ

Yes i have an rflink Adapter with an Arduino Mega plugged. I disable the Adapter in iobroker and unplug the Arduino.
But its the same

 sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: e
   Active: activating (auto-restart) (Result: exit-code) since Sat 2021-01-30 10
  Process: 2607 ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80 (code
 Main PID: 2607 (code=exited, status=217/USER)

Jan 30 10:52:51 raspberrypi systemd[1]: deconz.service: Failed with result 'exit

best regards

WernerSch, 2021-01-30:
Hello

i checked now the Conbee II adapter on a raspberry Pi 2 wirh Raspbian Buster light
There its works

sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: e
   Active: active (running) since Sat 2021-01-30 10:39:27 GMT; 5s ago
 Main PID: 528 (deCONZ)
    Tasks: 4 (limit: 2063)
   CGroup: /system.slice/deconz.service
           └─528 /usr/bin/deCONZ -platform minimal --http-port=80

Jan 30 10:39:27 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- REST A
Jan 30 10:39:28 raspberrypi deCONZ[528]: QStandardPaths: XDG_RUNTIME_DIR not set
Jan 30 10:39:29 raspberrypi deCONZ[528]: libpng warning: iCCP: known incorrect s

best regards

Smanar, 2021-01-30:
Os so probably from iobroker, but I don't know how it work ...
What is this story with password ?

Can you try to disable all services and run deconz with command line

```
$sudo systemctl disable deconz
$sudo systemctl disable deconz-gui
$sudo systemctl stop deconz
$sudo systemctl stop deconz-gui
$/usr/bin/deCONZ -platform minimal --http-port=80
```

WernerSch, 2021-01-30:
Hello

i do your test. Here is the result shown in putty

/usr/bin/deCONZ -platform minimal --http-port=80
QCoreApplication::arguments: Please instantiate the QApplication object first
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()

best regards

closed 2021-03-04 by github-actions[bot]