dresden-elektronik/deconz-rest-plugin #4292: Device name

Maddinthebrain, 2021-01-28:
## Device

- Product name: tint Fernbedienung white [https://www.mueller-licht.de/produktinformationen/artikel/404022/](url)
- Manufacturer: MLI
- Model identifier: tint-Remote white
- Device type : 
  - Remote

 

## Screenshots

Mimiix, 2021-01-28:
Without screenshots, we can't do anything.

Maddinthebrain, 2021-01-28:
> Without screenshots, we can't do anything.

Well I'm on it. But Github doesn't let me....
I try it again.

closed 2021-03-04 by github-actions[bot]

Maddinthebrain, 2021-03-20:
Can I support this anyhow?

Mimiix, 2021-03-20:
Need to wait for a dev to do it. Or you need to figure out yourself.

Maddinthebrain, 2021-03-21:
Okay, that's exactly my question how can help? What steps and tools are needed?

Maddinthebrain, 2021-03-22:
@Mimiix

Maddinthebrain, 2022-01-01:
Hello there, 

is there any action about this remote? Unfortunatly the new API featurer doesn't help here. The device only appears in the deconz gui but not in the phoscon app. How can I support?