dresden-elektronik/deconz-rest-plugin #4294: tuya ZigBee & RF AC Triac Dimmer S1-B(WZ)

stevenwfoley, 2021-01-29:
## Device
- Product name: ZigBee & RF AC Triac Dimmer
- Manufacturer: _TZE200_6qoazbre
- Model identifier: TS0601
- Device type:
  - Light
  - Switch

https://www.aliexpress.com/item/1005001942473093.html
https://www.superlightingled.com/zigbee-rf-smart-ac-switch-ssbwz-p-3810.html
http://www.iskydance.com/index.php?c=product_show&a=index&id=1616

## Screenshots
[image]
[image]

Smanar, 2021-01-29:
Hello, you have no light device created at all ?

stevenwfoley, 2021-01-29:
Its created as a Plug device, whereas I believe it should be a Light device. And it's not responding to any commands.
It also appears that the device might be mistakenly associated as some other device because of the "Fairy lights" model text.
[image]

Smanar, 2021-01-29:
Ok, so I don't see the problem.

- Can you show the device JSON, visible in the API or in third app
- This device have a switch entry, so we can use it to check the tuya command.

- Wire the entry
- Go in deconz/help/log with "info" "info_l2"
- press the switch and give me the command in log starting by "debug tuya" pls.

stevenwfoley, 2021-01-29:
Here is the device JSON:
{
  "etag": "5127781c5626a9cb4fb61758199cd1d0",
  "hascolor": false,
  "lastannounced": null,
  "lastseen": "2021-01-29T18:01Z",
  "manufacturername": "_TZE200_6qoazbre",
  "modelid": "TS0601",
  "name": "Smart plug 40",
  "state": {
    "alert": "none",
    "on": false,
    "reachable": true
  },
  "swversion": null,
  "type": "Smart plug",
  "uniqueid": "84:2e:14:ff:fe:0e:12:cb-01"
}

Smanar, 2021-01-29:
"Bad" luck, the json is good, so I think it s because tuya is using a new command.
And I don't see this device on other project, so I realy need the command visible in log to reproduce it.

stevenwfoley, 2021-01-29:
What about the device showing as a type of "Smart Plug"? Shouldn't it be a "Dimmable Light"?
The device does not have any physical switches, so I don't know how to have it send commands.
I was able to put it in a group in Phoscon, as you can see at the bottom of the last picture.

Smanar, 2021-01-30:
>What about the device showing as a type of "Smart Plug"? Shouldn't it be a "Dimmable Light"?

It s not me that choose ^^, it s written on the firmware. But you are right, I thing third app will be not able to use it like that.
I can force a now type in the api or just add the field "bri", but will be not the priority, first the on/off.

Here you have the wiring to add a switch
[image]

>I was able to put it in a group in Phoscon, as you can see at the bottom of the last picture.

Yep, but as it, you are sending group command, and this device don't use standard command, you need to use the bulb that is alone in phoscon, or play with the "single / all", I don't remember, but it s not easy with phoscon.

So it mean you don't use third app ?

stevenwfoley, 2021-01-30:
Thanks. This makes sense, and I now see that I can power on/off the device individually but not at the group level. It appears that this device will not function as I expected, so I'm going to return it. Please let me know if you want more info about the device for better integration into deconz/phoscon, before I return the device.

Smanar, 2021-01-30:
>It appears that this device will not function as I expected

What is missing for you ? the group feature ? This device realy seem usefull for me

stevenwfoley, 2021-01-30:
Most importantly for me is the zigbee groups cluster functionality, which is not working. I added it to a group with a philips hue dimmer, and it does not respond to any of the commands from the switch. The device also runs really hot.
I agree that the specs seem very attractive, but I just seems to lack proper zigbee compatibility.

Smanar, 2021-01-31:
Ha yes, you need to use a group feature made by your third app, you can't use zigbee one.
For exemple HUE use the cluster 0x0006 for on/off and 0x0008 for level control, so it s thoses command that are send in group, but most of tuya device use the cluster 0xEF00, so even if they support zigbee group command, they are not compatible for me.

Not possible to make a direct bind with a HUE remote and this device, you need to use third app. But the device stall nice for me, the temperature is more worrying  with chinese stuff without certifications ^^.

closed 2021-03-04 by github-actions[bot]