dresden-elektronik/deconz-rest-plugin #4298: Alcantara 2 acova

geekance, 2021-01-29:
## Device

- Product name: Alcantara 2 acova
- Manufacturer: ZEHNDER GROUP VAUX ANDIGNY
- Model identifier: ALCANTARA2 D1.00P1.01Z1.00
- Device type : Please remove all unrelated device types. 
  - Thermostat

## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

geekance, 2021-02-10:
Is the process to get this device whitelisted automatic once the request is open or should I do something else?

Mimiix, 2021-02-10:
There's not much more to do than waiting. A dev needs to pick it up :)

NicoB35, 2021-03-09:
My request #4129 (same heater) as been closed for no activity.
I'll try to keep this one alive.

closed 2021-04-22 by manup