dresden-elektronik/deconz-rest-plugin #4299: How to Add SYMFONISK Sound Controller

talun2075, 2021-01-29:
## Describe the question or issue you are having

I want to add the Ikea SYMFONISK Sound Controller over the Phoscon App but this Kind of switch is not in the app and I don´t know other ways to add a new switch to the system.
So I tryed the both Ikea Switches and the generic switch from "other".
I try to Reset the Switch with pressing 10 Seconds written in the manual and 4 times 5 seconds like othere ikea switches.
From the Changelog I have seen that this kind of Switch is supportet so I wonder how to add it.

Is there other way to add this switch except the Phoscon UI?

## Screenshots

## Environment

 - 
    Host system: Windows
    Running method: Windows
    Firmware version: (26680700)
    deCONZ version: (2.09.01)
    Device: (ConBee II)
    Do you use an USB extension cable: (yes)
    Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

cubertt, 2021-01-29:
just added the symfonisk some minutes ago, think i only looked for general light oder general switch and it apperad (after pressing the reset button 4 times). r

talun2075, 2021-01-31:
Hi,
I have found a solution. It was a defect CR2032.
After change I need 5 Minutes to add it to the system.
Greate wtf

closed 2021-01-31 by talun2075