dresden-elektronik/deconz-rest-plugin #4300: Livolo Zigbee Power Socket TI0001 Device Request

bsfaxi, 2021-01-29:
## Device

- Product name: Livolo Zigbee Power Socket.
- Manufacturer: LIVOLO.
- Model identifier: TI0001.
- Device type : Please remove all unrelated device types. 
  - Switch
  - Other: Power Socket

## Screenshots

[image]
[image]

Nothing in the On/Off cluster!
[image]

Here is a link to the product : https://www.aliexpress.com/item/4000343575116.html?spm=a2g0s.9042311.0.0.49316c37sVuqR1.

I found an interesting link on this brand (not the same product) here: https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Livolo-corner.md.

bsfaxi, 2021-02-04:
anyone please?

Mimiix, 2021-02-04:
Bumping wont work. Devs need to integrate it. But even if they do, chances are slim that it could be integrated soon. as theres 36 PR's waiting.

bsfaxi, 2021-02-04:
Thanks @Mimiix.
I understand. Bumping wasn't my goal. :)
I had to request for new devices in the past, and I had the habit to receive a response in the same day.
I have to be patient! ;)

Smanar, 2021-02-04:
On m y side I m not realy motivated, sorry, your device look realy strange (clusters are not standard) , and last time I havez tried with livolo device was not a success.

bsfaxi, 2021-02-04:
> On m y side I m not realy motivated, sorry, your device look realy strange (clusters are not standard) , and last time I havez tried with livolo device was not a success.

Hi @Smanar,
I know. :)
Please know that I have a dedicated DEV environment. We can play with it as you wish. :)

Smanar, 2021-02-04:
Lol, nice, but seriously there is no usefull cluster, so it can be hidden, or another magic one like for tuya, so lot of search to do.

bsfaxi, 2021-02-04:
Too much work was already done here as mentioned in the initial request :  https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Livolo-corner.md.

Extract:
_"On and OFF are not based on the Cluster 0x0006, but rather on cluster 0x0008. Don't ask me why ... Livolo migh have a reason for that !
On the Cluster 0x0008 (Level Control) Livolo are using those 2 parameters."_

Smanar, 2021-02-04:
Yeah I know, It that I have used for the livolo switch, to finaly give up, too much work for something not enought stable.

bsfaxi, 2021-02-08:
Hi @Smanar,
If a beer would motivate you, I would be more than happy to offer you one, virtually or whenever you come to Paris in person. ;)

Smanar, 2021-02-08:
Lol, you realy want this device ^^.
But seriously even on other project, there is so much problem with this kind of device, will be easier to find another one.

bsfaxi, 2021-02-08:
> Lol, you realy want this device ^^.
> 
> But seriously even on other project, there is so much problem with this kind of device, will be easier to find another one.

Let's give it a quick try then. I have 3 of them. If it's so much time, we stop.
BTW, could you point me to an equivalent Zigbee euro wall socket, that works already with deCONZ ?

Smanar, 2021-02-09:
I think I have still the old code for livolo somewhere on my hard disk, but too few time ATM.
Plug or "in wall" models ?

bsfaxi, 2021-02-09:
Thanks @Smanar .
It's an "in wall" socket.
link : https://www.aliexpress.com/item/4000343575116.html?spm=a2g0s.9042311.0.0.49316c37sVuqR1.

Smanar, 2021-02-09:
I m using the legrand netatmo one, good product but still not comptibale with Xiaomi and channel locked
This one is not bad https://www.niko.eu/fr-be/article/170-33505 from my memory swoop have already tested it too.

bsfaxi, 2021-02-10:
> I m using the legrand netatmo one, good product but still not comptaible with Xiaomi and channel locked
> This one is not bad https://www.niko.eu/fr-be/article/170-33505 from my memory swoop have already tested it too.
> 
> Edit:
> #2984

Thanks Smanar. They are indeed good ones.
The only show stopper for me is that they don't have a physical on/off button. If I loose my Zigbee network, or if grandfather wants to turn on the TV and the switch is off, you can imagine the situation... lol

Smanar, 2021-02-10:
>The only show stopper for me is that they don't have a physical on/off button. If I loose my Zigbee network, or if grandfather wants to turn on the TV and the switch is off, you can imagine the situation... lol

Ha yes, was a criter in the "in wall plug search", but the niko one have it. Can use too a remote with a direct bind (work even if the gateway is off)

bsfaxi, 2021-02-10:
That sounds really great. Thanks a lot.
I found one with a good price here: https://www.dmlights.fr/niko_home_control_prise_de_courant_connectee_commutable~18G15.
I will order one for test.
Meanwhile, could we have a quick test for the LIVOLO one. :)

Smanar, 2021-02-10:
lol tenacious.
Not with me, I can share the old file if someone want to test it, but I m realy not interessted, too much work to have too much hack in the code to make a not reliable device working.

bsfaxi, 2021-02-10:
I understand.
Who can help?

closed 2021-03-11 by github-actions[bot]