dresden-elektronik/deconz-rest-plugin #4305: hama Smart Radiator Thermostat

ruehcep, 2021-01-29:
## Device

- Product name: hama Smart Radiator Thermostat
- Manufacturer (as per Cluster Info): _TYST11_yw7cahqs
- Model identifier (as per Cluster Info): w7cahqs
- Device type:
- Device type :
   - Thermostat
 
https://www.hama.com/00176592/hama-smart-radiator-thermostat-for-hama-wlan-heating-control

The device is being identified as a light instead of a Thermostat and in the deconz software it seems to be a TRV Tuya.
I do have a linux machine with full OS to make tests if required.

## Screenshots
[image]
[image]

Smanar, 2021-01-30:
Finally ^^, at third try.

Will add it today, do you have a unix machine to make test ?

ruehcep, 2021-01-30:
All good things come in threes :-D
I have a raspberry running Raspbian, can also quickly setup a different distro if needed.
Many thanks!

Smanar, 2021-01-30:
Nice, so I have aded it on the future Tuya modification, so you will have almost all future tuya change.

I haven't find information about your device, so I have used a basic one, and enable all features. So just use it some time, and take a look after some time, in the json to check the working one or not.

To test it https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

```
sudo apt install deconz-dev
git clone https://github.com/KPWhiver/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

ruehcep, 2021-01-31:
Hi, I hope I am not missing anything basic here ... it is my my first not "out of the box" supported device in deconz.
I now do see a second object (see screenshot below) but otherwise nothing changed. In REST also everything looks the same, only seeing the on/off light (see below).
Could you  explain what you mean with "making two inclusions"?
Thanks for the support!
[image]
[image]

Smanar, 2021-01-31:
2 inclusions/pairings/addings.

ATM we have some problem with tuya stuff, deconz is not able to reconize them, and they are not zigbee compliant.
So at first inclusion, deconz memorise the device with some information, and make a "bad" entry in the api according to it. This device react like a "on/off light", it s for that you have this device created.

At the second inclusion, deconz will look in the database, the previous data to reconize it, and make the correct entry in the API.

It s for that, you need 2 inclusions and without deleting the device (to be sure data are still inside the DB)

closed 2021-03-04 by github-actions[bot]

bmgit26, 2021-04-16:
I installed the rest plugin from git and try to add the Hama thermostate in Phoscon. But the thermostate is not detected in Lights/Sensors/Switches. Do I need to add it in deconz-gui or should it work in phoscon too?

Mimiix, 2021-04-16:
Should only be found in deCONZ/RestAPI

Smanar, 2021-04-16:
Thermostat are not managed by phoscon, you can use phoscon to include the device, but you need a third app to use it.

bmgit26, 2021-04-16:
Great, thanks, I see it in deconz and can use it in Home Assistant.

sedony23, 2021-08-13:
> 2 inclusions/pairings/addings.
>
> ATM we have some problem with tuya stuff, deconz is not able to reconize them, and they are not zigbee compliant.
> So at first inclusion, deconz memorise the device [...]

Hi there,

I also just got these thermostats. How do you include them twice? I added the thermostat through the Phoscon WebUI once and if I try to pair it again, it doesn't show up as new device. I can see it in deCONZ. However I can not control it via homebridge-hue. Some more info would be appreciated. Thanks and all the best.

Daniel
[image]

Smanar, 2021-08-13:
You just need to set phoscon in permit join > add new sensors.
Then make the inclusion procedure 2 times.

Where the name "Thermostat 81" is from ? if it s not you that have set it, it mean the device inclusion have worked.

sedony23, 2021-08-14:
Thanks for your help! Hmm, I added it twice through sensors. I can now control the temperature through homebridge-hue with the Eve app. However I cannot set the mode (heating on or off) nor can I set a schedule. When I toggle the switch, it just goes switching the state and then right back. See screenshot. Any ideas?
[image]

Smanar, 2021-08-14:
Good question ....
From the code it seem this device have the "mode" and the "preset"
But I don't know how work homekit, you can't use direclty the API ? with Curl or Using a Rest client ? https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

I can explain with curl, but I need at least the device Json.

sedony23, 2021-08-14:
OK. I'll check and will post again here!

sedony23, 2021-08-15:
So I ran some tests. Here is what I get from the  REST API:

`{
  "config": {
    "heatsetpoint": 1000,
    "locked": true,
    "mode": "heat",
    "offset": 200,
    "on": true,
    "preset": null,
    "reachable": true,
    "schedule": {},
    "schedule_on": true,
    "setvalve": false,
    "windowopen_set": false
  },
  "ep": 1,
  "etag": "1628e2eb7c431e5de05361db3c4b115c",
  "lastannounced": "2021-08-14T19:18:14Z",
  "lastseen": "2021-08-15T06:51Z",
  "manufacturername": "_TYST11_yw7cahqs",
  "modelid": "w7cahqs",
  "name": "Thermostat",
  "state": {
    "lastupdated": "none",
    "lowbattery": null,
    "on": null,
    "temperature": 2250,
    "valve": null
  },
  "swversion": "20180727",
  "type": "ZHAThermostat",
  "uniqueid": "60:a4:23:ff:fe:b2:93:d0-01-0201"
}`

I can also change the mode from "heat" to "auto" or to "off". However that doesn't change anything on the thermostat. Setting the temperature works though. Any ideas?

Smanar, 2021-08-15:
Nice, thw a lot.
Have found the problem, the "mode" use the wrong code, we need to use DP_IDENTIFIER_THERMOSTAT_MODE_3 for this one.

Are you able to compile to test code modification, or I made the fix in "blind mode" ?

sedony23, 2021-08-15:
Great, thanks for the quick fix! I'm not sure if everything else is ok and what the device actually exposes/uses. Maybe this helps:
[https://www.zigbee2mqtt.io/devices/SEA801-Zigbee_SEA802-Zigbee.html](url)
[https://github.com/Koenkk/zigbee2mqtt/issues/5507](url)
It also seems to be very close to this one: [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3109](url)

Unfortunately I run Docker and can't compile and test easily as the whole rest of the system is working fine.

Smanar, 2021-08-15:
I only need test to confirm ^^.

I have made the PR, hoping I m right.

sedony23, 2021-08-15:
Will it be in the next beta release then? Then I could easily test it.

Smanar, 2021-08-15:
IDK, depend of PR validation, but as it's a small fix, yes we probalby have it in next version.

sedony23, 2021-08-20:
Did the fix make it into the new release? Then I could test it!

Smanar, 2021-08-20:
Nope, not yet, this issue will be closed in same time the PR will be validated.
I think you will have on the next version a beta one.

pillemats, 2021-11-08:
How to change manually ? In rest api ? How was the command?

Smanar, 2021-11-08:
Manualy ? with your hand ^^.
All tuya request are logued, so we will be able to compare it to see if it s the same device, or perhaps this device don't make request at all.

pillemats, 2021-11-08:
:-D oh sorry 

Here is the log 

`21:42:03:012 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0FF5
21:42:03:022 Websocket 192.168.178.204:33144 send message: {"attr":{"id":"56","lastannounced":null,"lastseen":"2021-11-08T20:42Z","manufacturername":"_TYZB01_8scntis1","modelid":"TS0216","name":"Sirene extern","swversion":"2019.6.23","type":"Warning device","uniqueid":"80:4b:50:ff:fe:1a:60:bb-01"},"e":"changed","id":"56","r":"lights","t":"event","uniqueid":"80:4b:50:ff:fe:1a:60:bb-01"} (ret = 329)
21:42:03:103 poll node 84:fd:27:ff:fe:92:81:41-01
21:42:03:103 Poll light node Spots
21:42:03:165 Poll APS request to 0x84FD27FFFE928141 cluster: 0x0006 dropped, values are fresh enough
21:42:03:463 rule event /config/localtime: 21:42:02.462 -> 21:42:03.462 (1000ms)
21:42:04:102 poll node bc:33:ac:ff:fe:74:3a:bd-01
21:42:04:102 Poll light node Fenster Flur
21:42:04:165 Poll APS request to 0xBC33ACFFFE743ABD cluster: 0x0006 dropped, values are fresh enough
21:42:04:306 Node data 0x000d6f001644f398 profileId: 0x0104, clusterId: 0x0001
21:42:04:315 0x000D6F001644F398: update ZCL value 0x01/0x0001/0x0021 after 0 s
21:42:04:315 [INFO] - No button map for: SmokeSensor-EM, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0001, command: 0x0A, payload: 210020C8, zclSeq: 50
21:42:04:316 	payload: 210020c8
21:42:04:316 binding for attribute reporting of ep: 0x01 cluster 0x0001 seems to be active
21:42:04:316 0x000D6F001644F398 (SmokeSensor-EM) create binding for attribute reporting of cluster 0x0500 on endpoint 0x01
21:42:04:317 queue binding task for 0x000D6F001644F398, cluster 0x0500
21:42:04:322 Websocket 192.168.178.204:33144 send message: {"attr":{"id":"141","lastannounced":null,"lastseen":"2021-11-08T20:42Z","manufacturername":"Heiman","modelid":"SmokeSensor-EM","name":"Kinderzimmer","swversion":"1.1.1","type":"ZHAFire","uniqueid":"00:0d:6f:00:16:44:f3:98-01-0500"},"e":"changed","id":"141","r":"sensors","t":"event","uniqueid":"00:0d:6f:00:16:44:f3:98-01-0500"} (ret = 328)
21:42:04:322 Websocket 192.168.178.204:33144 send message: {"config":{"battery":100,"enrolled":1,"on":true,"pending":[],"reachable":true},"e":"changed","id":"141","r":"sensors","t":"event","uniqueid":"00:0d:6f:00:16:44:f3:98-01-0500"} (ret = 175)
21:42:04:462 Idle timer triggered
21:42:04:462 Force read attributes for node Keller 2
21:42:04:462 Force binding of attribute reporting for node Keller 2
21:42:04:463 binding for attribute reporting SensorNode Haust�r Hinten of cluster 0x0400 seems to be active
21:42:04:465 Force read attributes for ZHALightLevel SensorNode Haust�r Hinten
21:42:04:466 binding for attribute reporting of ep: 0x02 cluster 0x0400 seems to be active
21:42:04:466 Force binding of attribute reporting for node Haust�r Hinten
21:42:04:468 rule event /config/localtime: 21:42:03.462 -> 21:42:04.462 (1000ms)
21:42:04:493 Poll APS request to 0xBC33ACFFFE743ABD cluster: 0x0008 dropped, values are fresh enough
21:42:04:603 read attributes of 0xBC33ACFFFE743ABD cluster: 0x0300: [ 
21:42:04:603 0x0003 
21:42:04:603 0x0004 
21:42:04:603 0x0007 
21:42:04:603 0x0008 
21:42:04:603 0x4001 
21:42:04:603 ]
21:42:04:603 add task 57187 type 19 to 0xBC33ACFFFE743ABD cluster 0x0300 req.id 127
21:42:04:603 Poll APS request 127 to 0xBC33ACFFFE743ABD cluster: 0x0300
21:42:04:665 Poll APS confirm 127 status: 0x00
21:42:04:679 Erase task req-id: 127, type: 19 zcl seqno: 94 send time 0, profileId: 0x0104, clusterId: 0x0300
21:42:04:728 Node data 0xbc33acfffe743abd profileId: 0x0104, clusterId: 0x0300
21:42:04:740 0xBC33ACFFFE743ABD: update ZCL value 0x01/0x0300/0x0003 after 0 s
21:42:04:740 0xBC33ACFFFE743ABD: update ZCL value 0x01/0x0300/0x0004 after 0 s
21:42:04:740 0xBC33ACFFFE743ABD: update ZCL value 0x01/0x0300/0x0007 after 0 s
21:42:04:740 0xBC33ACFFFE743ABD: update ZCL value 0x01/0x0300/0x0008 after 0 s
21:42:04:741 0xBC33ACFFFE743ABD: update ZCL value 0x01/0x0300/0x4001 after 0 s
21:42:04:742 Websocket 192.168.178.204:33144 send message: {"attr":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"id":"26","lastannounced":"2021-10-29T16:32:08Z","lastseen":"2021-11-08T20:42Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E14 WS 470lm","name":"Fenster Flur","swversion":"2.1.022","type":"Color temperature light","uniqueid":"bc:33:ac:ff:fe:74:3a:bd-01"},"e":"changed","id":"26","r":"lights","t":"event","uniqueid":"bc:33:ac:ff:fe:74:3a:bd-01"} (ret = 416)
21:42:05:204 add task 57190 type 21 to 0x0017880102874886 cluster 0x0004 req.id 131
21:42:05:216 Erase task req-id: 131, type: 21 zcl seqno: 95 send time 0, profileId: 0x0104, clusterId: 0x0004
21:42:05:252 verified group capacity: 23 and group count: 2 of LightNode 0x0017880102874886
21:42:05:252 0x0017880102874886 found group 0xFFF0
21:42:05:252 0x0017880102874886 found group 0x000A
21:42:05:282 poll node 00:17:88:01:02:ae:ea:9b-0b
21:42:05:282 Poll light node Bett Evelin
21:42:05:348 read attributes of 0x0017880102AEEA9B cluster: 0x0006: [ 
21:42:05:348 0x0000 
21:42:05:348 ]
21:42:05:348 add task 57191 type 19 to 0x0017880102AEEA9B cluster 0x0006 req.id 133
21:42:05:348 Poll APS request 133 to 0x0017880102AEEA9B cluster: 0x0006
21:42:05:450 Poll APS confirm 133 status: 0x00
21:42:05:451 Erase task req-id: 133, type: 19 zcl seqno: 96 send time 0, profileId: 0x0104, clusterId: 0x0006
21:42:05:454 rule event /config/localtime: 21:42:04.462 -> 21:42:05.451 (989ms)
21:42:05:463 Node data 0x0017880102aeea9b profileId: 0x0104, clusterId: 0x0006
[... cut ...]
21:43:23:559 Websocket 192.168.178.204:33144 send message: {"attr":{"id":"28","lastannounced":"2021-09-03T16:23:31Z","lastseen":"2021-11-08T20:43Z","manufacturername":"Philips","modelid":"LWB010","name":"Kugel 2","swversion":"1.50.2_r30933","type":"Dimmable light","uniqueid":"00:17:88:01:02:53:8f:63-0b"},"e":"changed","id":"28","r":"lights","t":"event","uniqueid":"00:17:88:01:02:53:8f:63-0b"} (ret = 336)
21:43:23:697 rule event /config/localtime: 21:43:22.693 -> 21:43:23.696 (1003ms)
21:43:23:931 poll node 80:4b:50:ff:fe:1a:60:99-01
21:43:23:931 Poll light node Sirene Intern `

pillemats, 2021-11-08:
Oh sorry :-D

Here is the log 

https://pastebin.com/mMxY9yLG

Smanar, 2021-11-09:
Its the mac adress 0x5C0272FFFEC21516  ?

Because it have worked

`21:42:54:966 Websocket 192.168.178.204:33144 send message: {"config":{"heatsetpoint":2300,"locked":true,"mode":"heat","offset":0,"on":true,"preset":null,"reachable":true,"schedule":{},"schedule_on":false,"setvalve":false,"windowopen_set":false},"e":"changed","id":"151","r":"sensors","t":"event","uniqueid":"5c:02:72:ff:fe:c2:15:16-01-0201"}`

So you receive command from the device, but not send  it using the API ?

The command for heat point for your device is 

>21:42:54:962 Tuya debug 5 : Status: 0 Transid: 3 Dp: 615 (0x02,0x67) Fn: 0 Data 230

And same on the code ....

Can you show the same log but this time trying with the API (or third app)

pillemats, 2021-11-09:
Yes it is. 

How was the command in rest api? 

Many time ago I done some things in rest api

Smanar, 2021-11-09:
But you have never tried the device ? How you can't know if it work or not ?

The temperature is still "Null" ?

Can try with Curl

`curl -H 'Content-Type: application/json' -X PUT -d '{"heatsetpoint": 2500}' http://IP:PORT/api/APIKEY/sensors/151/config`

pillemats, 2021-11-09:
i dont understand curl.... i run deconz on my win10 VM

I tried via restapi and get error 400
[image]

Smanar, 2021-11-10:
>is it normal that it takes some time (10-15 seconds) if trv get the status update and will actualise the state?

Yes, lot of TRV are lazy.

The "auto" is broken ATM for your device https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5471
But the "off" need to work, and it's "off" instead of off.

pillemats, 2021-11-10:
Ah ok…

Mmh this is strange…. I get always error 400 if i want to turn off

See here:

[image]

Smanar, 2021-11-10:
It's "off" not off.

pillemats, 2021-11-11:
oh sorry, my mistake.

i jst tried now and it will switch but the TRV does nothing.
The state was changed in homebridge and homekit but not on the TRV itself

Smanar, 2021-11-11:
If you have the GUI, can you show the log when making the mode = off command.
Logs with "info" and "info_l2"

pillemats, 2021-11-11:
it should be work right now.... but i get this in homebridge

[11.11.2021, 20:48:16] [Hue] Thermostat 151: homekit target heating cooling state changed from 0 to 1
[11.11.2021, 20:48:16] [Hue] Phoscon-GW: request 27: PUT /sensors/151/config {"mode":"auto"}
[11.11.2021, 20:48:16] [Hue] Phoscon-GW: request 27: api error 608: Could not set attribute
[11.11.2021, 20:48:21] [Hue] Thermostat 151: set homekit target heating cooling state from 1 to 0

Hope the log will help

[log.docx](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7522961/log.docx)

Smanar, 2021-11-12:
Seriously I m totaly lost in deconz version, there is new trv every 2 days, and repair/add one can broke another. And I m sure there is fix for TRV in all new versions.

But yes, ATM the "auto" command is broken and provoke exactlty this message

>pydeconz.errors.pydeconzException: /sensors/55/config/mode Could not set attribute

pillemats, 2021-11-13:
hopefully it will fixed asap :-)

pillemats, 2021-11-17:
How does it go on here ?

Smanar, 2021-11-17:
For the "auto" issue ? still in waiting list https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5471

pillemats, 2021-11-19:
Same problem is with another manufacturer

[image]
[image]

Smanar, 2021-11-19:
What does means "same problem" ? the missing "auto" feature ?

pillemats, 2021-11-19:
Oh sorry…. Yes the Auto feature don’t work

Smanar, 2021-11-20:
Have just updated the PR for this issue, if you can try it ? https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5471

pillemats, 2021-11-20:
Thanks! When should I try it ? I don’t understand I think

Smanar, 2021-11-20:
You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5436#issuecomment-962259657

But you need a linux machine with full OS.

pillemats, 2021-11-20:
Oh i don’t have it… only win10 with deconz

capstan1, 2021-12-13:
So I just got one connected to the conbee and Home Assistant. I can successfully change the temperature. What else should be working? Heating/off/auto doesn't seem to make any difference.

Smanar, 2021-12-13:
Hu ?
I realy don't remember this issue
Wich one deconz version are you using ? If you are on HA with the build in deconz version, you have an old version for sure.
What is the manufacture name of your device ?

capstan1, 2021-12-13:
Version 2.12.06 / 19.8.2021
Firmware 26390500
Didn't have the offer for an update in quite a while.
Manufacturer is shown as _TZE200_yw7cahqs
Model identifier TS0601

Also there is no battery reported....

thanks for tuning in again!

Smanar, 2021-12-13:
There was lot of update since your version, but even me I don't remember all.

capstan1, 2021-12-13:
So it just should be working in the future when HA devs implement newer firmware?

capstan1, 2021-12-16:
Okay, so I'll just wait for the next stable deconz version so it gets into the HA add-on!

Thank you!

hAMMERlized, 2021-12-22:
Good morning guys,

@Smanar i know this issue is closed but as there is already one for my model I didn't want to open a new one.

So I am also using then "_TZE200_yw7cahqs" Version of the Hama radiator with a Conbee 2 Stick and Homebridge. I was able to successfully pair the thermostat even without using the deCONZ GUI through VNC, just searched for a new sensor in the phoscon web app. So far so good.

I'm using the Eve App for controlling my radiators. Only thing that seems to work is heating control and setting the offset. 

"Mode", "Valve Control", "Schedules", "Child Lock", "Chalk Protection" have no function. Even the battery state is missing.

I've also noticed that the temperature is dropping automatically 2 degrees down to what I have set in the Eve App (e.g. from 20 to 18), or even down to 16. I think this has something to do with they "Away Mode" the radiators have coded insides or something like that.

I'm using the latest stable 2.13.04 version. Firmware also the latest.

Anything new you could tell me or helping me or at least help me with this temperature drops? For now like this the thermostat is not useable in my smart home.

Thanks in advance.

Smanar, 2021-12-22:
I m  a little lost in all tuya TRV. So your model is exaclty "_TZE200_yw7cahqs"

Can you show the command send by deconz when you make for exemple "mode" request ? You can see it in deconz log with "info".

>I've also noticed that the temperature is dropping automatically 2 degrees down to what I have set in the Eve App (e.g. from 20 to 18), or even down to 16. I think this has something to do with they "Away Mode" the radiators have coded insides or something like that.

You can't change the mode manually ? You send a 20 °C setpoint, and the device display 18 °C ?

hAMMERlized, 2021-12-22:
> I m  a little lost in all tuya TRV. So your model is exaclty "_TZE200_yw7cahqs"
> 
Yes, it's the exact model (check the Screenshots down below)
> 
> Can you show the command send by deconz when you make for exemple "mode" request ? You can see it in deconz log with "info".
> 
Sorry I'm somehow new into this materia. Where can I find the deconz log "info"? I've uploaded some pictures from the homebridge log after restarting it after pairing the radiators through the Phoscon web app. The pairing process is pretty easy. Didn't expect that. Phoscon does not show the radiators in the sensors list, but they are all in the deconz software listed and passesd through to the homebridge. Or did I miss something? I've also pressed the read button in the "Cluster Info" at "Basic" of each radiator. But still, I can just set the temperature and offset with the Eve App. No battery status and the the other options that I've told about.
> You can't change the mode manually ? You send a 20 °C setpoint, and the device display 18 °C ?

I can change the mode to On/Off. Once I have set the Mode to "off", the radiator display shows OF. So far so good. When hitting the Mode to "On" nothing happens and the radiators stays at Of. So I have to put the temperature manually through the Eve App to 30 degrees (radiator "On") and drop it down to, for example, 22 degrees.

By the way, Home App neither the Eve App show if the radiator is heating or not. In the Home App I see only the green temperature mode (not heating) but the radiators do heat.

I hope you can follow my bad school English skills. Thanks so far.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

capstan1, 2021-12-22:
> I've also noticed that the temperature is dropping automatically 2 degrees down to what I have set in the Eve App (e.g. from 20 to 18), or even down to 16. I think this has something to do with they "Away Mode" the radiators have coded insides or something like that.

Press the button on the valve itself and make sure the clock symbol isn't shown. This auto mode seems to have more times build in and does unexpected things

hAMMERlized, 2021-12-22:
> Press the button on the valve itself and make sure the clock symbol isn't shown. This auto mode seems to have more times build in and does unexpected things
Thanks for the quick response. Yea, I allteady thought that this should be some internal Programm running or something like that. That little clock is blinking while pushing the radiator. How do I get rid of it?

capstan1, 2021-12-22:
For me it disappeared every 2nd push.

hAMMERlized, 2021-12-22:
I've figured it out. While the clock is blinking, double tap the radiator and the temperature will be set to 20. Now set it to the temperature you want and hit the radiator once. The clock should disappear

[5 more replies]