dresden-elektronik/deconz-rest-plugin #4307: Can't connect to IKEA Shortcut button - No button handler for: TRADFRI SHORTCUT Button endpoint

brujoand, 2021-01-30:
## Describe the bug
Trying to add the IKEA Shortcut button doesn't work, and the logs show
```
[INFO] - No button handler for: TRADFRI SHORTCUT Button endpoint: 0x01 cluster: 0x0003 command: 0x01 payload[0]: 000
```

## Steps to reproduce the behavior
1. Start the pairing process by either adding as IKEA Remote/Dimmer or Other
2. Press the pair button 4 times in 5 seconds
3. Wait
4. Logs say there is no handler for the button endpoint

## Expected behavior
The button should be paired as support was added in 2.9.0

## Screenshots
Well, I don't think any are relevant, but I better write something here and not delete this header.

## Environment
 - Host system: PC
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.09.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system?: No
 
## deCONZ Logs
```
10:27:31:623 sync() in 24 ms
10:27:31:638 Current channel 15
10:27:31:642 CTRL got nwk update id 0
10:27:31:649 Device TTL 3475 s flags: 0x7
10:27:32:658 ZCL attribute report 0xBC33ACFFFE5F0088 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000
10:27:35:597 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
10:27:36:622 GW firmware version: 0x26660700
10:27:36:623 GW firmware version is up to date: 0x26660700
10:27:40:598 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
10:27:45:598 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
10:27:46:616 send permit join, duration: 59
10:27:46:617 send GP proxy commissioning mode
10:27:50:598 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
10:27:54:697 ZCL attribute report 0xBC33ACFFFE9B33D9 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000
10:27:54:702 ZCL attribute report 0xBC33ACFFFE9B33D9 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000
10:27:55:598 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
10:28:00:362 0xEA33 nwk changed to 0x5B00
10:28:00:363 DeviceAnnce of SensorNode: 0x680AE2FFFE25E078 [1]
10:28:00:364 	nwk address changed 0xEA33 -> 0x5B00 [2]
10:28:00:365 device announce 0x680AE2FFFE25E078 (0x5B00) mac capabilities 0x80
10:28:00:365 set fast probe address to 0x680AE2FFFE25E078 (0x5B00)
10:28:00:366 FP indication 0x0000 / 0x0013 (0x680AE2FFFE25E078 / 0x5B00)
10:28:00:366                       ...     (0x680AE2FFFE25E078 / 0x5B00)
10:28:00:367 device announce 0x680AE2FFFE25E078 (0x5B00) mac capabilities 0x80
10:28:00:376 discard double entry in binding queue (size: 2) for for 0x680AE2FFFE25E078, cluster 0x0006
10:28:01:297 [1] get node descriptor for 0x680ae2fffe25e078
10:28:01:492 Bind response success for 0x680ae2fffe25e078 ep: 0x01 cluster: 0x0006
10:28:01:520 FP indication 0x0000 / 0x8002 (0x680AE2FFFE25E078 / 0x5B00)
10:28:01:521                       ...     (0x680AE2FFFE25E078 / 0x5B00)
10:28:01:521 ZDP indication search sensors 0x680AE2FFFE25E078 (0x5B00) cluster 0x8002
10:28:01:522 ZDP indication search sensors 0x680AE2FFFE25E078 (0x5B00) clear timeout on cluster 0x8002
10:28:01:529 [2] get active endpoints for 0x680ae2fffe25e078
10:28:02:590 Bind response success for 0x680ae2fffe25e078 ep: 0x01 cluster: 0x0001
10:28:02:608 FP indication 0x0000 / 0x8005 (0x680AE2FFFE25E078 / 0x5B00)
10:28:02:609                       ...     (0x680AE2FFFE25E078 / 0x5B00)
10:28:02:609 ZDP indication search sensors 0x680AE2FFFE25E078 (0x5B00) cluster 0x8005
10:28:02:610 ZDP indication search sensors 0x680AE2FFFE25E078 (0x5B00) clear timeout on cluster 0x8005
10:28:02:697 FP indication 0x0000 / 0x8005 (0x680AE2FFFE25E078 / 0x5B00)
10:28:02:698                       ...     (0x680AE2FFFE25E078 / 0x5B00)
10:28:02:699 ZDP indication search sensors 0x680AE2FFFE25E078 (0x5B00) cluster 0x8005
10:28:02:703 Bind response success for 0x680ae2fffe25e078 ep: 0x01 cluster: 0x0001
10:28:03:690 FP indication 0x0000 / 0x8005 (0x680AE2FFFE25E078 / 0x5B00)
10:28:03:691                       ...     (0x680AE2FFFE25E078 / 0x5B00)
10:28:03:691 ZDP indication search sensors 0x680AE2FFFE25E078 (0x5B00) cluster 0x8005
10:28:03:696 FP indication 0x0104 / 0x0001 (0x680AE2FFFE25E078 / 0x5B00)
10:28:03:697                       ...     (0x680AE2FFFE25E078 / 0x5B00)
10:28:03:698 ZCL configure reporting rsp seq: 46 0x680AE2FFFE25E078 for ep: 0x01 cluster: 0x0001 attr: 0x0021 status: 0x00
10:28:03:702 FP indication 0x0104 / 0x0001 (0x680AE2FFFE25E078 / 0x5B00)
10:28:03:702                       ...     (0x680AE2FFFE25E078 / 0x5B00)
10:28:04:630 Bind response success for 0x680ae2fffe25e078 ep: 0x01 cluster: 0x0006
10:28:07:238 FP indication 0x0104 / 0x0003 (0x680AE2FFFE25E078 / 0x5B00)
10:28:07:239                       ...     (0x680AE2FFFE25E078 / 0x5B00)
10:28:07:240 [INFO] - No button handler for: TRADFRI SHORTCUT Button endpoint: 0x01 cluster: 0x0003 command: 0x01 payload[0]: 000
10:28:21:398 ZCL attribute report 0xBC33ACFFFE9B33D9 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000
10:28:26:594 ZCL attribute report 0xBC33ACFFFE9B33D9 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000
10:28:31:611 Current channel 15
10:28:31:615 CTRL got nwk update id 0
10:28:31:619 Device TTL 3415 s flags: 0x7
10:28:39:714 ZCL attribute report 0xBC33ACFFFE9B33D9 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000
10:28:42:687 ZCL attribute report 0xBC33ACFFFEA516BD for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000
10:28:45:597 send permit join, duration: 59
10:28:45:598 send GP proxy commissioning mode
10:28:45:802 FP indication 0x0104 / 0x0019 (0x680AE2FFFE25E078 / 0x5B00)
10:28:45:802                       ...     (0x680AE2FFFE25E078 / 0x5B00)
```

## Additional context
Tried adding this issue as #4306 but the bot didn't like it, so here goes again.

Mimiix, 2021-02-01:
The big question here: IS it working correctly? That button handler is also showing for other devices.

@ebaauw as you have integrated this one, can you provide some pointers here and close if needed?

brujoand, 2021-02-01:
Yeah so this is where it gets weird. In Home Assistant which is using the deconz integration, I find the entity `sensor.tradfri_shortcut_button_battery_level`. But in the the phoscon UI there is nothing added under Switches (or sensors for that matter). So I'm not really sure what's happening.

Mimiix, 2021-02-01:
> Yeah so this is where it gets weird. In Home Assistant which is using the deconz integration, I find the entity `sensor.tradfri_shortcut_button_battery_level`. But in the the phoscon UI there is nothing added under Switches (or sensors for that matter). So I'm not really sure what's happening.

That's normal. Phoscon needs to add support themselves. 

 Best would be to check the deconz_event in HA event listener.

brujoand, 2021-02-01:
Thanks, it looks like this was added recently https://github.com/dresden-elektronik/phoscon-app-beta/issues/297 so I'll dig further in that direction then.

Mimiix, 2021-02-01:
@brujoand IT is usable within HA though, as it produces deconz_event in HA with the button. No need for phoscon support there :)

brujoand, 2021-02-01:
ah true, I just find that the phoscon+deconz gives me "better" control over the IKEA buttons and switches compared to HA so I like keeping it there.

Better meaning stuff like supporting the ikea on/off button with on/off for short press and dim+/dim- for long press. In HA that's either lot's of python or AppDaemon + Custom Integration (which I guess is also a lots of python, hehe).

Mimiix, 2021-02-01:
> ah true, I just find that the phoscon+deconz gives me "better" control over the IKEA buttons and switches compared to HA so I like keeping it there.
>
> Better meaning stuff like supporting the ikea [...]

Fair enough :) Just wait for phoscon to update next release and check it then ^^

brujoand, 2021-04-17:
Hey, so I did some more digging here and set up VNC into deconz to view the actuall devices. The shortcut button is indeed added to deconz, and it does show up in Home-assistant. But while listening for 'deconz_event' in HA nothing shows up when pressing the button. I can't find the device under entities nor devices in HA but I can add it as a trigger to an automation in the UI, but saving that automation fails because it doesn't have a domain nor entity_id set. So it seems like the devices isn't presentet correctly to HA. Is there something I can do in deconz to change this?

SwoopX, 2021-04-17:
This sounds like a purely HA related topic now. If the device is sending the button events via API, then there's nothing we can do and you should seek some advice from HA universe.

brujoand, 2021-04-18:
@SwoopX Sorry, I'm not following you here. The events don't show up in HA as 'deconz_event', all other Tradfri buttons do. How do you draw the conclution that this then is a problem on the HA side? Is there a simple way of verifying that events are being sent from deconz?

closed 2021-05-10 by brujoand

mrclschstr, 2021-06-26:
Since phoscon beta 2.12.01 it finally works. Just give it a shot!