dresden-elektronik/deconz-rest-plugin #4309: 2Gang Wall Swicth TZ3000_owgcnkrh / TS0042

degrammont, 2021-01-30:
## Device

- Product name: Lonsonho 2 Gang Wall Switch
- Manufacturer: _TZ3000_owgcnkrh
- Model identifier: TS0042
- Device type : Switch

https://fr.aliexpress.com/item/4001095272467.html?spm=a2g0s.9042311.0.0.1e886c37LAyKkC

The device is visible as a light / the 2 Gangs are visible
Battery level (CR2032) is not reported
deconz_event not reported under HA (I'm using deconZ docker under Synology NAS)

## Screenshots

[image]

[image]

Mimiix, 2021-01-30:
Please add all screenshots.

Smanar, 2021-01-30:
Yep, miss the basic attribute panel, select the "basic cluster" 0x0000, then press the "read" button with making the device awake.
For information, this device will be easy to add, but I m waiting for other PR  will be merged first.

Smanar, 2021-01-30:
Yep, miss the basic attribute table.

degrammont, 2021-01-30:
I hope that's what you are looking for.

[image]
[image]

Thanks for your help.

Mimiix, 2021-01-30:
We need all clusters.

degrammont, 2021-01-30:
[image]
[image]
[image]

And for the second Gang
[image]
[image]
[image]

Smanar, 2021-01-31:
Hello, do you have a linux machine to test the code ? I will explain how to do.

degrammont, 2021-01-31:
I'm using the docker container on my Synology NAS.

Smanar, 2021-01-31:
Lol, ok so hard to test, the pr is submitted, hopping for no error from me.

closed 2021-02-15 by manup