dresden-elektronik/deconz-rest-plugin #4310: Device name Tuya AM43-0.45-40  Blind drive motor

Wimaware, 2021-01-30:
## Device

- Product name: AM43 Blinds drive
- Manufacturer: _TZE200_zah67ekd
- Model identifier: TS0601
- Device type : _Blinds motor drive
- 
[image]
[image]
[image]
[image]
[image]

## Screenshots

Smanar, 2021-01-30:
Hello, do you have a Linux machine to test the code ?

Wimaware, 2021-01-30:
I am running on a Windows machine

Smanar, 2021-01-30:
No problem, will be in all OS on the future version.

closed 2021-02-15 by manup