dresden-elektronik/deconz-rest-plugin #4312: Deconz migration with Docker

marithpl, 2021-01-30:
## Describe the question or issue you are having

I've made a migration of docker. I copy settings of images and copy files of Deconz.
It's seems to work, but in Deconz UI I have some unrecognized devices like 0x2B53 etc.
Is it possible to force update devices names? As I see in Phoscone I'm can only see devices with name in Deconz.

## Screenshots

[image]

## Environment

Host system: NAS Synology
Running method: Marthoc Docker container
Firmware version: 26660700
deCONZ version: 2.09.00 / 23.12.2020
Device: ConBee II
Do you use an USB extension cable: yes
Is there any other USB or serial devices connected to the host system? If so: -----

## deCONZ Logs

```
16:48:35:523 CTRL got nwk update id 0                                                                                                      
16:48:35:527 Device TTL 2841 s flags: 0x7                                                                                                  
16:48:41:710 GW firmware version: 0x26660700                                                                                               
16:48:41:711 GW firmware version is up to date: 0x26660700                                                                                 
16:49:35:518 Current channel 15                                                                                                            
16:49:35:522 CTRL got nwk update id 0                                                                                                      
16:49:35:526 Device TTL 2781 s flags: 0x7                                                                                                  
16:49:41:701 ZCL attribute report 0x0017880104B4FD57 for cluster: 0x0400, ep: 0x02, frame control: 0x08, mfcode: 0x0000                    
16:49:42:816 ZCL attribute report 0x847127FFFE24ECB0 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000                    
16:49:46:595 ZCL attribute report 0x0017880104B4FD57 for cluster: 0x0406, ep: 0x02, frame control: 0x08, mfcode: 0x0000                    
16:50:07:616 DeviceAnnce of SensorNode: 0x001788010867A59E [1]                                                                             
16:50:07:617 DeviceAnnce of SensorNode: 0x001788010867A59E [1]                                                                             
16:50:07:618 DeviceAnnce of SensorNode: 0x001788010867A59E [1]                                                                             
16:50:07:671 [INFO] - No button map for: SML001 endpoint: 0x02 cluster: 0x0402 command: 0x0A payload[0]: 000                               
16:50:07:671 ZCL attribute report 0x001788010867A59E for cluster: 0x0402, ep: 0x02, frame control: 0x08, mfcode: 0x0000                    
16:50:07:810 DeviceAnnce of SensorNode: 0x001788010867A59E [1]                                                                             
16:50:07:811 DeviceAnnce of SensorNode: 0x001788010867A59E [1]                                                                             
16:50:07:811 DeviceAnnce of SensorNode: 0x001788010867A59E [1]                                                                             
16:50:08:035 remove discovery request duplicate                                                                                            
16:50:09:682 ZCL attribute report 0x847127FFFE24ECB0 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000                    
16:50:15:342 APS-DATA.confirm id: 228 status: transaction expired (8 seconds)                                                              
16:50:35:520 Current channel 15                                                                                                            
16:50:35:524 CTRL got nwk update id 0                                                                                                      
16:50:35:528 Device TTL 2721 s flags: 0x7                                                                                                  
16:51:35:519 Current channel 15                                                                                                            
16:51:35:523 CTRL got nwk update id 0                                                                                                      
16:51:35:527 Device TTL 2661 s flags: 0x7                                                                                                  
^[c16:52:16:801 ZCL attribute report 0xCCCCCCFFFE5C4255 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000                 
16:52:24:390 new node - ext: 0x00158d0004621431, nwk: 0x47EE
```

## Additional context

Mimiix, 2021-02-01:
Those devices seem not to be paired correctly. Could it be that there's a version difference between the migration?

Either way, re-pairing should fix it.

marithpl, 2021-02-01:
It's what I did. Versions was the same.
But is very annoying some sensor I have behind wash machine etc. I use Deconz for 1 month and I had to re-pair devices about 3-4 times.

I don't know why sometimes end devices connect to Conbee which is on 2nd floor but they won't connect to device which is 1-2 meter away.

Now I turned off Xiaomi Gas Sensor because I suppose it makes a lot of problems in mesh network.

Mimiix, 2021-02-01:
> I don't know why sometimes end devices connect to Conbee which is on 2nd floor but they won't connect to device which is 1-2 meter away. 

No clue either, but that is what they do. Xiaomi devices tend to not switch parents. Would be interesting to see some logs.

closed 2021-03-04 by github-actions[bot]