dresden-elektronik/deconz-rest-plugin #4313: Level change stop

perfrankling, 2021-01-30:
## Feature request type
Extend REST-API capabilities

## Description
Implement a way to stop an ongoing brightness level change through the REST-API, leaving the brightness level at the current level, ie the same as executing the "Stop" command in the Cluster info panel for the Level Control Cluster in the deConz application.

## Considered alternatives

## Additional context

ebaauw, 2021-01-30:
PUT `{"bri_inc": 0}` to the light state.

perfrankling, 2021-01-30:
Thank you!

The bri_inc parameter is missing from the documentation at https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/lights/

ebaauw, 2021-01-30:
I'm afraid that's not the only thing missing from the API documentation...

It works for groups as well.

Mimiix, 2021-02-01:
Case close :)

closed 2021-02-01 by Mimiix