dresden-elektronik/deconz-rest-plugin #4321: Woox R7051 Smart Indoor Siren

EarlOfBurton, 2021-02-01:
## Device

- Product name: R7051 Smart Indoor Siren
- Manufacturer: _TYZB01_ynsiasng
- Model identifier: TS0219
- Device type : Please remove all unrelated device types. 
  - Siren

Product page see: https://wooxhome.com/uncategorised-c5/new-2020-product-c7/woox-r7051-smart-indoor-siren-p49
## Screenshots

  Required screenshots:
  - Endpoints and clusters of the node
  - Node Info panel
  - Basic Cluster attributes in the Cluster Info panel.
[image]
[image]
[image]
[image]

Mimiix, 2021-02-01:
Missing screenshot of total clusters.

EarlOfBurton, 2021-02-01:
Is this the right missing information?
[image]

Mimiix, 2021-02-01:
Missing IasZONE , Identify, Scenes, Groups and the unknown one too.

EarlOfBurton, 2021-02-01:
Hi @Mimiix, now I think I do understand, sending you all of the screenshots. If I missed something, please let me know.
Thanks.
## Basic
[image]
## Power configuration
[image]
## Identify
[image]
## Groups (1/2)
[image]
## Groups (2/2)
[image]
## Scenes (1/3)
[image]
## Scenes (2/3)
[image]
## Scenes (3/3)
[image]
## IAS Zone
[image]
## IAS WD
[image]
## Unknown
[image]
## Identify
[image]
## OTAU
[image]

Smanar, 2021-02-01:
Loool, seriously ? a new tuya cluster ?

EarlOfBurton, 2021-02-01:
> Loool, seriously ? a new tuya cluster ?

Hi @Smanar, I don't get the joke... Do they make so many/too many clusters? I'm fairly new to the "deep dive" into Zigbee...

Smanar, 2021-02-01:
Lol, no, don't worry, it s just 3/4 months we are on a special cluster for tuya, with all to discover, so much work done, and now we know it better you find a new one.

EarlOfBurton, 2021-02-02:
Hello friends, are you still missing data from my side, since it has still the mark "incomplete issue"? If so, could you let me know what it is, please?

Mimiix, 2021-02-03:
@EarlOfBurton Sorry, forgot to update. Busy day.

EarlOfBurton, 2021-02-25:
Bump to keep it alive.

Smanar, 2021-10-30:
You speak about the DDF file ?
Yes, no one know the command used by this device (or at least I haven't find it in google).
But like I have said, this device use a new cluster, so we probably need to write a code part just for it, and atm the priority is more cleaning the actual code than adding more line code.

kkoenders, 2021-11-22:
When you are going to code new clusters, doe I need to reopen this ticket? Or are you going to track all stale (and closed) tickets requesting integration?

Smanar, 2021-11-22:
Hu, honestly  if I don't need them, I will not re open closed ticket, there is so much of them.
But you can easily re open them yourself.

The priority will be the migration of old code to the ddf core, so first the already known cluster.

closed 2021-12-21 by github-actions[bot]