dresden-elektronik/deconz-rest-plugin #4323: Woox R7052 Smart SOS Button

EarlOfBurton, 2021-02-01:
## Device

- Product name: Woox R7052 Smart SOS Button
- Manufacturer: _TZ_3000_ssp0maqm
- Model identifier: TS0215
- Device type : Please remove all unrelated device types. 
  - Switch

Product page: https://wooxhome.com/new-c8/woox-r7052-smart-sos-button-p50
## Screenshots
[image]
[image]
[image]

Mimiix, 2021-02-01:
Missing screen of the device on deconz (and what clusters it houses).

EarlOfBurton, 2021-02-01:
Is this the screenshot I forgot to do?
[image]

Mimiix, 2021-02-01:
Yes. But now also missing Time, IAS zone and IAS ACE.

EarlOfBurton, 2021-02-01:
Okay, I'm sorry, I'm new to the process. Is this what you need?
## Basic
[image]
## Time
[image]
## Power Configuration
[image]
## IAS Zone
[image]
## IAS ACE (part 1 of 2)
[image]
## IAS ACE (part 2 of 2)
[image]
## OTAU
[image]

Mimiix, 2021-02-02:
Yes. Well, not me, but the developers.

EarlOfBurton, 2021-02-02:
Hello friends, are you still missing data from my side, since it has still the mark "incomplete issue"? If so, could you let me know what it is, please?

EarlOfBurton, 2021-02-24:
Bump to keep it active

Mimiix, 2022-07-26:
Re opened on user request.

Nomeboj, 2022-08-03:
Bump to keep it active

Mimiix, 2022-10-12:
@Smanar Can this be done you think?

Smanar, 2022-10-12:
Yep, can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_ssp0maqm",
  "modelid": "TS0215",
  "vendor": "Woox",
  "product": "R7052 Smart SOS Button",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0401",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0500"
        ],
        "out": [
          "0x0501"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent",
          "awake": true
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0501"
    }
  ]
}
```

With this DDF the device will be reconized but not working.
Pls check deconz log with flag "info" and "info_l2" and pressing button, you will see error about missing button map, can you share them ?

But I think you will have thoses one

```
        "immaxKeyfobMap": {
            "vendor": "Immax",
            "doc": "Smart Keyfob 07046L",
            "modelids": ["Keyfob-ZB3.0"],
            "map": [
                [1, "0x01", "IAS_ACE", "ARM", "1", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm day/home zones only"],
                [1, "0x01", "IAS_ACE", "ARM", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Disarm"],
                [1, "0x01", "IAS_ACE", "EMERGENCY", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Emergency"],
                [1, "0x01", "IAS_ACE", "ARM", "3", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm all zones"]
            ]
        },
```

Smanar, 2022-10-12:
My DDF is based on the one on the first post, and this one is not a tuya.

EarlOfBurton, 2022-10-12:
Hi friends,
thanks for providing the start of the DDF.
The Log says:
``19:42:11:912 [INFO] - No button map for: TS0215A, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: EMERGENCY (0x02), payload: None, zclSeq: 110
19:42:11:942 Websocket 172.30.32.1:37604 send message: {"attr":{"id":"103","lastannounced":null,"lastseen":"2022-10-12T17:42Z","manufacturername":"_TZ3000_ssp0maqm","modelid":"TS0215A","name":"Switch 103","swversion":null,"type":"ZHASwitch","uniqueid":"bc:33:ac:ff:fe:29:6b:2b-01-0500"},"e":"changed","id":"103","r":"sensors","t":"event","uniqueid":"bc:33:ac:ff:fe:29:6b:2b-01-0500"} (ret = 328)``

Is this what you need?
Since I'm having 73 nodes it's pretty hard to get the right portion of the log to me...

There is a small typo in the DDF,
you wrote ``"modelid": "TS0215"``
but it should be ``"modelid": "TS0215A"``

Couldn't get it working without the "A" at the end.

Smanar, 2022-10-13:
Ha thx, have correctyed the DDF

>Is this what you need?

Yes exactly and this one is the same than 

`No button map for: TS0215A, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: EMERGENCY (0x02), payload: None,`

`[1, "0x01", "IAS_ACE", "EMERGENCY", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Emergency"],`

So we can suppose all are same.
You can edit the file "button_maps.json" you can find it in the "devices" folder too

And add inside

```
        "TuyaKeyfobMap": {
            "vendor": "Woox",
            "modelids": ["TS0215A"],
            "map": [
                [1, "0x01", "IAS_ACE", "ARM", "1", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm day/home zones only"],
                [1, "0x01", "IAS_ACE", "ARM", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Disarm"],
                [1, "0x01", "IAS_ACE", "EMERGENCY", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Emergency"],
                [1, "0x01", "IAS_ACE", "ARM", "3", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm all zones"]
            ]
        },
```

EarlOfBurton, 2022-10-13:
Hi and thanks,
I implemented the part to button_maps.json
then I had to restart deCONZ the changes to take effect.
Now I can see an action connected to the device in Home Assistant (before there was nothing):

[image]
but when I use it in an Automation, **nothing happens**.
I tried it with two more automations to be sure there's nothing wrong with that - same result. Nothing happens.
I pressed it, pressed it long, pressed it twice, pressed it really long like 10 secsonds - the automation was never triggered.

I relogged with INFO and INFO_L2, maybe this can help you to identify the problem?
`22:43:07:138 Websocket 172.30.32.1:37758 send message: {"attr":{"id":"105","lastannounced":null,"lastseen":"2022-10-13T20:43Z","manufacturername":"_TZ3000_ssp0maqm","modelid":"TS0215A","name":"Woox R7052 Smart SOS Button 1","swversion":null,"type":"ZHASwitch","uniqueid":"bc:33:ac:ff:xx:xx:xx:xx-xx-xxxx"},"e":"changed","id":"105","r":"sensors","t":"event","uniqueid":"bc:33:ac:ff:xx:xx:xx:xx-xx-xxxx"} (ret = 347)`

The API Information from Phoscon states as follows (I edited the etag and uniqueid field for safety reasons):
`{
    "config": {
        "enrolled": 0,
        "on": true,
        "pending": [],
        "reachable": true
    },
    "ep": 1,
    "etag": "ff5bfac77c156167f0a6f67xxxxxxxxx",
    "lastannounced": null,
    "lastseen": "2022-10-13T20:53Z",
    "manufacturername": "_TZ3000_ssp0maqm",
    "mode": 1,
    "modelid": "TS0215A",
    "name": "Woox R7052 Smart SOS Button 1",
    "state": {
        "buttonevent": null,
        "lastupdated": "none",
        "lowbattery": null,
        "tampered": false
    },
    "type": "ZHASwitch",
    "uniqueid": "bc:33:ac:ff:xx:xx:xx:xx-xx-xxxx"
}`

Or do you need something else I can support you with?
Kind regards.

Smanar, 2022-10-14:
>"buttonevent": null

I have something wrong.
You still have same error message in deconz when using a button ?

>No button map for: TS0215A, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: EMERGENCY (0x02), payload: None

EarlOfBurton, 2022-10-14:
Hi @Smanar,
I think there was no message like "No button map".
Btw. the "Woox R7052 Smart SOS Button" has only one button, in case that is important for the button map.

Now there is something weird happening.
I deleted the node via deCONZ, to have a new clean pairing.
I reset the Woox device.
Then I tried to re-pair it with Phoscon, as I did before. But Phoscon will no more recognise it, but in deCONZ I see it with the randomly generated name (0x48BD):
[image]

I restarted the deCONZ addon, no luck.
I shutdown the Raspi, took it off power for 1 min, rebooted it.
Still the same, Phoscon is not recognising the device again, although I can see in deCONZ that it has loaded the correct DDF file...

Can you help me?

Kind regards.

EarlOfBurton, 2022-10-14:
@Smanar, I got it working again...
it seems that the DDF-file that I created, `woox_r7052_ts0215a.json` got deleted as well as the new `button_maps.json` was overwritten during restart or reboot of deCONZ and the changes are gone. I had to reupload the files in the usr/share/deCONZ/devices/woox -folder.
Strange.
At the moment, with the by deCONZ cleaned button_maps.json I get the following when pressing the button:
`19:16:03:547 [INFO] - No button map for: TS0215A, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: EMERGENCY (0x02), payload: None, zclSeq: 42`

Does this help to get it working?

Smanar, 2022-10-15:
Yes :(
Ha use docker system, lot of stuff are not permanent.

>Btw. the "Woox R7052 Smart SOS Button" has only one button, in case that is important for the button map.

Only one button ?
So yes, no need to use the button map, can make something faster

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_ssp0maqm",
  "modelid": "TS0215A",
  "vendor": "Woox",
  "product": "R7052 Smart SOS Button",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0401",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0500",
          "0x0501"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "cl": "0x0501",
            "cmd": "0x02",
            "eval": "Item.val = 1002"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0501"
    }
  ]
}
```

EarlOfBurton, 2022-10-15:
Hi @Smanar,
i replaced the device-file with your definition, re-paired, now I get
`21:15:21:120 [INFO] - No button map for: TS0215A, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: EMERGENCY (0x02), payload: None, zclSeq: 17`

Btw. now it is recognised as "Remote Control" and before it was recognised as "Switch" - but both are ok to me.

Smanar, 2022-10-15:
>now it is recognised as "Remote Control" and before it was recognised as "Switch"

I don't see what happen, the device type is a ZHAswitch

You can have the error in logs, but now the device don't use the buttonmap, it work direclty with the api
If I m right every time you use the button you receive a notification with state/buttonevent = 1002 in websocket (it s visible too on phoscon/help/API Information/event/sensor/ ..)

EarlOfBurton, 2022-10-15:
Hi @Smanar,
true.
In Phoscon it looks like this:
[image]

and the websocket, as you describe is:

[image]

But in the automations in Home assistant I cannot select "button press" as trigger.
I do have the device woox_r7052_sos_button1 but no trigger as I have it from other switches.
What am I doing wrong?

Smanar, 2022-10-15:
Not your fault ^^

It seem there is still missing the state/buttonevent 

Are you sure HA haven't replaced your DDF ?
You need to have 

```
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "cl": "0x0501",
            "cmd": "0x02",
            "eval": "Item.val = 1002"
          }
        },
```

EarlOfBurton, 2022-10-15:
True, my mistake. Maybe I uploaded the wrong ddf.
Now with state/buttonevent existing in DDF (I double checked). Un-paired. Deleted old DDF, uploaded new DDF, re-paired:
[image]

[image]

Still no "button press" as trigger available.

EarlOfBurton, 2022-10-15:
How do I know that deCONZ is using the recently uploaded `woox_r7052_ts0215a.json` and not an old cached one?
At the moment, when I `cat` the file, I get:
`root@core-deconz:/usr/share/deCONZ/devices/woox# cat woox_r7052_ts0215a.json 
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_ssp0maqm",
  "modelid": "TS0215A",
  "vendor": "Woox",
  "product": "R7052 Smart SOS Button",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0401",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0500",
          "0x0501"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "cl": "0x0501",
            "cmd": "0x02",
            "eval": "Item.val = 1002"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0501"
    }
  ]
}
`

Smanar, 2022-10-15:
>Now with state/buttonevent existing in DDF (I double checked). Un-paired. Deleted old DDF, uploaded new DDF, re-paired:

Can just use the "hot reload" command in the DDF editor.
- right clic on the device
- "edit DDF"
- look on preview is the DDf is good, with all code in state/buttonevent
- edit what you want
- make "hot relaod"

Bind/report are already set, so no need to reconfigure the device.

But the state/buttonevent is still "null", you can see nothing in logs with the flag" DDF" ?

I don't see what I m making bad, perhaps with more information 

```
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "cl": "0x0501",
            "cmd": "0x02",
           "fn": "zcl",
           "ep": 1,
            "eval": "Item.val = 1002"
          }
        },
```

EarlOfBurton, 2022-10-15:
It seems like it is not accepting anything below "awake": true
It is ignoring the rest, see screenshot below.
is it possible that there is some bad indentation or some typo?
[image]

EarlOfBurton, 2022-10-15:
@Smanar should it maybe be `"zcl"` instead of `"cl"` below the `"parse"`-line??

Smanar, 2022-10-16:
I realy hate the HA OS ....

I have just checked the json, all is fine
I think the editor don't use the same you are looking in the console.

It's possible too making the sate/buttonevent chnage using the editor, but you need in the preview the fingerprint part 

```
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0401",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0500",
          "0x0501"
        ]
      },
```

If you can see this part in the "preview", select state/buttonevent in the editor and you can fill all fields manually.

Nomeboj, 2022-10-18:
Hi.

I created same thing for my SOS button. I changed  "manufacturername": "_TZ3000_zsh6uat3"

So DDF looks like:

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_zsh6uat3",
  "modelid": "TS0215A",
  "vendor": "Woox",
  "product": "R7052 Smart SOS Button",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/woox/woox_r7052_ts0215a.json",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0401",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0500",
          "0x0501"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "cl": "0x0501",
            "cmd": "0x02",
            "ep": 1,
            "eval": "Item.val = 1002",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0501"
    }
  ]
}

Added this to button_maps.json:

"TuyaKeyfobMap": {
            "vendor": "Woox",
            "modelids": ["TS0215A"],
            "map": [
                [1, "0x01", "IAS_ACE", "ARM", "1", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm day/home zones only"],
                [1, "0x01", "IAS_ACE", "ARM", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Disarm"],
                [1, "0x01", "IAS_ACE", "EMERGENCY", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Emergency"],
                [1, "0x01", "IAS_ACE", "ARM", "3", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm all zones"]
            ]
        },

restarted everything.
In Home assistant i can see remote it is not switch but no entities. However in HA logbook i can see info when button is presed:
**Remote control fired event '1002'**

In log i can see :

No button map for: TS0215A, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: EMERGENCY (0x02), payload: None, zclSeq: 179

And this is strange if button map is updated. 
Are you sure that mapping is correct for EMERGENCY button:

[1, "0x01", "IAS_ACE", "EMERGENCY", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Emergency"], 

Regards

Nomeboj, 2022-10-18:
Hi.

Tried with 1234. Re added switch, i can see in DDF that is updated still no entity in HA.
Logs shows:

19:25:02:546 [INFO] - No button map for: TS0215A, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: EMERGENCY (0x02), payload: None, zclSeq: 202
19:25:02:546 Websocket 192.168.0.58:34976 send message: {"e":"changed","id":"10","r":"sensors","state":{"buttonevent":1234,"lastupdated":"2022-10-18T17:25:02.545","lowbattery":null},"t":"event","uniqueid":"a4:c1:38:11:30:61:52:3b-01-0500"} (ret = 183)

Smanar, 2022-10-18:
For me it mean it work.
The buttonevent  = 1002 mean button 1, relased button.

For the fact it's not visible in HA perhaps this link can help you https://www.home-assistant.io/integrations/deconz/#remote-control-devices

Nomeboj, 2022-10-18:
Ok, so this means that we will not have entity but we need to create automation based on event trigger?

Smanar, 2022-10-18:
Honnestly IDK how work HA, but I m seing this question regulary, on the last, one user have post capture but I don't find again the issue 

BTW, we forget the battery level

Miss state/battery and bind

```
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0001",
         "report":[
            {
               "at":"0x0021",
               "dt":"0x20",
               "min":600,
               "max":6000,
               "change":"0x00000001"
            }
         ]
      }
```

EarlOfBurton, 2022-10-18:
Hi @Smanar, thank you for your help, this is not an easy journey ;)

I got the DDF now replaced, finally I had to rename the file so that HA OS reads the new definition. It seems to use a cached file from ... somewhere?
I can see the battery indicator in deCONZ.
But in Home Assistant I can't see either the battery state nor I can use the button event as a trigger.
With my simple minded brain it seems Home Assistant does not know that there is button or the battery state, still receiving events but not knowing that this is a button press?
[image]

Comparing it to i.e. an Ikea 2-button switch:
[image]

My DDF is now:
```
{
	"schema": "devcap1.schema.json",
	"manufacturername": "_TZ3000_ssp0maqm",
	"modelid": "TS0215A",
	"vendor": "Woox",
	"product": "R7052 Smart SOS Button",
	"sleeper": true,
	"status": "Gold",
	"path": "/devices/woox/woox_r7052_ts0215a.json",
	"subdevices": [
	{
		"type": "$TYPE_SWITCH",
		"restapi": "/sensors",
		"uuid": [
			"$address.ext",
			"0x01",
			"0x0500"
		],
		"fingerprint": {
			"profile": "0x0104",
			"device": "0x0401",
			"endpoint": "0x01",
				"in": [
					"0x0000",
					"0x0500",
					"0x0501"
				]
		},
		"items": [
			{ "name": "attr/id" },
			{ "name": "attr/lastannounced" },
			{ "name": "attr/lastseen" },
			{ "name": "attr/manufacturername" },
			{ "name": "attr/modelid" },
			{ "name": "attr/name" },
			{ "name": "attr/swversion" },
			{ "name": "attr/type" },
			{ "name": "attr/uniqueid" },
			{
				"name": "config/enrolled",
				"public": false
			},
			{ "name": "config/on" },
			{ "name": "config/pending" },
			{ "name": "config/battery" },
			{ "name": "config/reachable" },
			{
				"name": "state/buttonevent",
				"awake": true,
					"parse": {
					"cl": "0x0501",
					"cmd": "0x02",
					"ep": 1,
					"eval": "Item.val = 1002",
					"fn": "zcl"
					}
			},
			{ "name": "state/lastupdated" },
			{ "name": "state/lowbattery" }
		]
	}
	],
	"bindings": [
		{
			"bind": "unicast",
			"src.ep": 1,
			"cl": "0x0500"
		},
		{
			"bind": "unicast",
			"src.ep": 1,
			"cl": "0x0501"
		},
		{
			"bind":"unicast",
			"src.ep":1,
			"cl":"0x0001",
			"report":[
				{
					"at":"0x0021",
					"dt":"0x20",
					"min":600,
					"max":6000,
					"change":"0x00000001"
		   		}
			]
		}
	]
}
```

The debug says:
```
22:51:26:021 Websocket 172.30.32.1:41750 send message: {"e":"changed","id":"106","r":"sensors","state":{"buttonevent":
1002,"lastupdated":"2022-10-18T20:51:26.012","lowbattery":null},"t":"event","uniqueid":"bc:33:ac:ff:fe:xx:xx:xx-xx-xxxx"} (ret = 184)
```

EarlOfBurton, 2022-10-19:
Hi @Smanar,
this is interesting, the battery field says "0" (screenshot below). Is there something wrong with the DDF I use? ( [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4323#issuecomment-1282993416](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4323#issuecomment-1282993416) )

[image]

Smanar, 2022-10-19:
Your DDF seem fine, the binding is good for me.
Can try with a more brutal version with replacing
```

        {
          "name": "config/battery",
         }
```

by

```

        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 3600,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
             "eval": "Item.val = Attr.val / 2",
            "fn": "zcl"
          },
          "default": 0
        },
```
but remember you need some time for battery report.

EarlOfBurton, 2022-10-20:
@Smanar, a battery value seems to have arrived - it took ~ 1,5 days ;) - Can it be that the percent calculation is not correct?
[image]

Unfortunately there still is no trigger to select in HA as it is with other deCONZ-bound switches. Any idea how to fix this?

Kind regards.

Mimiix, 2022-10-20:
> @Smanar, a battery value seems to have arrived - it took ~ 1,5 days ;) - Can it be that the percent calculation is not correct? [image]
> 
> This is what deCONZ says about the battery: [image]
> 
> Unfortunately there still is no trigger to select in HA as it is with other deCONZ-bound switches. Any idea how to fix this?
> 
> Kind regards.

Do the events come through?

Mimiix, 2022-10-20:
In that case , you need to ask @kane610 to add support for it on home assistant. You can already use the event though by making an automation that triggers on the event.

EarlOfBurton, 2022-10-20:
Hi  @Kane610 - would it be possible to add this device so that the button trigger can be used an on any other switch in automations in home assistant?
Kind regards.

Mimiix, 2022-10-20:
> Hi @Kane610 - would it be possible to add this device so that the button trigger can be used an on any other switch in automations in home assistant? Kind regards.

You can already so this by using a event as a trigger in ha.

Nomeboj, 2022-10-21:
> In that case , you need to ask @Kane610 to add support for it on home assistant. You can already use the event though by making an automation that triggers on the event.

Do we need to open new request or we can ask on discord maybe?
I tried with deconz event in HA to trigger and it is working fine, but maybe it would be better to have entity :-)

Regards

Smanar, 2022-10-21:
> Can it be that the percent calculation is not correct?

I m reading the value (so 200 on your capture) and dividing it by 2, so you can't have 0, there is something wrong
You are using this modification ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4323#issuecomment-1284352270

On the other woox device we are using for report

```
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0001",
         "report":[
            {
               "at":"0x0021",
               "dt":"0x20",
               "min":300,
               "max":43200,
               "change":"0x00000001"
            }
         ]
      },
```

Kane610, 2022-10-21:
> Do we need to open new request or we can ask on discord maybe? I tried with deconz event in HA to trigger and it is working fine, but maybe it would be better to have entity :-)
>
> Regards

Sorry this will not be possible, the events are momentary interactions not suitable for being exposed as an entity. You can probably create a template entity to achieve what you want.

EarlOfBurton, 2022-10-23:
Initially I thought that DDF was for enthusiast users with some coding skills so that there is a lot of work off the deCONZ/Phoscon developers. But getting a simple device with one button not working in 60 comments, well, is very disencouriging.
If dresden electronics wants us end-users to contribute, we needed a rather extensive explanation on one device.
I know that one webpage about DDF, this is not enough. I think we needed a maybe ~ 1 hour video that goes through all the parameters, which formats they can have and how to save and load successfully a DDF. From "unknown" device to "fully working and implemented".
Maybe DDF was meant that way and still is in Kindergarden - I don't know. Maybe someone from the developers could tell is the idea how/who DDF should be used.
Still, I appreciate a lot your stamina supporting us enthusiast-users that extensive. It's a hacking project more or less, how I see it today. Since you often don't get a direct access to the manufacturers but have to reverse-engineer a lot of things, I guess?

Kind regards.

Smanar, 2022-10-23:
>But getting a simple device with one button not working in 60 comments

But it still not working ?

EarlOfBurton, 2022-10-23:
@Smanar, yes, unfortunately it is not working for me.

I tried to go the 'event trigger' route, which would be another hack that I don't like, but since the event trigger options are extremely bad documented in Home Assistant (none of your business, I know) and lacking of examples, I can't get it working.
All my other switches (Xiaomi, IKEA remote, IKEA switch, Philips Hue remote) work as expected with 'button press' triggers.

Kind regards.

Mimiix, 2022-10-23:
@Kane610 Can you provide a sample perhaps?

Smanar, 2022-10-23:
You mean it's possible on HA for an ikea remote but not this switch ?
But they are working same on deconz side, why there is difference on HA ? I don't think it's from the plugin.

EarlOfBurton, 2022-10-23:
True.
I have five IKEA remotes, seven IKEA switches, one Philips hue dimmer/switch, one Xiaomi Switch and the one Woox switch.
All except the Woox switch have "Trigger on buttonpress" options.
I made screenshots for all "events" in Home Assistant under devices in deCONZ:
[image]
[image]
[image]
[image]
[image]

Kind regards

Mimiix, 2022-10-23:
Hi,

But it does provide events right? If that's the case deconz support is finished and correctly. 

Ha is not managed in this repository. @kane610 is the developer of the integration in HA. If he says it isnt possible, it isn't.

EarlOfBurton, 2022-10-23:
No happy end on this thread for a one button device after 1,8 years and ~60 comments.
I liked this button beacuse of its nice SOS label which I wanted to set an alarm to emergency authorities via Internet. 
Getting a DDF working - until the end, which is Home Assistant in this case - has some room for improvement, it seems.

Thanks for trying, though.
Kind regards & have a nice sunday.

Nomeboj, 2022-10-23:
Hi.

I am adding happy end on this then :-)

Add to configuration.yaml

template:
  - trigger:
      platform: event
      event_type: deconz_event
    binary_sensor:
      - name: SOS
        auto_off: 5
        state: "true"

This will create binary sensor SOS with auto off 5 seconds. So when you press SOS button, it will trigger deconz event and binary sensor will change state to on and will go to off after 5 seconds.

Regards.

Kane610, 2022-10-23:
> Hi,
> 
> But it does provide events right? If that's the case deconz support is finished and correctly.
> 
> Ha is not managed in this repository. @Kane610 is the developer of the integration in HA. If he says it isnt possible, it isn't.

Sorry. I said entity is not possible. What's talked about now are device registry support which is a different thing and can be done here

Kane610, 2022-10-23:
> No happy end on this thread for a one button device after 1,8 years and ~60 comments. I liked this button beacuse of its nice SOS label which I wanted to set an alarm to emergency authorities via [...]

Certain aspects of the home assistant integration are specific per device and can thus not be automatically supported. I intend to start looking at the devices endpoint to start automatically generate device triggers for new devices

EarlOfBurton, 2022-10-23:
@Kane610 That's great news.
I finally could implement it in an automation without generating a template binary sensor like that:
```
- id: '1663952665245'
  alias: 'SOS-Button: Toggle Siren'
  trigger:
  - platform: event
    event_type: deconz_event
    event_data:
      id: woox_r7052_smart_sos_button1
      event: 1002
  action:
  - service: siren.turn_on
    entity_id: siren.woox_r7051_sirene1
  - delay:
      hours: 0
      minutes: 0
      seconds: 2
      milliseconds: 0
  - service: siren.turn_off
    entity_id: siren.woox_r7051_sirene1
  mode: single
```
For the future it would be awesome to have automatically generated device triggers...

Kind regards.

Kane610, 2022-10-23:
> @Kane610 That's great news. I finally could implement it in an automation without generating a template binary sensor like that:
>
> ```
> - id: '1663952665245'
>   alias: 'SOS-Button: Toggle Siren' [...]

Yes the event is really all you need. But device triggers simplifies it to the next level

EarlOfBurton, 2022-10-29:
Hi! After updating to deCONZ 2.19.1 the DDF file was lost.
Would it be possible to integrate the file into deCONZ? (See file attached, I had to rename it from .json to .txt because github does not allow .json)

[woox_r7052_ts0215a.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9893813/woox_r7052_ts0215a.txt)

Smanar, 2022-10-29:
Oups, sorry , the PR  was already submitted for me.

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6481

[5 more replies]

closed 2022-11-08 by manup