dresden-elektronik/deconz-rest-plugin #4325: Woox R7060 Smart Garden Irrigation Control Device name

EarlOfBurton, 2021-02-01:
## Device

- Product name: Woox R7060 Smart Garden Irrigation Control
- Manufacturer: _TZ_3210_eymunffl
- Model identifier: TS0101
- Device type : Please remove all unrelated device types. 
  - Other: Garden Watering Switch

Product page: https://wooxhome.com/products-c10/other-c4/woox-r7060-smart-garden-irrigation-control-p61

## Screenshots
[image]
[image]
[image]
[image]

EarlOfBurton, 2021-02-01:
Now adding _all_ of the cards, since I know what is needed (sorry for that).
## Node Info
[image]
## Basic
[image]
## Groups (1/2)
[image]
## Groups (2/2)
[image]
## Scenes (1/3)
[image]
## Scenes (2/3)
[image]
## Scenes (3/3)
[image]
## On/Off (1/2)
[image]
## On/Off (2/2)
[image]
## Tuya specific Cluster
[image]
## Power Configuration
[image]
## OTAU
[image]
## Time
[image]

EarlOfBurton, 2021-02-24:
Bump to keep it active

Martinvdm, 2021-02-27:
Interesting, hoping this device gets supported

Nomeboj, 2021-03-08:
Hi.

Is there any chance that this device will be supported? 
It would be great because it is battery operated, no cables, easy install and I would say cheap product if we compare it to some other vendors :-)

Mimiix, 2021-03-09:
@Smanar This is Tuya.

ippo112, 2021-03-26:
Hi,
any news about the possible integration of R7060 with Deconz?

Mimiix, 2021-03-26:
@Smanar  :)?

dotcom84, 2021-04-07:
+1

Since this seems to be the only real smart irrigation solution that is both affordable and works using Zigbee and thus (in theory) would not require it's own gateway...

dotcom84, 2021-04-07:
Cheap enough that I'd sponsor a set of hardware for the brave dev implementing this device...

ippo112, 2021-04-18:
I integrated in HA with deconz, but I'm not able to see the battery state.
Any suggestions?

Nomeboj, 2021-04-18:
Hi.
Is this working and only battery state is a problem?
What deconz version you are using?

ippo112, 2021-04-18:
yes, it's paired as Smart plug.

Firmware:  26680700
version sw: 2.10.04 / 14/3/2021

Nomeboj, 2021-04-19:
Hi.

Thx it is working for me also. 
Only issue is battery state if there is battery reading at all in this device.

I need to say that this device is really great for battery operated irrigation system, but lets wait couple of months to see how performs :-)

Regards.

htodep, 2021-05-02:
He!

Really curious if somebody has managed to get the battery state of the device.
@Nomeboj Was it easy to get it working in HA? How is it working now?
This is probably a reason for me to ditch the Hue hub.

Kind regards.

Nomeboj, 2021-05-02:
Hi.
It was easy to integrate it, it is working excellent for the past two weeks.

Regards

dotcom84, 2021-05-02:
Unfortunately my experience with the device is different.
It connects just fine and works for 1 or 2 days. After that it is still displayed as "reachable", but does not show any reaction when switching it on and off (even if displayed correctly in Phoscon).

htodep, 2021-05-02:
But it's still reachable? As in it turns on and off? Distance between hub
is oke?

Nomeboj, 2021-05-02:
Hi.

Device is outside and my deconz gateway is inside (10m apart concrete wall). I had same problem that was solved with ikea extender that is near woox r7060.
Try with ikea extender or maybe zigbee bulb that will extend zigbee mesh.

Regards.

dotcom84, 2021-05-02:
Yeah, it's reachable and shown as such. Distance shouldn't be an issue: There is a "repeater" (wired ZigBee device) only about 80cm away.
I can re-pair the device with Phoscon to get it up and running for another day or two any time...

Nomeboj, 2021-05-02:
Hi.

Can you check in VNC is it actually connected via repeater?

ottelo9, 2021-06-01:
Alternative device for Germany (Lidl):
https://www.lidl.de/de/parkside-smarter-bewaesserungscomputer-zigbee-smart-home/p375570
for only 25€

thinker08131, 2021-06-30:
@dotcom84 Did you solve the problem with the valve? My Valve acts the same...few days it works fine, and then it does not respond for remote control. Zigbee2MQTT shows it is available and the LQI is around 125. Is that enought? I am just starting with HA and I am not very familiar with the system ;)

htodep, 2021-06-30:
@dotcom84 I've had the same issue, try to add the device through a router and not directly to the controller. That worked for me. The LQI went from 50 to 237

dotcom84, 2021-06-30:
> @dotcom84 Did you solve the problem with the valve? My Valve acts the same...few days it works fine, and then it does not respond for remote control. Zigbee2MQTT shows it is available and the LQI is [...]

I tried everything with my initial device - e.g. resetting it a couple of times (with and without a repeater placed only 50cm away, etc.). However, no configuration would make it work in a reliable way for more than a day or two.
What finally resolved the issue for me: Replacing it with another R7060 (bought three initially). The replacement has now been working for more than two weeks without any hiccups...

thinker08131, 2021-06-30:
@htodep  It means that the valve needs really good signal to work.. What do you mean through a router? I have HA installed on the docker and ConBee2 plug to USB. I think the only way to increase the signal strength is another device to extend the mesh..

thinker08131, 2021-06-30:
@dotcom84  I bought another R7060 yesterday and I will install it this afternoon. I hope this one will be working better.

htodep, 2021-06-30:
@dotcom84 Sorry, wanted to tag @thinker08131. 
Well it is a battery device so a good signal is probably what you want.
I'm not running Deconz anymore but ZHA because of the native integration in HA. In ZHA you can add a device through a device. So the device is always linked to that particular device. Mine was just a simple light from Innr.

thinker08131, 2021-06-30:
OK, i get it. I did not found that option in HA (to add device through a device), so probably the only way is to install another device very close to the valve. I bought simple light switch and want to use it as "Digital Input" and control the pressure switch after the valve. It is not battery powered and I hope it will increase LQI enought.

Mimiix, 2021-07-28:
Reopened on user request

Smanar, 2021-07-28:
Hello, the device work natively on deconz ? If you use only the on/off cluster ?

EarlOfBurton, 2021-07-28:
Hi,

yes, the on/off works.

Smanar, 2021-07-28:
So you just need the device be visible in the API (and battey level) ?

EarlOfBurton, 2021-07-29:
You see that I asked for a device integration months ago - nothing happened.

I stopped asking for integration of devices because it is lost time (a lot of work for all the screenshots, data, log files, ... they ask for - and nothing happens in the end).
This thread has over 40 entries but still no action regarding integration.

I don't remember if it also receives the battery level.
On/off works, as I wrote.

It registers as weird device . Surely I paired it using Phoscon App, there is no other way using the deCONZ integration, AFAIK.
Then I looked into deCONZ via VNC as well to see if pairing worked, since there is no device support from Phoscon's side.

I'm using Home Assistant on a Raspberry Pi, no docker/container/... - the "classic way" or however you would call it.

Smanar, 2021-07-29:
Sorry, on my side, I have just see this issue some days ago, because of the new forum, have skipped too Mimiix notifications.
But NVM this device realy look easy to integrate.

Ok so first try

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch woox https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2"

RenePhilipse, 2021-08-02:
Hi @Smanar. Sorry, I was also only looking at the new forum post and didn't look here. I'm not very into Linux and testing using command lines as a above, but I will give it a try! I don't have an extra Pi but I will try using a Linix VM. I will keep you posted on any progress.

Question out of curiosity: Shouldn't it be detected as a 'switch' instead of a 'light'? Seems more logical to me.

RenePhilipse, 2021-08-02:
> 
> 
> Sorry, on my side, I have just see this issue some days ago, because of the new forum, have skipped too Mimiix notifications. I don't read all issues.
> But NVM this device realy look easy to integrate.
> 
> So first try
> 
> If I m right at first inclusion you will have a new "sensor", a zhabattery, and after a second one you will have a "light" device for it.

Hi Smanar,
I created a linux VM and followed your instructions (and the info from the link) to install everything. I think everything went fine because I can start Deconz and see my conbeeII stick. However, I can't find/link with any devices, neither via GUI or Phoscon. I also tried to link a Blitwolf outlet but that also doesn't work. Maybe I did something wrong, but do you have any direction where to look? (FYI: I currently use Home Assistant with the Deconz plugin in my daily setup, so that's why I do not have very much experience with the installation part).

Smanar, 2021-08-02:
If you have the GUI I think the problem can be visible on log
Try in deconz / help / debug view, with flag 'info" "info_l2" "error" and "error_l2"

RenePhilipse, 2021-08-02:
> 
> 
> If you have the GUI I think the problem can be visible on log
> Try in deconz / help / debug view, with flag 'info" "info_l2" "error" and "error_l2"

Thanks for your response smanar, I got it fixed already and I am now at the next step. See the update in my previous post.

Smanar, 2021-08-02:
Ok so small problem ^^
You have not the same device than him.

I have see you have different MAC adress start, for you it s 0x847127 and him 0xbc33ac
Can you check  the manufacture number, 0x1002 and if you have the same model ID / Manufacture name ?

RenePhilipse, 2021-08-02:
Yes, as far as I can see and understand, this is all the same. See my screenshots below:

[image]
[image]

Smanar, 2021-08-02:
No, you are right.

No error on log when making the compilation and copying the file ?

Can you share some logs when doing the inclusion ? with "info" and "info_l2" ?

RenePhilipse, 2021-08-03:
> 
> 
> No, you are right.
> 
> No error on log when making the compilation and copying the file ?
> 
> Can you share some logs when doing the inclusion ? with "info" and "info_l2" ?

Oke Smanar, I fired up my test linux VM again. After I was done yesterday, I placed the conbee stick back into my production PI, and I had lost the connection to all my zigbee devices. Fortunately a simple reconnect did the trick and all the names etc. were still there. Now I plugged the stick back into the test VM again and search for devices, it connects all my existing zigbee devices from my production PI automatically (with default names though). I thougt the connection data is not an the stick but in the deconz program? How can it then be that all the devices connects also to the test environment?

However, I did the reconnect and catch the logs. During building the test environment using your command prompts I did not see any errors. Is there a way for me to doublecheck if I am using the right test software?

```
10:44:47:238 new node - ext: 0x847127fffea1ec6c, nwk: 0x94A8
10:44:47:238 device announce 0x847127FFFEA1EC6C (0x94A8) mac capabilities 0x80
10:44:47:239 set fast probe address to 0x847127FFFEA1EC6C (0x94A8)
10:44:47:239 FP indication 0x0000 / 0x0013 (0x847127FFFEA1EC6C / 0x94A8)
10:44:47:240                       ...     (0x847127FFFEA1EC6C / 0x94A8)
10:44:47:240 device announce 0x847127FFFEA1EC6C (0x94A8) mac capabilities 0x80
10:44:47:460 APS-DATA.indication from child 0x94A8
10:44:47:461 FP indication 0x0104 / 0x000A (0x847127FFFEA1EC6C / 0x94A8)
10:44:47:462                       ...     (0x847127FFFEA1EC6C / 0x94A8)
10:44:47:848 poll node 84:2e:14:ff:fe:3a:4a:ea-01-0b04
10:44:47:849 Poll ZHAPower sensor node Power 4
10:44:48:175 [1] get node descriptor for 0x847127fffea1ec6c
10:44:48:547 APS-DATA.indication from child 0x94A8
10:44:48:548 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 3, node: 0x94A8
10:44:48:548 DB pushZdpDescriptorDb()
10:44:48:548 DB save zll database items 0x00000893
10:44:48:549 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('2', 'normal', '00:21:2e:ff:ff:06:75:44-01', 'Configuration tool 2', '65520', '1', 'ConBee II', 'dresden elektronik', '0x26660700', '{"attr/id":"2","attr/lastannounced":null,"attr/lastseen":"2021-08-03T08:41Z","attr/manufacturername":"dresden elektronik","attr/modelid":"ConBee II","attr/name":"Configuration tool 2","attr/swversion":"0x26660700","attr/type":"Configuration tool","attr/uniqueid":"00:21:2e:ff:ff:06:75:44-01","state/reachable":true}')
10:44:48:553 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('3', 'normal', '60:a4:23:ff:fe:4f:ba:2d-01', 'On/Off plug-in unit 3', '65520,4', '1', 'SP 220', 'innr', '2.3', '{"attr/id":"3","attr/lastannounced":null,"attr/lastseen":"2021-08-03T08:43Z","attr/manufacturername":"innr","attr/modelid":"SP 220","attr/name":"On/Off plug-in unit 3","attr/swversion":"2.3","attr/type":"On/Off plug-in unit","attr/uniqueid":"60:a4:23:ff:fe:4f:ba:2d-01","state/alert":null,"state/on":null,"state/reachable":true}')
10:44:48:554 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('4', 'normal', '58:8e:81:ff:fe:d9:90:1a-01', 'Extended color light 4', '65520,9', '1', '14149506L', 'LIDL Livarno Lux', '', '{"attr/id":"4","attr/lastannounced":null,"attr/lastseen":"2021-08-03T08:44Z","attr/manufacturername":"LIDL Livarno Lux","attr/modelid":"14149506L","attr/name":"Extended color light 4","attr/swversion":null,"attr/type":"Extended color light","attr/uniqueid":"58:8e:81:ff:fe:d9:90:1a-01","config/colorcapabilities":null,"config/ctmax":65279,"config/ctmin":null,"state/alert":null,"state/bri":254,"state/colormode":"xy","state/ct":153,"state/effect":"none","state/hue":null,"state/on":null,"state/reachable":true,"state/sat":254,"state/x":24903,"state/y":10092}')
10:44:48:556 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('6', 'normal', 'ec:1b:bd:ff:fe:98:69:65-01', 'Color temperature light 6', '65520,2,6', '1', 'TRADFRI bulb GU10 WS 400lm', 'IKEA of Sweden', '2.0.029', '{"attr/id":"6","attr/lastannounced":null,"attr/lastseen":"2021-08-03T08:43Z","attr/manufacturername":"IKEA of Sweden","attr/modelid":"TRADFRI bulb GU10 WS 400lm","attr/name":"Color temperature light 6","attr/swversion":"2.0.029","attr/type":"Color temperature light","attr/uniqueid":"ec:1b:bd:ff:fe:98:69:65-01","config/colorcapabilities":null,"config/ctmax":65279,"config/ctmin":null,"state/alert":null,"state/bri":128,"state/colormode":"ct","state/ct":454,"state/on":false,"state/reachable":true}')
10:44:48:560 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('7', 'normal', 'ec:1b:bd:ff:fe:99:48:2a-01', 'Color temperature light 7', '65520,2,7', '1', 'TRADFRI bulb GU10 WS 400lm', 'IKEA of Sweden', '2.0.029', '{"attr/id":"7","attr/lastannounced":null,"attr/lastseen":"2021-08-03T08:44Z","attr/manufacturername":"IKEA of Sweden","attr/modelid":"TRADFRI bulb GU10 WS 400lm","attr/name":"Color temperature light 7","attr/swversion":"2.0.029","attr/type":"Color temperature light","attr/uniqueid":"ec:1b:bd:ff:fe:99:48:2a-01","config/colorcapabilities":null,"config/ctmax":65279,"config/c [... cut ...]
10:44:53:983 Poll light node Color temperature light 7
10:44:53:997 MAC poll fastEnddeviceProbe() 0x847127FFFEA1EC6C
10:44:53:998 skip check bindings for client clusters (no group)
10:44:54:266 skip check bindings for client clusters (no group)
10:44:54:501 skip check bindings for client clusters (no group)
10:44:54:675 Wait 29s till query finished
10:44:54:763 skip check bindings for client clusters (no group)
10:44:54:880 poll node 60:a4:23:ff:fe:88:b9:74-01
10:44:54:881 Poll light node On/Off plug-in unit 8
```

RenePhilipse, 2021-08-04:
Hi @smanar. Thanks for the information, I am learning a lot from you at the moment.

I retry it today (removed the device and added it again) and this time, the battery icon appeared successfully!
[image]

So it seems that your change has done the trick and can be merged in production branch.

Since I still want to learn:
-What makes a battery icon appear in the deconz gui (or defines a battery icons should be shown)? Did I do something wrong?
-There is a questionmark displayed when I open the clusters view. Is this an issue? Or is this purely esthetic icon?

Thanks again for your help and work!

Smanar, 2021-08-04:
>the battery icon appeared successfully

It's not something mandatory in deconz, and even if it work, you can need some hours for the icon be visible. It s not something that appear on all inclusions.

>There is a questionmark displayed when I open the clusters view. Is this an issue? Or is this purely esthetic icon?

Can you show me this questionmark ?

And no this code is to use the device in the API, not in deconz, IDK if this devie appear in phoscon but you need to test in third app or at least direclty with the API to be sure it s working.

RenePhilipse, 2021-08-05:
> It's not something mandatory in deconz, and even if it work, you can need some hours for the icon be visible. It s not something that appear on all inclusions.
> And this don't depend of me, I have changed nothing on deconz, only in the API

Ah, oke. Learned something again! If the battery shows in Home Assistant, I am happy already :-).

> Can you show me this questionmark ?

Yes, I meant his questionmark (also, this is just a question out of interest. I dont think that this depends on you, same as the battery icon you just explained):
[image]

> And no this code is to use the device in the API, not in deconz, IDK if this device appear in phoscon but you need to test in third app or at least direclty with the API to be sure it s working.

Aha, I didn't know that! I also have created a VM with an test Home Assistant environment then and connected it to the test deconz environment. The battery level is also showing in HA! (see image below). The only thing that I noticed was that the device in Home Assistant was named "Battery 7" while normally it is named as the device, so "Smart plug 17". I can rename it in HA, so it is not a big issue, but I wanted to mention it to you at least.
[image]

Smanar, 2021-08-05:
Ha, this one, it s a kind of "device type", you can have a bulb, a switch, or nothing too.
I have take a look fastly on google and it seem all "smart plug" have it.

For the name it s a long story ^^. For this device you have 1 "light" entry + 1 "sensor "entry, the name is not the same for both (for exemple for multi sensor, you can use "temperature room" "pressure room" and 'humidity room"), but HA mix all entries.
Something different in HA, it s you can set a new "global name" that don't impact the name of entry in the API.

So I m seing the battey level is working, but the on/off too ?

Smanar, 2021-08-05:
No thx it s perfect, have just done the PR.

closed 2021-09-04 by github-actions[bot]

rachedamer, 2022-04-23:
Any news on this? I have the device connected to a Conbee II on a raspi running Phoscon and switching it on and off works fine only for around half a day. After that it does not react anymore even though Phoscon does not display it as unreachable but as a regular switched on light bulb (which thinking of it is weird because it is displayed as a power outlet in the list of all devices, but as a light bulb within a room). 

Don't think it is the zigbee connection since it is right in between two permanently powered light bulbs. Also, I used brand new batteries.

Speaking of batteries: Any news on the battery status issue?

Smanar, 2022-04-24:
Your model is the "_TZ_3210_eymunffl" too ?

EarlOfBurton, 2022-04-24:
Hi @rachedamer
Hi @Smanar

my model is _TZ3210_eymunffl.
[image]

@rachedamer: Is yours different?
I'm running on recent deCONZ 2.14.1 with RaspBee II firmware 26720700

And yes, also for me it is buggy. But supporting more than all the data I cannot do.
A good year and 56 comments seem not to be enough to implement it properly, it seems.
So I gave up on it.

How I work around the foes:
No battery reading for me as well, which is a pretty risky, because imagine the batteries go bad after valve is turned on...

For connection issues I found out for me:
Every time I reboot deCONZ or the reboot the host of home assitant that I'm running on I need to push the green and red buttons on the irrigation device one time so that they talk to each other again.
Since I'm restarting deCONZ seldom (all 3-4 months) I can somehow live with that.

For me it seems that there is kind of a handshake from the irrigation device with the zigbee network and it is kept alive and never redone. When a deCONZ reboot happens the handshake is no more valid although it is still in the mesh.
This is definitely some stupid explanation from my side since I don't know anything about how zigbee works.
But this is definitely how my irrigation device works together with my RaspBee II.

Try opening and closing the valve with the green and red buttons on the irrigation device after any deCONZ/zigbee-reboot and see if the connection is stable then.

Maybe @Smanar can help solving the "handshake-problem" after restart and also integrate battery readout? Would be awesome...

Kind regards

Yardco, 2022-04-24:
Don't want to spark a "me too" chain of reactions, but im seeing exactly the same behaviour as @EarlOfBurton; No battery readout, not talking to home assistant anymore after rebooting host/deCONZ until I either press the button on the device physically or read the attributes in VNC. The latter makes me think that the connection is not necessarily broken, but for whatever reason no "conversation" is made until one of the actions mentioned prior is performed. Until now the device thankfully never became unresponsive after its turned on..

rachedamer, 2022-04-24:
Thanks a lot for your input! My model is _TZ3210_eymunffl too, my Conbee II firmware is 26660700, version 2.14.01 / 6.2.2022. Noticed the same behaviour: After pressing the green button, it works fine again. But since I reboot my Raspi once a day, this is not very viable… Since I just bought the irrigation control, maybe it's a good idea to send it back and replace it with another one. Any ideas which one might work instead?

Smanar, 2022-04-24:
Ok seem there is a new version https://github.com/Koenkk/zigbee2mqtt/issues/11751
The battery was working before for them too, but no more ATM.

ottelo9, 2022-04-25:
https://github.com/Koenkk/zigbee2mqtt/issues/12163

bsrdjan, 2022-06-04:
Same as @Homeboy05, no significant issues with deconz 6.12.1 (except battery). Sometimes there is ca. 5 sec delay when the on/off command sent from Home Assistant after day or two of inactivity, like the ventil is "sleeping". 
I tried deconz 6.14.1 and the connection has been lost, device could not be paired:
https://github.com/home-assistant/addons/issues/2509

Tetragramato, 2022-06-30:
Hello, i see this MR : https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5179

That is not merged, but ready to merge... Any news ?

Thanks ;) 👍

Mimiix, 2022-06-30:
> Hello, i see this MR : #5179
> 
> That is not merged, but ready to merge... Any news ?
> 
> Thanks ;) 👍

Probably wont be merged as it's not a DDF. 

@Smanar  can this be done with a DDF?

bsrdjan, 2022-06-30:
Would it provide battery status support as well?

Tetragramato, 2022-06-30:
I saw that https://forum.phoscon.de/t/woox-smart-garden-irrigation-control/1958

BTW i have the same kind of DDF for this device, and don't have the battery status, even editing it with other values : 

`
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3210_eymunffl",
  "modelid": "TS0101",
  "vendor": "WOOX",
  "product": "TS0101",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/woox/woox_r7060_ts0101.json",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_BATTERY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0001"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/battery",
          "refresh.interval": 3600
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 600,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
`

Smanar, 2022-06-30:
Try with adding the binding

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3210_eymunffl",
  "modelid": "TS0101",
  "vendor": "WOOX",
  "product": "TS0101",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/ts0101.json",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_BATTERY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0001"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/battery",
          "refresh.interval": 3600
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 43200
        }
      ]
    }
  ]
}
```

And if after 1 day you still have nothing replace 

```
        {
          "name": "state/battery",
          "refresh.interval": 3600
        },
```

by

```
        {
          "name": "state/battery",
          "awake": true,
          "refresh.interval": 4000,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        }, 
```

[18 more replies]