dresden-elektronik/deconz-rest-plugin #4326: view mesh map remotely from other device

superjaegermaster, 2021-02-01:
## Feature request type

It would be nice to have way from a PC or a Mac that has deconZ app install to connect remotely and view the mesh map

## Description

X11 didnt work for me in stretch to the RPi I have deconz + key on or Im really not using it correctly.
I would like to leave my RPI without GUI and keep it headless and lite
I think the quick win would be being able to connect from the windows/mac app (192.168.1.86 for ex) to a remote GW (192.168.1.10 for ex) and visulaize the mesh map from there, maybe via the API?

This feature would prevent people from having to workaround to view mesh map :
- installing GUI on a headles RPI (Raspi OS lite for ex)
- unplugging the conbee key to put it on a computer which changes the location from which user would analyze the mesh map and signal strength

## Considered alternatives

View mesh map in read only from web GW

## Additional context

  Add any other context or screenshots about the feature request here.

superjaegermaster, 2021-02-01:
added mockup image

Larsn1, 2021-02-02:
Maybe possible with PR 4136

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4136

superjaegermaster, 2021-03-08:
Hello, any news about if this can be integrated ?

would love to see this.

Thank you and best regards.

bemardev, 2021-03-22:
Did [this PR](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4605) too to support this functionnality.
You can see an example of returned JSON and of an implementation on jsfiddle.

superjaegermaster, 2021-06-25:
hello team, any news on whether this can be implemented , or at least permit mesh map to be viewed in some official way ?

Best regards,

ColinRobbins, 2022-01-08:
Any progress?
I have excatly the same situation - A raspberry pi running deconz and home assistant.  It totally headless, with no GUI elements installed.  So X11, VNC etc will not work.

mknet, 2022-09-28:
Any news on that topic? Thx!

queeek, 2023-01-29:
would be really highly appreciated if this could be done ;-)

SebastianMoinhinwech, 2025-01-19:
Any News on this? Maybe as Mesh MAP on the Deconz WebGUI?