dresden-elektronik/deconz-rest-plugin #4328: Aqara Single Switch Module T1 (With Neutral)

EarlOfBurton, 2021-02-01:
## Device

- Product name: Aqara Single Switch Module T1 (With Neutral)
- Manufacturer: LUMI
- Model identifier: lumi.switch.n0agi1
- Device type : Please remove all unrelated device types. 
  - Switch

Product page: https://www.aqara.com/en/single_switch_T1_with-neutral.html

## Screenshots
### Node
[image]
### Basic
[image]
### Phoscon
[image]
### Device temperature configuration
[02_device_temperature_configuration](https://user-images.githubusercontent.com/16398996/106503112-86553c80-64c5-11eb-81dc-e24e66763430.png)
### Identify
[image]
### Groups 1/2
[image]
### Groups 2/2
[image]
### Scenes 1/4
[image]
### Scenes 2/4
[image]
### Scenes 3/4
[image]
### Scenes 4/4
[image]
### On/Off 1/2
[image]
### On/Off 2/2
[image]
### Alarms
[image]
### Time
[image]
### Simple Metering 1/2
[image]
### Simple Metering 2/2
[image]
### Electrical Measurement
[image]
### Lumi specific
[image]
### OTAU
[image]
### 15 Home Automation On/Off
[image]
### 1F Home Automation On/Off
[image]
### 29 Home Automation On/Off
[image]
### F2 Green Power
[image]

 Home Automation On/Off

Smanar, 2021-02-01:
Hello, It s not working natively ?
Or it create too much devices, or it s just because it miss power measurement ?

EarlOfBurton, 2021-02-01:
Hi @Smanar, thanks for the reply. It is missing the simple metering and electrical measurement, as well as temperature and so on. The only feature that is on deconz is "on/off", that works. Kind regards.

SwoopX, 2021-02-15:
Device temperature is not exposed via API. I highly doubt the simple metering and electrical measurement clusters are truly available and usable. Like I got to know Xiaomi, any power/consumption values are taken from the analog input (basic) clusters. You can verify that by turning the wtisch on and reading the attribute 0x0055 from both endpoints (double click on the value, then press read and not read config).

Additionally, we would need like 15 mins of log data, having info level 5 enabled. During that timeframe, please also make sure that you turn the the switch on and off a couple of times.

EarlOfBurton, 2021-02-17:
Hello @SwoopX,
I did the logging, with INFO_L2 and ERROR_L2 activated. I hope that is what you need. Otherwise I could do a log run once more.
the device name is "aqara_t1_with_neutral_1" and there is no other device in this test (unless raspbee2).

Attached as load to the Aqara T1 is a 60 Watt "old school" bulb, so that there may come some useful reading out of the data.

I think the "at the moment" power measurement is in the "Electrical Measurement Cluster" as id 0x050B "Active Power" and needs to be divided by 10 to get the Watt-reading (maybe the id 0x0605 "AC Power Divisor" that reads "10" has that meaning.

The summation power over time that adds up is in the "Simple Metering Cluster", under id 0x000 "Current Summation Delivered", needs to be divided by 1000 (id "0x0302", "Divisor") and reads in kWh.

The hardware device has two terminals to add an external (old school) switch as well, that toggles the current "zigbee" state. I suppose this is so that you always can "hardware" switch or have a "smooth" transition from and old school installation to a smart home.
I also performed a few "on", "off", and "toggle" cycles via deCONZ via the "On/Off Cluster" at about timestamp 16:56:00.

I made a 20 seconds video clip (aqara_t1.mp4, see attached to this github-document) for you to see the test environment and the "external switch contacts" in action.

Also the log file is attached to this github-document as 20200217_deconz_log_aqara_single_switch_t1_with_neutral.txt-file, I hope this is how you like to have the raw data.
[20200217_deconz_log_aqara_single_switch_t1_with_neutral.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5997348/20200217_deconz_log_aqara_single_switch_t1_with_neutral.txt)

https://user-images.githubusercontent.com/16398996/108240689-5eafd680-714b-11eb-9deb-e9a1daa61e30.mp4

Thanks for all the work of you and your colleagues!
Kind regards

SwoopX, 2021-02-18:
Thanks for providing the requested info. The results are ... rather unexpected.

Now first I'm really surprised to only see the on/off attribute reports, and literally nothing else. However, this might be related to the fact that Aqara and Mija ZB3.0 devices do seem to behave very differently and the Aqara's tending to more adhere to the expected zigbee behavior.

Can you please manually bind the simple metering, electrical measurement and lumi specific cluster as described in here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...#manually-bind-server-clusters ? Then, please wait some minutes and take another log of 10 mins. The log settings are fine for the moment.

EarlOfBurton, 2021-02-19:
Do I understand it right that I can bind all three parameters in "one logging session"?

EarlOfBurton, 2021-02-19:
Hi @SwoopX ,
I hope I set the manual bindings right.
If you needed/wanted, I could setup a remote vnc connection to this raspi/deconz via WAN. It is a test environment only, so it would be possible to "mess around".

Attached you will find the file 20210219_log_aqara_single_switch_t1_with_neutral.txt
About 19:20:xx: Bound  all three simple metering, electrical measurement and lumi specific
About 19:31:xx started logging with 60W classic light blub as load
About 19:33:xx switching light about 10x on and off with "onboard" button on the device
About 19:33:50 keeping the light off for one minute (to get different measurement values)
About 19:35:00 turning the light on and off with the "external switch terminals" about 10 times
About 19:35:50 keeping the light turned on for different measurement values
About 19:38:00 turning off/on/toggle via deCONZ about 10 times - "last state" of this session: "off"
About 19:38:50 keeping turning the light "on" via deCONZ for measurement values
About 19:45:00 turned light off for different values
About 19:48:00 logging ended

Kind regards.

[20210219_log_aqara_single_switch_t1_with_neutral.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6012584/20210219_log_aqara_single_switch_t1_with_neutral.txt)

EarlOfBurton, 2021-03-14:
Bump to keep it alive.

SwoopX, 2021-03-27:
Although I somehow doubt that it will make and difference, I peeked at z2m for this device. There it is handled pretty much standard, so I added the code for the power and consumption sensors to be created. The interesting part will be to see if anything will drip in or not.

andudi, 2021-04-12:
I just tried the pull request #4660 above:
- my Aqara T1 Switch is now found as OnOff switch output
- the power/energy sensor is not found
- the input swiches are not found as well

is there some progress soon?
can this be improved in the plugin code? or must it be done in deconz itself?

thanks for clarify, I am willing to help if needed.

SwoopX, 2021-04-12:
I got no clue what you're trying to say, except for the absent measurement sensors. Can you please share a screenshot of the lumi specific cluster?

andudi, 2021-04-14:
@SwoopX I do not have screenshots, but I have some console results for now:

I captured the binding process with the CLI plugin, it looks like this:

<-EP 0x0149 0x0149   1 (01)
<-EP 0x0149 0x0149  41 (29)
<-EP 0x0149 0x0149  21 (15)
<-EP 0x0149 0x0149  31 (1F)
<-EP 0x0149 0x0149 242 (F2)
<-CLUSTER 0x0149 0x0149 ep 0x01 profile 0x0104 deviceid 0x0100 deviceversion 0x01
<-CLUSTER 0x0149 0x0149 0x01 In  0x0000 BASIC_CLUSTER_ID
<-CLUSTER 0x0149 0x0149 0x01 In  0x0002 unknown
<-CLUSTER 0x0149 0x0149 0x01 In  0x0003 IDENTIFY_CLUSTER_ID
<-CLUSTER 0x0149 0x0149 0x01 In  0x0004 GROUPS_CLUSTER_ID
<-CLUSTER 0x0149 0x0149 0x01 In  0x0005 SCENES_CLUSTER_ID
<-CLUSTER 0x0149 0x0149 0x01 In  **0x0006** ONOFF_CLUSTER_ID
<-CLUSTER 0x0149 0x0149 0x01 In  0x0009 ALARMS_CLUSTER_ID
<-CLUSTER 0x0149 0x0149 0x01 In  0x000A TIME_CLUSTER_ID
<-CLUSTER 0x0149 0x0149 0x01 In  **0x0702** SIMPLE_METERING_CLUSTER_ID
<-CLUSTER 0x0149 0x0149 0x01 In  **0x0B04** unknown
<-CLUSTER 0x0149 0x0149 0x01 In  0xFCC0 unknown
<-CLUSTER 0x0149 0x0149 0x01 Out 0x0019 OTAU_CLUSTER_ID
<-CLUSTER 0x0149 0x0149 ep 0x29 profile 0x0104 deviceid 0x0100 deviceversion 0x01
<-CLUSTER 0x0149 0x0149 0x29 In  0x0012 unknown
<-APS attr 0x0149 1 0x0000 0x0005 0x42 lumi.switch.n0agl1
<-ZCL serverToClient 0x0149 1 for cluster 0x0004 8A F0 FF 
<-APS attr 0x0149 1 0x0000 0x0005 0x42 lumi.switch.n0agl1
<-APS attr 0x0149 1 0x0000 0x0006 0x42 Aug  8 2020
<-CLUSTER 0x0149 0x0149 ep 0x15 profile 0x0104 deviceid 0x0100 deviceversion 0x01
<-CLUSTER 0x0149 0x0149 0x15 In  0x000C unknown
<-CLUSTER 0x0149 0x0149 ep 0x1F profile 0x0104 deviceid 0x0100 deviceversion 0x01
<-CLUSTER 0x0149 0x0149 0x1F In  0x000C unknown
<-ZCL serverToClient 0x0149 1 for cluster 0x0004 FF 01 F0 FF 
<-CLUSTER 0x0149 0x0149 ep 0xF2 profile 0xA1E0 deviceid 0x0061 deviceversion 0x01
<-CLUSTER 0x0149 0x0149 0xF2 Out 0x0021 unknown
<-BIND 0x0149 1 0 1 0x54EF44100009DACE 0x01 0x0000 0xFFF0
<-LQI 0x0000   01 0 1 0x54EF44100009DACE 0x0149 1 1 2 00 01 FF 
<-LQI 0x0000   01 0 1 0x54EF44100009DACE 0x0149 1 1 2 00 01 FF 
<-LQI 0x0000   01 0 1 0x54EF44100009DACE 0x0149 1 1 2 00 01 FF

It looks similar than what can be found in the deCONZ pictures of @EarlOfBurton above, the onoff cluster is present as well as the 0x0702 metering cluster and the interesting 0x0B04 power cluster.

Then I read sensors and ligths with the rest plugin:
   "3" : {
      "etag" : "65a6f4b3fc50490aa4ea80d35eb578f7",
      "hascolor" : false,
      "lastannounced" : null,
      "lastseen" : "2021-04-14T16:40Z",
      "manufacturername" : "LUMI",
      "modelid" : "lumi.switch.n0agl1",
      "name" : "On/Off light 3",
      "state" : {
         "alert" : "none",
         "on" : false,
         "reachable" : true
      },
      "swversion" : "Aug  8 2020",
      "type" : "On/Off light",
      "uniqueid" : "54:ef:44:10:00:09:da:ce-01"
   }

One new entry in the lights list, but no new sensors are bound yet!!!

Swtich on/off is already possible by button, device switch, with  REST and even with Phoscon.

At every switch, the new state is sent to the gate and can be shown with the REST or CLI:

<-LQI 0x0000   01 0 1 0x54EF44100009DACE 0x0149 1 1 2 00 01 FF 
<-ZCL attribute report 0x0149 0x0006 1 00 00 10 **01** 
->ZCL default response seq id 0x09 command id 0x0A, status 0x00
[... 49 lines cut ...]
      },
      "ep" : 1,
      "etag" : "afb6c2422921b344de69aeff940911dc",
      "lastseen" : "2021-04-14T16:47Z",
      "manufacturername" : "LUMI",
      "modelid" : "lumi.switch.n0agl1",
      "name" : "Power 11",
      "state" : {
         "current" : null,
         "lastupdated" : "2021-04-14T16:47:01.293",
         **"power" : 40,**
         "voltage" : null
      },
      "type" : "ZHAPower",
      "uniqueid" : "54:ef:44:10:00:09:da:ce-01-0b04"
   },

Again, the actual power is 4.0W, the factor can be found in another attribute to 1/10.

Another REST sensor show the total energy in kWh, but it is reseted at every restart of the device...

### My conclusion:
- the metering devices can be bound while reading one of the hidden attributes during the gate is open
- even when the REST API is showing the meters, Phoscon App do not! (yet?)
- when failing, place the antenna very close to the gate

SwoopX, 2021-04-14:
Ah, shxx. That one is on me. Forgot that the Xiaomi's are excluded from the automatic bindings.

> Another REST sensor show the total energy in kWh, but it is reseted at every restart of the device...

Interesting. My smart plug keeps the values even after a device reset. However, I assume the divisor for the consumption is correct, so that doesn't require any changes and the values are in Wh already.

> even when the REST API is showing the meters, Phoscon App do not! (yet?)

Correct. Phoscon doesn't show any power/consumption in general.

closed 2021-04-22 by manup

cobch, 2021-12-27:
I have installed the aqara switch module T1 today but no report of power in deconz. What is the status of power report in deconz? I am running HA via raspberry pi.

andudi, 2021-12-29:
Am Montag, 27. Dezember 2021, 19:31:49 CET schrieb cobch:
> I have installed the aqara switch module T1 today but no report of power in
> deconz. What is the status of power report in deconz? I am running HA via
> raspberry pi.
I can just say that it works for me but I use my own home system.
I read/write to the deconz direct over the REST API by curl commands.
But in Phoscon the Power Sensor is not shown, just the (Light)Switch.
Maybe the feature is only available in low level access?

cobch, 2021-12-29:
@andudi I first did not make the correct wiring and that was why the power did not report correctly... I manage to installed the switch and even the power off memory so it is automatically on after a power outage. Thanks for your help (-:

MrStrategy, 2022-01-21:
I would also like to know how to get power and energy via the rest interface,
Unfortunately I'm not so much in the details to fully understand everything discussed within this thread.

Actually I'm using the deconz / phoscon applications to bring Xiaomi (and other zigbee) devices into my home automation.
I'm using the rest interface to gather all the needed data. And it works quite well for all the deviced I used so far (mainly temperature sensors and some lights)

In my current setup I'm interacting with the Phoscon frontend to add the devices and to manage them.
I can do the same with the Aqara SSM-U01 and it is added as a On/Off switch in the lights section. This seems to be correct according to what i read so far,

Now I try to summarize what I read in this thread:

1. The power / energy monitoring will never show up in phoscon as it is not intended to be shown there.
The phoscon app is an abstraction layer above deconz focusing on light control. So the sensor data of power and energy is not there.

2. The REST API contains the data based on the underlying deconz system. If the sensors data is defined within deconz it should be available in the REST API even if it does not show up in phoscon.
(In my scenario there is no additional sensor data in the REST messages and the light message does not contain any data with regards to energy / power.)

3. @SwoopX  stated:

Ah, shxx. That one is on me. Forgot that the Xiaomi's are excluded from the automatic bindings.

Does this mean the process by simply adding the sensor via Phoscon is not sufficient to get the power / energy data in the REST API? Do I need to open up the deconz app and expose the specific measurements there?

I'm currently running the deconz app headless (platform minimal) and there was no need till now to use the deconz itself. So i do have no experience and cannot quickly validate my assumptions.
Maybe some ou you experts can confirm above or give me some tips how to proceed?

Thanks.

andudi, 2022-01-24:
hei Matthias

I used the following CURL syntax on my raspi:

# first I define some shortcuts:
$ path=localhost:80/api/<application ID>
$ format="json_pp -json_opt pretty,canonical"

# now I can get the config:
curl $path/config | $format

# and the list of lights:
$ curl $path/lights | $format
{
   "1" : {
...
   },
   "2" : {
      "etag" : "dd48815737a735c8c840123f30e326b4",
      "hascolor" : false,
      "lastannounced" : "2022-01-06T21:13:34Z",
      "lastseen" : "2022-01-24T19:28Z",
      "manufacturername" : "LUMI",
      "modelid" : "lumi.switch.n0agl1",
      "name" : "On/Off light 2",
      "state" : {
         "alert" : "none",
         "on" : false,
         "reachable" : true
      },
      "swversion" : "Aug  8 2020",
      "type" : "On/Off light",
      "uniqueid" : "54:eg:77:10:00:09:da:ce-01"
   },
}

# I can now switch on/off the switch like
$ curl -X PUT $path/lights/2/state -d '{"on":true}' | $format

# and I can read out the sensor states with energy consumption and actual power
curl $path/sensors | $format
{
   "12" : {
      "config" : {
         "on" : true,
         "reachable" : true,
         "temperature" : 0
      },
      "ep" : 1,
      "etag" : "1774c82ed1c27f12392d945029c88d26",
      "lastannounced" : "2022-01-06T21:13:34Z",
      "lastseen" : "2022-01-24T19:30Z",
      "manufacturername" : "LUMI",
      "modelid" : "lumi.switch.n0agl1",
      "name" : "Consumption 12",
      "state" : {
         "consumption" : 548,
         "lastupdated" : "2022-01-24T19:30:38.874"
      },
      "swversion" : "Aug 8 2020",
      "type" : "ZHAConsumption",
      "uniqueid" : "54:eg:77:10:00:09:da:ce-01-0702"
   },
   "13" : {
      "config" : {
         "on" : true,
         "reachable" : true,
         "temperature" : 0
      },
      "ep" : 1,
      "etag" : "8707559f80087435bba8123a4bca4a46",
      "lastannounced" : "2022-01-06T21:13:34Z",
      "lastseen" : "2022-01-24T19:30Z",
      "manufacturername" : "LUMI",
      "modelid" : "lumi.switch.n0agl1",
      "name" : "Power 13",
      "state" : {
         "lastupdated" : "2022-01-24T19:30:38.912",
         "power" : 0
      },
      "swversion" : "Aug 8 2020",
      "type" : "ZHAPower",
      "uniqueid" : "54:eg:77:10:00:09:da:ce-01-0b04"
   },
}

since the consumption is resetting while rebooting the device, I think it is not really
interesting feature, but to know the actual power can be nice

Andreas

Am Freitag, 21. Januar 2022, 14:06:36 CET schrieb Matthias:
> I would also like to know how to get power and energy via the rest
> interface, Unfortunately I'm not so much in the details to fully understand
> everything discussed within this thread.
>
> Actually I'm [...]

MrStrategy, 2022-02-16:
Hey Andreas,

thanks for sharing the feedback and the actual code. 
Unfortunately I could not get it to work with the commands you provided but I finally found a solution:

1. Connect the aqara switch via Phoscon web frontend to deconz (this is what I always do).
2. Open up deCONZ GUI and perform the read commands on the nodes 0x0B04 and 0x0702 as you have recommended it earlier in this thread I already get some values in deCONZ with regards to power and consumption.
3. Checking the API via commandline or phoscon showed no power or consumption readings.
4. This did the trick for me:  **Open phoscon app again and search for Sensors -> Others.**
  The search ended showing the green message new sensors were found. No new sensors are shown in Phoscon but the readings for power and consumption are now available in the REST API...

I'm not sure if step 3 is really necessary, but I do have only 3 devices and it took me some hours to figure out the actual process. For the first two devices I got the data rather by accident and on the third device I finally found above process.

It would be nice if someone else with the same problem could confirm the process and also check if step 3 is really necessary...

BR
Matthias

andudi, 2022-02-16:
cool if you found a solution.
I recently buyed a second device, and it just worked after binding with the
Phoscon web frontend without problems, the sensors are just shown in the REST
API like the first one...

7leman7, 2022-08-16:
Seems to be not covered above - is it possible to setup the inputs of Aqara T1 module in deCONZ (i.e. switch the input mode between momentary switch, regular switch etc.)?

flopp999, 2022-09-02:
> @andudi I first did not make the correct wiring and that was why the power did not report correctly... I manage to installed the switch and even the power off memory so it is automatically on after a power outage. Thanks for your help (-:

How did you do to put T1 to ON after a power failure. I didn't found it above

cobch, 2022-09-02:
It should be there if I am not wrong
[image]

flopp999, 2022-09-02:
Thanks. I looks correct but it did worked :(
Tested on two new devices. Bought them last week.

cobch, 2022-09-02:
Strange. It is working at home. When I switch of the electricity I have all the relay on again after without doing anything...

flopp999, 2022-09-02:
I will give it more tried during the weekend. It stores the setting that I choose but do not work.

flopp999, 2022-09-02:
Tried on same machine where I have deconz(not GUI) running by installing XFCE.
I needed to stop deconz to be able to connected to ConBee II but I think i read about this, so this is normal.
Can I send some printscreens so maybe we can find some differences

Smanar, 2022-09-02:
There is some device that don't support it, but they don't store the setting.
And you can stop deconz, the setting is memorised on the device.

A firmware difference ?

flopp999, 2022-09-02:
> There is some device that don't support it, but they don't store the setting. And you can stop deconz, the setting is memorised on the device.
> 
> A firmware difference ?

I am using ConBee II firmware 26780700.

It remembers the setting after I re-power the T1.
Anything else I can send to verify something? :)

flopp999, 2022-09-02:
I checked an old(2-3 years) T1, same behavior.
It said Previous but if was off after power came back.

Smanar, 2022-09-02:
When I say firmware I mean device firmware, can explain why a device is working and not another.
Have you try with the setting "on" or "off" insteaod of "previous" ?

flopp999, 2022-09-02:
> When I say firmware I mean device firmware, can explain why a device is working and not another.
> 
> Have you try with the setting "on" or "off" instead of "previous" ?

Will check firmware of T1, sorry :)

Yes, tried all three and it doesn't use the chosen when power comes back

Thanks for helping me

Smanar, 2022-09-02:
You are using a "1 state" switch (the simple push) or a "2 states" switch for the wiring ?

cobch, 2022-09-02:
@Smanar tell me if I can help with anything. You save my a... a couple of years ago with my Legrand dimmer (-:

flopp999, 2022-09-02:
I have same values as this pictures. Am I looking at correct page for firmware of T1?
my old T1 I have 2 states-button.
New once I have not used any button yet

Smanar, 2022-09-03:
And for users where the device is working ? Are you using the "1 state" or the "2 state" ?

For it s logic for the "1 state", because the device don't work with switch state but with action. But for the "2 state" I can understand for this feature is disabled and the device use the switch state, not this setting.

flopp999, 2022-09-04:
15 000C and id 0055 that is watt, correct?
1F 000C and id 0055 that is kWh or Wh, correct?
[image]

flopp999, 2022-09-04:
I was missing the 01 0702 Simple metering and 01 0B04 Electrical Measurement.
I was playing around.
I added T1 in Phoscon(it has been in use for a while) then I had activated Add Sensor as @MrStrategy wrote above and clicked on Read simple descriptors, then they came up

[image]

flopp999, 2022-09-04:
I also noticed that "01 0006 On/Off" disappears when I click on "Read simple descriptors" but I comes back when I switch ON or OFF the device

flopp999, 2022-09-05:
> I was missing the 01 0702 Simple metering and 01 0B04 Electrical Measurement. I was playing around. I added T1 in Phoscon(it has been in use for a while) then I had activated Add Sensor as @MrStrategy wrote above and clicked on Read simple descriptors, then they came up
> 
> [image]

But I now noticed that I do not get any devices in API

Smanar, 2022-09-05:
The cluster 0x000C can be anything, the used cluster is generaly the Electrical Measurement or the Simple metering.

>and clicked on Read simple descriptors, then they came up
>But I now noticed that I do not get any devices in API

Try the same thing but with phoscon in permit join (add new light) it at this moment where deconz can create/edit API entries.

flopp999, 2022-09-05:
OK
I tried to do it when "add new light" was pressed but now I don't see them.
I opened the Debug View and found this:
```
17:29:45:222 void DeRestPluginPrivate::pushZdpDescriptorDb(quint64, quint8, quint16, const QByteArray&),1137: assertion 'changes == 1' failed
17:29:45:241 void DeRestPluginPrivate::pushZdpDescriptorDb(quint64, quint8, quint16, const QByteArray&),1137: assertion 'changes == 1' failed
17:29:45:252 void DeRestPluginPrivate::pushZdpDescriptorDb(quint64, quint8, quint16, const QByteArray&),1137: assertion 'changes == 1' failed
17:29:45:432 void DeRestPluginPrivate::pushZdpDescriptorDb(quint64, quint8, quint16, const QByteArray&),1137: assertion 'changes == 1' failed
17:29:45:471 void DeRestPluginPrivate::pushZdpDescriptorDb(quint64, quint8, quint16, const QByteArray&),1137: assertion 'changes == 1' failed
```
Can this help you to figure out if the problem is in the device or deconz can be updated to solve it?
I can help you debug or anything you want me to do.

flopp999, 2022-09-05:
> OK I tried to do it when "add new light" was pressed but now I don't see them. I opened the Debug View and found this: "add new light" was scanning and I pressed "Read Simple descriptors"
> 
> ```
> 17:29:45:222 void DeRestPluginPrivate::pushZdpDescriptorDb(quint64, quint8, quint16, const QByteArray&),1137: assertion 'changes == 1' failed
> 17:29:45:241 void DeRestPluginPrivate::pushZdpDescriptorDb(quint64, quint8, quint16, const QByteArray&),1137: assertion 'changes == 1' failed
> 17:29:45:252 void DeRestPluginPrivate::pushZdpDescriptorDb(quint64, quint8, quint16, const QByteArray&),1137: assertion 'changes == 1' failed
> 17:29:45:432 void DeRestPluginPrivate::pushZdpDescriptorDb(quint64, quint8, quint16, const QByteArray&),1137: assertion 'changes == 1' failed
> 17:29:45:471 void DeRestPluginPrivate::pushZdpDescriptorDb(quint64, quint8, quint16, const QByteArray&),1137: assertion 'changes == 1' failed
> ```
> 
> Can this help you to figure out if the problem is in the device or deconz can be updated to solve it? I can help you debug or anything you want me to do.

Let me check so I have root rights before you answer :)

Smanar, 2022-09-05:
`assertion 'changes == 1' failed`

Is an output in SQL core part. It mean the SQL request haven't updated the database.
I can't be sure but it can be just for exemple if the device is already in the database without new value to change.

But you have Electrical Measurement and the Simple metering Cluster ATM ?
If yes you need to have thoses new sensor in the API ?

flopp999, 2022-09-12:
I found a way to set the T1 ON after power outage.
This is working even if 4003 is set to OFF
[image]
[image]

Smanar, 2022-09-12:
A device attribute that work only if the gateway is running ?

cobch, 2022-09-15:
@flopp999 I have the same issue with the T1. The only way to get the "power OFF memory" is with the help of the "Restore power on outage" option you descibed above.

My problem is that I got a T1 relay behind the plug that is serving the Home Assistant / deconz server. If I get a power outage on that plug it will not wake up. Do you have another idea how to do?

flopp999, 2022-09-15:
> The only way to get the "power OFF memory" is with the help of the "Restore power on outage" option you descibed above.

So you don't solve it by using "restore power on outage"?

I don't have any other idea.
Anyone know if you can set it by using an Aqara Hub?

psaggiomo, 2022-09-24:
Hi. I can see the power in 15 000C and id 0055, but I don't see "01 0702 Simple metering" and "01 0B04 Electrical Measurement".
Also I don't see Power Sensor in Home Assistant. Any ideas? Thanks

psaggiomo, 2022-09-27:
I haven't the cluster. In Device Compatibility List to Deconz is supported Power measurement.

In this screenshot I see watt in cluster 000C and attribute 0x0055:
[image]

cobch, 2022-09-30:
Same problem for me. I got 2 running for at least 1 year without any problem but when I reset one of them the power measurment did not came back. See the picture below. On the left the one running as before and on the right the one I added today where electrical measurment is missing.
[image]

Smanar, 2022-09-30:
And both are "lumi.switch.n0agi1" ?
Because they realy look different.

cobch, 2022-09-30:
No sorry, I got the wrong smart plug (-: Below is the 2 different "lumi.switch.n0agl1". One has been resetting recently and it is impossible to get the power any longer. The other one has been inside my network for at least one year... Simple metering is missing as you see )-:
[image]

Smanar, 2022-09-30:
This device probalby use the Lumi specific cluster.
But I don't see DDF for "lumi.switch.n0agl1" so this problem is from the legacy code, it have worked before ?

cobch, 2022-09-30:
I realize now that they were not supported and yet working really well since (-:
To answer you questions. Yes it has work, see the cluster "Electrical measurement" on the left device. Yes, the device stopped after reseting, I guess due to newer deconz/Hassio version. Yes, the cluster and devices looks excatly as the one on the left. The newer does not has any cluster called electrical measurment. BUT, I can see the electrical power (not energy consumption yet) from here:
000C -> 0x0055

Smanar, 2022-09-30:
I m asking that because there is no DDF on this device.
So if the cluster was present and no more ATM, I think it's more a problem with the device, perhaps it will appaear later, or if you re-include it, or If if you remake all requests using the first round on the node title.

On the code, for the device "lumi.switch.n0agl1" I m only seing code to use the cluster 0x0b04, so without this cluster, it can't work.

I don't see anycode using the cluster 0x000C and the cluster 0x0055 for the consumption.
Only on some DDF, for exemple for a xiaomi plug that have 2 versions, one with the classic cluster and one with the 0x000C https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/xiaomi/xiaomi_zncz04lm_smart_plug_v24.json

This plug probably had the same issue.

So if the device have never worked, and the cluster is not existant for sure, I think you can solve the issue with the same kind of DDF.

cobch, 2022-10-01:
I am not a pro but is there something I am missing here (-: It says the module is working in Deconz:
https://zigbee.blakadder.com/Aqara_SSM-U01.html

Not sure to understand what you mean with DDF /-:

cobch, 2022-10-01:
@Smanar I am a bit confuse since without doing anything, the cluster appears this morning (-:
The problem I have now is only the wrong energy consumption. A factor 1000 wrong somewhere? I hope oterwise my bill will be expensive next month.
[image]

Smanar, 2022-10-01:
Can you try that https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6372#issuecomment-1264291026 ?

cobch, 2022-10-03:
@Smanar The grid total comes from the main energy monitoring (Tibber Pulse). The devices that is not working here (aqara relay) is calles "Media" and even "Dishwasher". Look the total consumption is creasy: 3000 kWh in one day.

[36 more replies]