dresden-elektronik/deconz-rest-plugin #4330: Issue with Raspberry Pi 4 and Conbee II, cant find devices

MartinTerp, 2021-02-02:
## Describe the bug
Im setting up deconz for my dad, and I have a raspberry pi which im setting up his conbee II on, but when I search for devices, it finds nothing, im testing with a ikea bulb.

Things i've tried:
- move raspberry to stronger psu
- moved deconz from docker to host os ( from http://phoscon.de/apt/deconz )
- eventhough I run is at root, root is still a member of dialout group.
- usb extension cord for conbee II
- tried other zigbee device ( philips hue motion sensor )
- if I try to add the bulb to MY conbee II, it joins the cluster right away
- upgraded to 0x26680700

There is no errors in the log.

## Steps to reproduce the behavior
Try and pair a device like normal

## Expected behavior
Found the zigbee device

## Screenshots

## Environment
 - Host system: Raspberry Pi 4 Model B
 - Running method: Raspbian, deconz via apt (not docker)
 - Firmware version: 0x26680700
 - deCONZ version: 2.07.01 / 12/8/2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which?
 SSD via USB

## deCONZ Logs
```
12:51:57:503 send permit join, duration: 59
12:51:57:503 send GP proxy commissioning mode
12:51:57:503 enqueue event config/localtime for /config/
12:51:58:010 Mgmt_Lqi_req zdpSeq: 34 to 0x00212EFFFF065BCF start index 0
12:51:58:027 ZDP status = 0x00 -> SUCCESS
12:51:58:027 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
12:51:58:027 ZDP Mgmt_Lqi_rsp zdpSeq: 34 from 0x00212EFFFF065BCF total: 0, startIndex: 0, listCount: 0
12:51:58:502 enqueue event config/localtime for /config/
12:51:59:503 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
12:51:59:504 enqueue event config/localtime for /config/
12:52:00:411 Mgmt_Lqi_req zdpSeq: 35 to 0x00212EFFFF065BCF start index 0
12:52:00:418 ZDP status = 0x00 -> SUCCESS
12:52:00:418 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
12:52:00:418 ZDP Mgmt_Lqi_rsp zdpSeq: 35 from 0x00212EFFFF065BCF total: 0, startIndex: 0, listCount: 0
12:52:00:503 enqueue event config/localtime for /config/
12:52:01:504 enqueue event config/localtime for /config/
12:52:02:503 enqueue event config/localtime for /config/
12:52:02:811 Mgmt_Lqi_req zdpSeq: 36 to 0x00212EFFFF065BCF start index 0
12:52:02:826 ZDP status = 0x00 -> SUCCESS
12:52:02:826 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
12:52:02:826 ZDP Mgmt_Lqi_rsp zdpSeq: 36 from 0x00212EFFFF065BCF total: 0, startIndex: 0, listCount: 0
12:52:03:503 enqueue event config/localtime for /config/
12:52:04:503 enqueue event config/localtime for /config/
12:52:05:211 Mgmt_Lqi_req zdpSeq: 37 to 0x00212EFFFF065BCF start index 0
12:52:05:218 ZDP status = 0x00 -> SUCCESS
12:52:05:218 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
12:52:05:218 ZDP Mgmt_Lqi_rsp zdpSeq: 37 from 0x00212EFFFF065BCF total: 0, startIndex: 0, listCount: 0
12:52:05:503 enqueue event config/localtime for /config/
12:52:06:503 enqueue event config/localtime for /config/
12:52:06:503 sql exec SELECT conf FROM zbconf ORDER BY rowid desc limit 1
```
after this it just seems to be the same 
```
12:53:19:626 ZDP status = 0x00 -> SUCCESS
12:53:19:626 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
12:53:19:626 ZDP Mgmt_Lqi_rsp zdpSeq: 68 from 0x00212EFFFF065BCF total: 0, startIndex: 0, listCount: 0
12:53:20:503 enqueue event config/localtime for /config/
12:53:21:504 enqueue event config/localtime for /config/
12:53:22:010 Mgmt_Lqi_req zdpSeq: 69 to 0x00212EFFFF065BCF start index 0
``` 

## Additional context

Mimiix, 2021-02-02:
Hi! Are you able to provide some logs:

To enable logging. Open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

MartinTerp, 2021-02-02:
The logs provided is with:
/usr/bin/deCONZ -platform minimal --http-port=80 --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1

MartinTerp, 2021-02-02:
Tried to update to v2.9.1, no difference.

Mimiix, 2021-02-02:
Are you able to provide APS level 2.

MartinTerp, 2021-02-02:
Provided here: https://termbin.com/i25q

Also, tried to change channel, but then i got:
```
08:04:40:354 change channel with nwkUpdateId = 1
08:04:40:365 change channel to 11, channel mask = 0x00000800
08:04:41:465 network configuration verified!
08:05:29:849 Websocket disconnected 192.168.1.153:10463, state: 0, close-code: 1001, reason:
08:05:31:188 New websocket 192.168.1.153:10662 (state: 3)
08:05:33:279 Websocket disconnected 192.168.1.153:10662, state: 0, close-code: 1001, reason:
08:05:34:422 New websocket 192.168.1.153:10694 (state: 3)
08:05:35:872 Current channel 25
08:05:35:876 CTRL got nwk update id 1
08:05:35:880 Device TTL 2984 s flags: 0x7
08:05:45:006 New websocket 192.168.1.153:10940 (state: 3)
08:05:45:854 channel 25 does not match channel mask 0x00000800 (TODO)
08:05:45:854 channel is 25 but should be 11, start channel change
08:05:45:855 Skip automatic channel change, TODO warn user
08:05:55:854 channel 25 does not match channel mask 0x00000800 (TODO)
08:05:55:855 channel is 25 but should be 11, start channel change
08:05:55:855 Skip automatic channel change, TODO warn user
08:05:55:856 send permit join, duration: 59
08:05:55:856 send GP proxy commissioning mode
08:06:05:853 channel 25 does not match channel mask 0x00000800 (TODO)
08:06:05:854 channel is 25 but should be 11, start channel change
08:06:05:854 Skip automatic channel change, TODO warn user
08:06:15:854 channel 25 does not match channel mask 0x00000800 (TODO)
08:06:15:855 channel is 25 but should be 11, start channel change
08:06:15:855 Skip automatic channel change, TODO warn user
08:06:25:853 channel 25 does not match channel mask 0x00000800 (TODO)
08:06:25:854 channel is 25 but should be 11, start channel change
08:06:25:854 Skip automatic channel change, TODO warn user
08:06:35:854 channel 25 does not match channel mask 0x00000800 (TODO)
08:06:35:854 channel is 25 but should be 11, start channel change
08:06:35:855 Skip automatic channel change, TODO warn user
08:06:35:868 Current channel 25
```

Switched it back to 25

Smanar, 2021-02-02:
Arf bad luck, hopping for you don't have this issue > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4005

Try to change the channel again, and if possible with the GUI (take care you need to press button on the good order > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help)

MartinTerp, 2021-02-02:
Well I dont NEED to change channel, was more to test to see if that would solve my cant-pair-anything issues

Smanar, 2021-02-02:
I  had someone else with a similar problem

>13:12:04:686 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt

Have you installed deconz on pi user or root ?

MartinTerp, 2021-02-02:
```
root@homeassistant:/tmp# sudo find / -name "zll.db"
/app/docker/deconz/config/zll.db
/root/.local/share/dresden-elektronik/deCONZ/zll.db
/home/pi/.local/share/dresden-elektronik/deCONZ/zll.db
```

The first one is from the docker mount
the /home/pi is from the normal "service deconz start" and the /root/ one is from just running it as root, for debug

Smanar, 2021-02-02:
Ok, so not the same issue at all ...

MartinTerp, 2021-02-06:
@Smanar do you know, if I can try and take my own conbee  and plug it in the raspberry?
Will that ruin my own cluster somehow or is all cluster info stored on the host?

Smanar, 2021-02-06:
It will not work (because not the same network key for exemple)
I don't think there is danger, the devices configuration is stored on the host, and the network configuration on the conbee.

But make a backup before (using phoscon feature), I don't know all the "deconz magic", and I can't be sure deconz will not try to write something on your device.

MartinTerp, 2021-02-06:
Think I will just leave my own cluster be as it is :D one thing is that my dad is irritated, but don't want the wife to be also ;)

Smanar, 2021-02-06:
Yeah, I m realy sorry, but I realy don't have idea, no special error on logs, and you seem skilled enought for linux checks

>if I try to add the bulb to MY conbee II, it joins the cluster right away

It s your conbee + your machine ?

You are using the usb 2.0 port ?

MartinTerp, 2021-02-06:
Yes it my own setup, pc + conbee II
where my dads is raspberry + conbee II, tried using both usb ports

MartinTerp, 2021-02-08:
So, I tried to install my dads conbee II on a mac mini I have, and it works!
So its a configuration issue on the raspberry ( or something else ), eventhough I use the same configuration.

ideas? The raspberry is not complaining about power

MartinTerp, 2021-02-08:
New status

tried to boot up and install on the SD card and boot up the raspberry again, that also works.
maybe it is a power issue then :-/

MartinTerp, 2021-02-08:
Im just going to stick with the sd card for now
if its the ssd making noice or taking to much power i dont know.
thanks for helping @Smanar

closed 2021-02-08 by MartinTerp

Smanar, 2021-02-08:
Perhaps the power, as you are using the USB extension ? You can make a test with using it too with the SSD (if it use usb too)

drgnomage, 2021-12-14:
Same issue, I can't connect anything to brand new Conbee II.  Everything worked on my old Sonoff bridge but that sadly died so I know the devices function correctly.  The power for my RPi 4 is the official RPi 4 power supply so power shouldn't be an issue.  No SSD for me but I am running off of a thumb drive and also have an arduino attached.

Mimiix, 2021-12-14:
@drgnomage please open an forum topic for this

drgnomage, 2021-12-14:
@Mimiix I've just loaded up the Conbee on my laptop with the same docker-compose config and it works fine so the thread is probably right, it's a power issue.

Nguimjeu, 2022-01-15:
I'm dealing with I think the same issue. My setup:
- Raspberry PI 8GB
- SSD Extension board for NVMe M.2
- Ubuntu server 64-bit (booting from the SSD)
- Homeassistant (supervised)
- Conbee 2
- USB-C 5V, 3A Power supply

If it's power issue how do you guy suggest to solve it?

Smanar, 2022-01-16:
Please open a new issue, it s not so simple.
Have you the extension cable, the firwmare number is displayed on phoscon, it s working on aPC, ect ...

Mimiix, 2022-01-16:
@Nguimjeu a forum post :)

Nguimjeu, 2022-01-16:
> Please open a new issue, it s not so simple. Have you the extension cable, the firwmare number is displayed on phoscon, it s working on a PC, ect ...

OK noted.
It works flawlessly on a PC (Windows) and I've even flashed the latest firmware.
Thanks for the reply.