dresden-elektronik/deconz-rest-plugin #4331: raspbee II not accessable

rene-mueller, 2021-02-02:
## Describe the bug

I have restarted my raspi. after that the hardware is not available. i cant reset or access informations

## Screenshots

[image]

## Environment

not available

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: not available
 - deCONZ version: deconz/unknown,now 2.07.01-raspbian-buster-stable
 - Device: RaspBee II
 - Is there any other USB or serial devices connected to the host system? If so: Which? NO

## deCONZ Logs

## Additional context

Mimiix, 2021-02-02:
Hi,

The above occurs when the Addon doesn't see the Raspbee. Did you click "Firmware update?" What is your Ha addon config?

rene-mueller, 2021-02-02:
Hi @Mimiix,
I updated the firmware a few months ago and then restarted everything. I then had to set up the entire zigbee devices again.

```
{
            "name": "Hue",
            "anyOn": true,
            "hosts": [
                "127.0.0.1:4530"
            ],
            "users": {
                "XXXXXXXXXXXXXXX": "XXXXXXXXXX"
            },
            "lights": true,
            "linkButton": false,
            "nativeHomeKitLights": true,
            "nativeHomeKitSensors": true,
            "sensors": true,
            "wallSwitch": true,
            "effects": true,
            "platform": "Hue"
        }
```

Mimiix, 2021-02-02:
I have no clue where that comes from...?

rene-mueller, 2021-02-02:
Same. Is it possible that the chip is broken?

Mimiix, 2021-02-02:
Where did you get that json from?

rene-mueller, 2021-02-02:
From homebridge. Sorry I selected the wrong running method

Smanar, 2021-02-02:
Just by curiosity what is the result of 

`GCFFlasher_internal -l`

rene-mueller, 2021-02-03:
@Smanar
```
pi@homebridge:~ $ GCFFlasher_internal -l
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee
```

sonk-ai, 2021-02-03:
@rene-mueller @Smanar 
I am having the exact same issue as @rene-mueller RaspBee shown with 0x0000 0x0000 after updating
the Firmware and DeCONZ package following this guide: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian

rene-mueller, 2021-02-03:
I have already done a general apt update and upgrade. So I should reset the gateway completely. Can you (@sonk-ai) tell me how you did that?

Smanar, 2021-02-03:
Just to be sure can you check if you still have this line too https://prnt.sc/xx5j5l

Smanar, 2021-02-03:
How is deconz on homebridge ? it s a normal installation or using VM/docker ?
Are you able to run deconz with logging enabled ?

rene-mueller, 2021-02-03:
Hey @Smanar it's a normal installation. Can you tell me what I need to do for this?

Smanar, 2021-02-03:
If you are using desktop > deconz / help / log with "info" "info_l2" "error" "error_l2"
If you are using headless > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags

sonk-ai, 2021-02-04:
> I have already done a general apt update and upgrade. So I should reset the gateway completely. Can you (@sonk-ai) tell me how you did that?

I actually deleted every Light in DeCONZ-GUI directly and afterwards I also did a "Reset Gateway" in Phoscon. 
>Settings > Gateway

rene-mueller, 2021-02-04:
@sonk-ai this does not work for me. There is an error that the gateway is not accessible.

rene-mueller, 2021-02-04:
Before the error (from screenshot):

The interface was accessible but there was no lights. Just one switch. And the reset button also doesn't worked.

rene-mueller, 2021-02-04:
@Smanar 
[deconz.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5925900/deconz.log)

rene-mueller, 2021-02-05:
@Smanar @Mimiix In the meanwhile i tryed to reset the gateway by deleting the configuration in filesystem. The gateway still didn't set up after that. It didn't find any lamps. After skipping this step, the same screen was shown as in the initial issue above.

Mimiix, 2021-02-05:
Asked @manup to check this issue.

rene-mueller, 2021-02-05:
ok @Mimiix in the log file are senetive informations. how can i change the the apikey?

Smanar, 2021-02-05:
Don't worry, not something dangerous, but to create a new one

`curl http://IP:PORT/api/ -X POST -d '{"devicetype": "Newkey"}'`

To delete it

`curl http://IP:PORT/api/KEY/config/whitelist/OLDKEY -X DELETE`

manup, 2021-02-06:
>I am having the exact same issue as @rene-mueller RaspBee shown with 0x0000 0x0000 after updating

Note that's always the case 0x0000 0x0000 is shown for RaspBee since it's just a UART connection not providing the vendor/product id's of USB devices.

>After a general apt update and upgrade it all seems to work again, though I had to reset the Gateway completely.

A reset isn't needed, usually such problems are on the basic level like deCONZ connecting to the firmware fails for some reason which needs to be figured out. Sometimes it's the serial port configuration changed after a system update, firmware gone rogue (seen in earlier firmware versions), or Zigbee parameters which need to be checked for modifications.

For the RaspBee II I'd recommend to update to the latest firmware version 0x26690700 as it has quite some fixes.
Did you already have this version installed (I can't tell from the logs)?

rene-mueller, 2021-02-27:
@manup i have tryed the Update. Now the connection and the software works again. :)

Thank you!

Mimiix, 2021-03-01:
#caseclosed

closed 2021-03-01 by Mimiix