dresden-elektronik/deconz-rest-plugin #4338: Problem after restarting deConz plugin in Home Assistant

Gawronnek, 2021-02-02:
## Describe the question or issue you are having
In HA I have several devices that are presented in Phoscon as Heiman TS0011, unfortunately after each restart of deConz, they behave weird, as if they are, because turning them on / off activates other entities when they are assigned to them, but they do not supply power to the devices connected to them .
After rebooting these devices, go to deConz (VNC) and click Read Simple Descriptor (s) in each of them, and then everything works fine. I'm a layman, so I have no idea why this is happening, but maybe there is a way to trigger this action with some automation that starts with deConz?
I also do not know what information I would have to provide, if it is possible to make a correction in the deConz add-on to HA.
I would like to add that I speak English poorly and I use a translator, and I did not find any help on Polish-language forums.
## Screenshots

## Environment
ConBee 2 firmware 26680700 
Phoscon ver 2.07.01/.12.200

Raspberry Pi 4 
Home Assistant OS 5.10

 - Do you use an USB extension cable: yes 

## deCONZ Logs

## Additional context

Smanar, 2021-02-02:
Hello, do you have the device JSON ? If you can't find it, I need at least the manufacture name (if it s not heiman).
It s wired or battery device (switch ?) ?

Gawronnek, 2021-02-02:
is that it?
        "etag": "905bda0f20ada945b5c3f45ebfb2f547",
        "hascolor": false,
        "lastannounced": "2021-01-30T09:59:06Z",
        "lastseen": "2021-02-02T19:59Z",
        "manufacturername": "_TYZB01_xfpdrwvc",
        "modelid": "TS0011",
        "name": "Swiatlo przedpokoj",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
-------------------------------------------
        "etag": "5486b6fe51ef4a677f86c73ea4fe1ec8",
        "hascolor": false,
        "lastannounced": "2021-01-29T16:00:39Z",
        "lastseen": "2021-02-02T20:04Z",
        "manufacturername": "Heiman",
        "modelid": "TS0013",
        "name": "Swiatlo sciana",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
---------------------------------------------
        "etag": "239c00362e103640c6565733eb250a75",
        "hascolor": false,
        "lastannounced": "2021-02-02T19:58:29Z",
        "lastseen": "2021-02-02T20:00Z",
        "manufacturername": "_TYZB01_rifa0wlb",
        "modelid": "TS0011",
        "name": "Zawor kuchnia",
        "state": {
            "alert": "none",
            "on": true,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",

Smanar, 2021-02-02:
Yes perfect thx ^^.

So you have 3 differents models TS0011 and one TS0013 ?
And after the reset some of them act on the light connected to other device instead of their own light ? And all is fine after having read the basic attribute on all of them ?

Gawronnek, 2021-02-02:
As I mentioned, I am a layman and I speak little English, I do not understand everything you are asking. In total, I have 2 flush-mounted switches, 3 surface-mounted switches, it is light and one device for closing the water valve. I also have other devices, but I have no problem with them. These surface-mounted switches just need to be turned on physically and they work without any problems until the next restart.
The last counts "uniqueid" is in all 01

I have one connected so that its launch starts another entity and it works, but the bulb connected to it does not work, I checked it does not get voltage, manual activation with a physical button, returns normal operation, as well as calling in VNC Read Simple Descriptor

Smanar, 2021-02-03:
Your device look "tuya standard"

>I have one connected so that its launch starts another entity and it works, but the bulb connected to it does not work, I checked it does not get voltage, manual activation with a physical button, returns normal operation, as well as calling in VNC Read Simple Descriptor

I m not sure to understand the difference with the previous one ?

But yes I have checked on my side for the "Read simple descriptor"
- Endpoint the case is not checked
- Router the case is checked

But it don't have impact on my side.

So to resume to have your devices working, you need read the simple descriptor or use it manualy one time ?

And all your problemactic devices are wired but see on deconz as end device (gray bar at left instead of yellow on title) ?

New mystery ....
Can you show a log when you try to use  a device with HA (before reading descriptor or use it) ?
You can find logs in deconz/help/logs.

Gawronnek, 2021-02-03:
> 
> So to resume to have your devices working, you need read the simple descriptor or use it manualy one time ? **Yes**
> 
> And all your problemactic devices are wired but see on deconz as end device (gray bar at left instead of yellow on title) ? **Yes**
> 
> New mystery ....
> Can you show a log when you try to use a device with HA (before reading descriptor or use it) ?
> You can find logs in deconz/help/logs.   -  tomorrow evening

Gawronnek, 2021-02-04:
I do not know if this is what you mean, but here after the restart I tried to turn on the icon in HA, the icon changed color in Phoscon is also on, but the bulb is not lit.
16:23:44:316 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0xEC1BBDFFFE687D5A (wait reading or unsupported)
16:23:44:565 ZCL attribute report 0xEC1BBDFFFE687D5A for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
16:23:44:569 	0xEC1BBDFFFE687D5A force poll (2)

I turned it off here, in Phoscon it seems to be turned off
16:25:15:819 ZCL attribute report 0xEC1BBDFFFE687D5A for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
16:25:15:827 	0xEC1BBDFFFE687D5A force poll (2)

and here after the repair, clicking on VNC in Read Simple Descriptor (s), it works fine
16:27:51:492 ZCL attribute report 0xEC1BBDFFFE687D5A for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
16:27:51:500 	0xEC1BBDFFFE687D5A force poll (2)
16:27:54:652 Current channel 20

I don't know how to save the whole log, because I have a lot of zigbbe devices and the log scrolls quickly, but these entries are correct

Smanar, 2021-02-04:
I realy don't have idea what happen.
You can't trigger a simple descriptor request using the API, but deconz do one automaticaly for light device (the yellow bar)

Something you can check in logs using aps and aps_l2 (can disable all other)

```
18:52:28:601 APS-DATA.indication srcAddr: 0x938a, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -43
18:52:28:606 	asdu: 08e90b0000
18:52:28:634 APS-DATA.indication srcAddr: 0x938a, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -43
18:52:28:640 	asdu: 08490a00001000
```

IDK why there is 2 lines, but it s the command send by deconz to the device. To have them, select the cluster 0x0006 ("on/off") then in the cluster panel you can send "on and "off" command.

>srcAddr: 0x938a

Is the network adress of the device

>srcEp: 0x01

Is the endpoind, I think the problem can come from it.

>profile: 0x0104, cluster: 0x0006

You will have same

>18:52:28:606 	asdu: 08e90b0000
>18:52:28:640 	asdu: 08490a00001000

You will have almost the same thing, just with 0/1 difference for "on" and "off", and ofc all the first char will be differents.

IDK if Rudolph is available on HA, but you can ask that to discord too : https://discord.gg/QFhTxqN there is a romm just for HA.

Gawronnek, 2021-02-04:
Thanks for your help, but I have too little knowledge and I don't understand what you are writing to me now.
Is there any way to fix these 8 devices with some automation without going to VNC?

Smanar, 2021-02-04:
No, I don't think, but this working mode is not normal.

Gawronnek, 2021-02-04:
It is difficult, perhaps it is the fault of the hardware manufacturer's software, but there are people with such equipment on the Polish forum on facebook and they do not report this problem. But you can live with it.
I do not know enough about it, so far poor knowledge of the English language. Many thanks for your help, for making you want to.

Gawronnek, 2021-02-13:
I solved this problem by switching to zigbee2mqtt from Conbee 2

closed 2021-02-13 by Gawronnek