dresden-elektronik/deconz-rest-plugin #4340: Tuya door window sensor (TS0203)

pavime, 2021-02-03:
## Device

- Product name: Tuya Contact Sensor
- Manufacturer: _TYZB01_ncdapbwy
- Model identifier: TS0203
- Device type :
  - Sensor

## Screenshots
[image]
[image]
[image]
[image]

Thank you for your great job
Regards

pavime, 2021-02-28:
Hello,
I upgraded to version 2.9.3, removed and re-discover this sensor, it's still unknown
Thank you for this so good gateway
Regards

Smanar, 2021-03-01:
Hello, this sensor is not yet added.
Do you have a linux machine to test code modifications ?

pavime, 2021-03-02:
Hello,
No, sorry, only a Windows PC and a RPi 4 with HAOS 5.12 and the ConBee II stick
Thank you

closed 2021-04-02 by github-actions[bot]