dresden-elektronik/deconz-rest-plugin #4342: ConBee II "bricked" after docker firmware update

davewood, 2021-02-04:
## Describe the question or issue you are having

I tried to flash a new firmware but got a timeout error.
firmware-update.sh doesnt set a timeout so a default of 3 seconds seems to be applied.

the ConBee II still worked.

I added a timeout of 60 seconds to the script. `-t 60` and ran the flasher again.
this time I still got the 3s timeout error and afterwards the stick didnt work anymore.

Now I keep getting these messages every 10s in `/var/log/messages` if i insert the conbee2. to me it seems the stick is in a connect/disconnect loop.

## Screenshots
I dont have any screenshots

## Environment
 - GCFFlasher V3_17
 - deCONZ_ConBeeII_0x26660700.bin.GCF
 - host: debian buster
 - Device: ConBee II
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - No other USB or serial devices connected

## deCONZ Logs
```
Feb  4 12:27:25 xps kernel: [217206.231445] usb 1-1: new full-speed USB device number 49 using xhci_hcd
Feb  4 12:27:25 xps kernel: [217206.381045] usb 1-1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Feb  4 12:27:25 xps kernel: [217206.381051] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb  4 12:27:25 xps kernel: [217206.381054] usb 1-1: Product: ConBee II
Feb  4 12:27:25 xps kernel: [217206.381057] usb 1-1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Feb  4 12:27:25 xps kernel: [217206.381060] usb 1-1: SerialNumber: DE2219384
Feb  4 12:27:25 xps kernel: [217206.382716] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
Feb  4 12:27:25 xps mtp-probe: checking bus 1, device 49: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1"
Feb  4 12:27:25 xps mtp-probe: bus: 1, device: 49 was not an MTP device
Feb  4 12:27:25 xps mtp-probe: checking bus 1, device 49: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1"
Feb  4 12:27:25 xps mtp-probe: bus: 1, device: 49 was not an MTP device
Feb  4 12:27:28 xps kernel: [217210.199132] usb 1-1: USB disconnect, device number 49
```

## Additional context

this is supposed to work according to your docker-deconz repository:
`The script has been tested and verified to work for Conbee on amd64 Debian linux` 
https://github.com/dresden-elektronik/docker-deconz#updating-conbeeraspbee-firmware

I dont have a windows system available.

SwoopX, 2021-02-04:
Just fyi, the repository you referred to is a fork from marthoc, so stricktly no official running method.

From the log messages, it appears like modemmanager is not playing nicely.

davewood, 2021-02-04:
hi and thanks for the info, I know that this is a fork and im pretty sure my modemmanager is working. What do you mean by "not playing nicely" and what in my logs makes you think thats the case?

Mimiix, 2021-02-04:
What is the host device? What hardware and Os? 

You seem to have left out that info and replaced it with other stuff.

davewood, 2021-02-04:
its a Dell XPS 13 running debian buster.

Mimiix, 2021-02-04:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian

You should be able to use that.

davewood, 2021-02-04:
success!

 - install dependency
`apt install libqt5serialport5`
 - get amd64 deb package
`wget http://deconz.dresden-elektronik.de/ubuntu/stable/deconz-2.07.01-qt5.deb`
 - extract files
`ar x deconz-2.07.01-qt5.deb data.tar.xz`
`tar xvJf data.tar.xz`
 - change into extracted bin directory
`cd usr/bin`
 - stop ModemManager
`systemctl stop ModemManager`
 - firmware image has also been extracted locally into usr/share
 - flash firmware, make sure you are using the correct device path
```
./GCFFlasher_internal -t 60 -d /dev/ttyACM0 -f ../share/deCONZ/firmware/deCONZ_ConBeeII_0x26660700.bin.GCF`
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2219384-if00 (ConBee II)
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 164106 bytes: |==============================|
verify: .
SUCCESS
Wait 10 seconds until application starts
```
 - start ModemManager
`systemctl start ModemManager`

Mimiix, 2021-02-04:
Yay! 

Case closed?

closed 2021-02-04 by davewood

Mimiix, 2021-02-04:
Case closed! 🎉