dresden-elektronik/deconz-rest-plugin #4343: Aqara Door & Window Sensor T1 MCCGQ12LM

AntiHeld889, 2021-02-04:
## Device

- Product name: Aqara Door & Window Sensor T1 MCCGQ12LM
- Manufacturer: LUMI
- Model identifier: lumi.sensor_magnet.agl02
- Device type :
  - Sensor

## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2021-02-04:
Where did you get those? :)

AntiHeld889, 2021-02-04:
A few stores on Aliexpress are now offering the new Zigbee 3.0 devices.

for example here:
https://de.aliexpress.com/store/911222034?spm=a2g0o.detail.1000007.1.64873fb2N9MFwT

But you have to look carefully to find the right Zigbee 3.0 devices.

closed 2021-02-22 by manup