dresden-elektronik/deconz-rest-plugin #4345: Xiaomi Door Sensor (lumi.sensor_magnet) - incorrect pairing

Nazarbayevv, 2021-02-04:
## Describe the question or issue you are having

I add new Xiaomi Door Sensor via Phoscon App and I can see sensor status (open/close) from App, But I can not see any Attributes in claster from Deconz. When I restart deconz, node are disappearing. And sensor status does not change in Phoscon App. I can see record in database, but without swversion info.    
P.S. I have 2 Aqara door semsor, which work fine

## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26390500
 - deCONZ version: 2.07.01 / 08.12.2020
 - Device: RaspBee I
 - Do you use an USB extension cable:  -
 - Is there any other USB or serial devices connected to the host system? cooler to gpio

## deCONZ Logs

## Additional context

Nazarbayevv, 2021-02-04:
[log_xiaomi_door_sensor.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5928739/log_xiaomi_door_sensor.txt)

SwoopX, 2021-02-04:
This should solve your challenge https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

Nazarbayevv, 2021-02-05:
@SwoopX It's a magic! You solved my issue. Thank you very much:)

closed 2021-02-05 by Nazarbayevv