dresden-elektronik/deconz-rest-plugin #4347: Blitzwolf BW-SS7 (tuya stuff)

handy80, 2021-02-05:
checked with Smanar in discord if this device was reported/requested already, come out with negative. so here's the details.

## Device

- Product name: Blitzwolf BW-SS7
- Manufacturer: _TYZB01_digziiav
- Model identifier: TS0003
- Device type : (2gang switching relay)
  - Switch

## Screenshots

[image]

[image]

maybe additional information to be found here: https://www.zigbee2mqtt.io/devices/BW-SS7.html

after I pressed read on Cluster Info, Infos changed slightly:
[image]

handy80, 2021-02-05:
and basic Cluster Info, selecting the second channel:
[image]

handy80, 2021-02-05:
groups:
[image]
scenes:
[image]
on/off:
[image]
OTAU:
[image]
Time:
[image]

Smanar, 2021-02-05:
Ha, my bad, lol was a plug in my mind ^^.
But this device realy seem nice, all cluster in good place, seem realy zigbee compliant.

Exccept your problem with missing endpoint during inclusion, I think using both button during inclusion can help for detection.

closed 2021-03-06 by github-actions[bot]