dresden-elektronik/deconz-rest-plugin #4348: TYZB01 light sensor

seafoodrice, 2021-02-05:
## Device

- Product name: Light sensor
- Manufacturer: _TYZB01_4mdqxxnn
- Model identifier: TS0222
- Device type : Please remove all unrelated device types.   
  - Sensor

## Screenshots
[image]

[image]
[image]
[image]

appleimperio, 2021-02-18:
Im waiting for this one too!

kiwikern, 2021-03-11:
Is any information needed to add this device? I also have one and am happy to help out.

seafoodrice, 2021-03-11:
This ticket is not resolved. Any Dev able to help?

Mimiix, 2021-03-11:
@Smanar  Do you have some time to implement this?

Smanar, 2021-03-11:
Can try, do you have a linux machine to test code modification ?

appleimperio, 2021-04-03:
Any dev available for help, if more information is needed just let me know. Thanks

Smanar, 2021-04-03:
Do you have a linux machine with classic OS to test code modifications ?

appleimperio, 2021-04-05:
Im not sure if this is what you want but a raspberry pi work, or i need another linux distribution?

Smanar, 2021-04-05:
Ok so first try

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch TYZB01 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry,it can freeze during compilation, my pi 3 do, so remove "-j2"

Then  try to include the device.

appleimperio, 2021-04-05:
Im having this error E: Unable to locate package deconz-dev

Thanks for your help

Smanar, 2021-04-05:
Have you installed deconz on the raspberry too ?
https://phoscon.de/en/conbee2/install#raspbian

This procedure will install too the "package stuff"

appleimperio, 2021-04-06:
Yes through a Docker and im doing this in the Docker console where the deconz is installed. I also try this steps https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2674#issuecomment-615889079 from another post of yours

Smanar, 2021-04-06:
Better to use a real OS, docker are virtual one, it s possible, but need lot of skill and I can't help you for that,

appleimperio, 2021-04-08:
Ok got the plugin compiled. Now when i add the device I get a Ready message but no new device is added. Thanks

Mimiix, 2021-04-08:
@appleimperio did you check the rest api?

appleimperio, 2021-04-08:
Yes its in there. I don't know where to see if the values are changing but since its a light sensor when i cover and uncover it the green light next to the device in the rest api start blinking. What else i can do to help make this work. Thanks!

Smanar, 2021-04-08:
>the green light next to the device in the rest api start blinking

You don't mean deconz instead ? The api don't have interface.
It's strange this kind of device is not visible in phoscon, but you haven't third app to make test ?
Else take a look direclty in the api `http://IP:PORT/api/KEY/sensors`

IP and PORT are the same than for deconz and KEY is a deconz API key.

appleimperio, 2021-04-08:
Ok this what im getting 
"ep":1,"etag":"d2da5f3a60f75eb24fb5c86af027670e","lastseen":"2021-04-08T14:51Z","manufacturername":"Tuya","modelid":"TS0222","name":"LightLevel 17","state":{"dark":false,"daylight":false,"lastupdated":"2021-04-08T14:51:26.110","lightlevel":14625,"lux":29},"type":"ZHALightLevel","uniqueid":"5c:02:72:ff:fe:c7:2c:13-01-0400"},"3":{"config":{"alert":"none","battery":87,"delay":180,"duration":60,"group":"1","on":true,"reachable":true}

Smanar, 2021-04-08:
Nice thx ^^, making the PR

appleimperio, 2021-04-09:
Thank you very much!

closed 2021-04-25 by manup