dresden-elektronik/deconz-rest-plugin #4349: Zemismart Tuya Zigbee Wall Switch 2 Gang (TS0042, _TZ3000_adkvzooy)

JasperMC, 2021-02-05:
## Device

- Product name: Zemismart Tuya Zigbee Wall Switch 2 Gang 
- Link to product: https://nl.aliexpress.com/item/4001227549409.html?spm=a2g0s.9042311.0.0.4d764c4dORIR1I
- Manufacturer: _TZ3000_adkvzooy
- Model identifier: TS0042
- Device type : Please remove all unrelated device types. 
  - Switch

Current behavior: On an older version the two switches were identified as lights in instead of switches, but pressing the buttons didn't work. In the current version of deconz (Version 2.09.00, ConBee II on firmware 26680700) the device does not register in the GUI but is visible in the VNC application. There are also variants with one or three switches. From what I can see, those use the same PCB design and chips.

Pictures of the device:

Assembled:
[image]

Disassembled:
[image]

## Screenshots

Clusters:
[image]

Node Info:
[image]

Basic cluster:
[image]

Power Configuration:
[image]

OTAU Cluster:
[image]

Smanar, 2021-02-05:
Added here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4157 
Do you have a linux machine to test the modification ?

JasperMC, 2021-02-05:
> Added here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4157 
> 
> Do you have a linux machine to test the modification ?

Thank you sir. I'm running on Docker currently but I can spin up a VM to try. Could you share the instructions for getting the right branch/version to test?

Smanar, 2021-02-05:
Full explanation here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

```
sudo apt install deconz-dev
git clone --branch tuya_switch https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

JasperMC, 2021-02-05:
Thanks! I'm having immense trouble installing deconz-dev. I tried the apt install and the suggested method in #2811, but I keep getting "Unexpected (" syntax error on this line:

```
sudo sh -c "echo 'deb http://phoscon.de/apt/deconz (cat /etc/os-release | grep -oP '(?<=VERSION_CODENAME\=).*')-beta main' > /etc/apt/sources.list.d/deconz.list"

JasperMC, 2021-02-05:
Hi Smanar,

I've managed to compile the plugin.

I can now successfully pair the switch in the Phoscon Web App. After doing so, it appears as a Switch under switches but still creates two lights (On/Off 9 and On/Off 10 in my case) that don't really do anything. I don't see any data coming through either when I listen for deconz_event in Home Assistant.

Don't know if that is expected behavior, just sharing my findings so far :)

Smanar, 2021-02-06:
Yes and no ^^.
It s notmal you have it, but weneed to prevent it.

ATM deconz have some problem to reconize tuya stuff, they are a little "exotic".

For this kind of devices you will have 2 (or more ) useless entries in "lights" sections.
You can delete them using the API, it s possible too with some third app, else using command line

`curl -X DELETE http://IP:PORT/api/KEY/lights/ID`

JasperMC, 2021-02-06:
Thanks! I'll give the deleting a try. As for now, pressing the buttons is not coming through yet.

If there's anything I can do to help let me know. Screenshots or doing a screenshare 😊

Smanar, 2021-02-06:
>As for now, pressing the buttons is not coming through yet.

Ha, this is not normal, on the "sensor" entry, you need to have a field "buttonevent" and this field containt the button value.

JasperMC, 2021-02-10:
Where would I find the sensor entry? In a GUI or the API?

Smanar, 2021-02-10:
Ha yes , my bad.
If you have just compiled the code, you have the new api code, but still the old json, the file is called  button_maps.json.
On raspberry you can find it in /usr/share/deCONZ/devices

The modification is here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4157/commits/39a33e9a14cfb1017ed753161ea0d49120c14b90#diff-a3adce467fe93b9c612c198c187dad34f5284d6c9adf66aa28127f0f619979b5

JasperMC, 2021-02-11:
Thanks! I updated the mapping and the switch works perfectly now :) 

I see you comitted the changed to master. Are they in the Roanoke 2.9.3? Where possible, I'd like to stay on the official releases vs the compiled plugin for testing :)

Smanar, 2021-02-11:
Easy to check, this issue will be closed when the PR will be validated. So not yet.

JasperMC, 2021-02-11:
> Easy to check, this issue will be closed when the PR will be validated. So not yet.

Clear, thanks! I tested the button extensively and all features are working from what I can see. Short press, double press and long press work fine on all buttons.

closed 2021-02-15 by manup