dresden-elektronik/deconz-rest-plugin #4353: deCONZ Service doesn´t start - (code=exited, status=217/USER)

AlexIOIIO, 2021-02-05:
## Describe the bug

deCONZ Service doesn´t start "systemctl status deconz"

## Steps to reproduce the behavior

Follow ConBee2 install steps for raspbian
Install wiringpi to fullfill Dependencies
enable deconz systemctl service
start deconz service (no error occurs)
Try to open http://x.x.x.x/pwa = Browser shows 'website not reachable'
systemctl status deconz:
● deconz.service - deCONZ: ZigBee gateway -- REST API
Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
Active: activating (auto-restart) (Result: exit-code) since Fri 2021-02-05 22:28:12 CET; 9s ago
Process: 19217 ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80 (code=exited, status=217/USER)
Main PID: 19217 (code=exited, status=217/USER)

Start deCONZ with "/usr/bin/deCONZ -platform minimal --http-port=80"
= Service is running and I´m now able to open the Admin Website http://x.x.x.x/pwa

## Expected behavior

deCONZ Service doesn´t start "systemctl status deconz"

## Screenshots

## Environment

Host system: Raspberry Pi
Running method: Raspbian
Firmware version: 26660700
deCONZ version: (2.07.01)
Device: ConBee II
Do you use an USB extension cable: no
Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

"pi" User was renamed

SwoopX, 2021-02-06:
You fiddled a bit more then just renaming the user, as the error tells us. It can't find uid 1000 any more.

AlexIOIIO, 2021-02-06:
Hey SwoopX,

thanks for your reply.
You´re right, "pi" was deleted, not renamed. Sorry for my mistake.
Good to know, uid 1000 ist neccessary to start deConz Service.

Regards,
Alex

SwoopX, 2021-02-07:
Well, not exactly necsessearry, but you should familiarize with service drop-in files to have everything configured to your needs. I could have sworn I created a wiki article on how to run deconz as another user years ago but I can't find it anymore 🤷

closed 2021-03-08 by github-actions[bot]