dresden-elektronik/deconz-rest-plugin #4355: EcoDim.07 LED Smart Dimmer

ottonix, 2021-02-06:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Light
  - Lock
  - Remote
  - Sensor
  - Siren
  - Switch
  - Thermostat
  - Other: 

## Screenshots

ottonix, 2021-02-06:
https://www.ecobright.nl/led-dimmer/smart-led-dimmer-zigbee-zwave-0-200w-voor-alle-afdekramen/

closed 2021-02-06 by ottonix

N0isMakeR, 2021-03-01:
Hi i wanted to ask if this means the device is now supported by deConz. Im having trouble finding the exact same device with phoscon app.

ottonix, 2021-03-06:
Hi, it is supported by deConz. I implemented in light as a Dimmer Switch. With homebidge i can use it also in Apple Home.