dresden-elektronik/deconz-rest-plugin #4357: Controlling QS-ZIGBEE-C01 with zigbee switch (eg. IKEA E1743) (retry)

Sjonnie2018, 2021-02-06:
## Describe the question or issue you are having
I am trying to figure out how to control the QS-ZIGBEE-C01 roller shutter module with an IKEA E1743 On/Off switch. The module is identified as a light switch but I can't control it with the On/Off switch. Probably because the module is in essence not a light. ;) I have been using the E1743 successfully to control regular lights but I can't get this combination to work.
Environment

## Screenshots
Not applicable

## Environment
DeCONZ Commercial Version 2.07.01

    Host system: Raspberry Pi 3B Rev 1.2
    Running method: Raspbian
    Firmware version: 26660700
    deCONZ version: 2.07.01
    Device: ConBee II
    Do you use an USB extension cable: yes
    Other USB devices connected to the host system: 2 x USB SSD

## deCONZ Logs
Not applicable

## Additional context
If this is not possible in DeCONZ I can fall back on Domoticz. But I would appreciate a pointer to a solution there as weel (I tried... didn't work).

SwoopX, 2021-02-06:
You probably can't do that, since both devices need to have a window covering cluster (cover a blue server cluster, the switch a grey client cluster)

Sjonnie2018, 2021-02-06:
That's too bad! It would be nice to have the module controlled on a Zigbee level as you can with "regular" lights. Maybe @Smanar can think of something either at the Zigbee level or in Domoticz. He is very knowledgeable of both platforms :)

SwoopX, 2021-02-07:
Don't get me wrong, but....

a) please don't tag other devs out of the blue
b) another dev with propbably don't tell you any different that I did
c) I already tried with more sophisticated devices
d) yes, of course it can be archieced what you seek with the help of whatever middleware...

closed 2021-02-10 by Sjonnie2018