dresden-elektronik/deconz-rest-plugin #4359: BTICINO F20T60A

igiulisol, 2021-02-07:
## Cannot connect device to Conbee II

I can't see de device 
- Reset the device
- Open network

But the device is not associated.

## Screenshots

[DEConzShot.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5938716/DEConzShot.pdf)

## Environment

My environment :
 - Host system: PC Centos8
 - Running method: Marthoc Docker container 
 - Firmware version: 264a0700
 - deCONZ version: 2.09.00
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? Ino

- REST-API Plugin installed
## deCONZ Logs

See attach
[Log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5938704/Log.txt)

## Additional context

igiulisol, 2021-02-07:
I forgot the REST-API screen shot.
[DEConzShot2.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5938767/DEConzShot2.pdf)

Smanar, 2021-02-07:
Ha ?
You are the second one recently with the same problem https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2456#issuecomment-774497636

You have tried the same pairing procedure ?

closed 2021-03-08 by github-actions[bot]