dresden-elektronik/deconz-rest-plugin #4360: Sonoff SNZB-03 MotionSensor - random detections

chrisma4600, 2021-02-07:
## Describe the bug

I have 2 Sonoff SNZB-03 Motion Sensors installed in my enviroment. Both of them recognize motion even though there is no motion.
I can not see any regularity.  The motion ist detected between 1 and 2 hours also at daytime as in night.   I can rule out, that theire ist any motion in deed the sensor triggers motion
Both Sensors have same behavior

## Steps to reproduce the behavior

use a SNZB-03   you will see

## Expected behavior

to trigger motion only when motion is 
## Screenshots

[image]
[image]
[image]
[image]
[image]

## Environ

 - Host system: Raspberry Pi 
 - Running method: Raspbian
 - Firmware version: 26690700
 - deCONZ version:  2.09.01
 - Device: RaspBee II
 - Do you use an USB extension cable: 

## deCONZ Logs

## Additional context

LindsayReid, 2021-02-08:
I too have the same random motion detects. I thought it might be a battery problem but tried a new one and still get spurious alerts.

Mimiix, 2021-02-08:
Someone reported this earlier (there's a few issues) somewhere, i picked up this url: https://tech.scargill.net/another-itead-sonoff-week/#comment-69873

It is the device that seems to be unreliable here and there's nothing we can do about that 😢.

chpego, 2021-03-02:
Hi there !
2 of my 4 sensors were not working properly, I decided to open the box of all the sensors and lift up the switch with the battery, then close the box again, and strangely one of the two that had the weird behavior (every 90 minutes) disappeared.
I'm not sure if it's just a hardware problem either.

joshhodgson, 2021-03-05:
I'm getting this too, though I doubt it's a deconz issue!

Mimiix, 2021-03-05:
As this seems to be a hardware issue, i suggest closing this issue for now as we can't do anything to fix it.

closed 2021-03-05 by Mimiix