dresden-elektronik/deconz-rest-plugin #4362: Can't find Hue Lights from Raspbee II

ikon106, 2021-02-07:
## Describe the question or issue you are having

## Screenshots

## Environment

## deCONZ Logs

## Additional context

manup, 2021-02-08:
Usually the easiest approach to move lights from the Hue bridge to deCONZ network is:

1. Start light search in the Phoscon App (might be helpful to raise search time in the Phoscon App Gateway Settings);
2. In the Hue App delete the light while it is powered and paired;
3. The Hue bridge now sends a reset command to the light and it will automatically join the deCONZ network.

Since the lights didn't join to the deCONZ network yet I guess they aren't in factory new state, the logs also show no indication of a Hue light joining. You might need to pair them to the Hue bridge again and try above steps.

ikon106, 2021-02-09:
I re-paired them to the bridge and started a 10-min search in phoscon. Towards the beginning of the search, I deleted the lights from the Hue app. Phoscon was still not able to find them.
Here is the debug from right before the beginning of the search to right after I deleted the lights from the Hue app: https://pastebin.com/nqMyYm3p
Here is the debug from the full search, incuding the above paste: https://pastebin.com/H19tQkex

ikon106, 2021-02-09:
I forgot to mention that the Hue Bridge is v1 (the round one). I don't know if that's relevant.

ikon106, 2021-03-03:
I still have not received a solution to my problem.

Mimiix, 2021-03-03:
What happens when you remove the fan from the RPI? Also: What case are you using?

ikon106, 2021-03-11:
Thanks for answering, unfortunately, unplugging the fan from the RPI did nothing.
I am using the following [case.](https://www.amazon.se/gp/product/B07V2ZLB9K/ref=ppx_yo_dt_b_asin_title_o01_s02?ie=UTF8&psc=1)

closed 2021-04-10 by github-actions[bot]

wvrneer, 2021-05-25:
I have a similar problem. No way to connect a Osram Lightify Flex strip on a new Raspbee II.
Before, the strip was successfully connected on a Raspbee.
For testing I resetted the strip serveral times and tried to connect it alternately on both devices.
Every search was successful with Raspbee, but not a single one with Raspbee II.

Is there a problem with Raspbee II ?

Adam-Langley, 2021-07-12:
@wvrneer I have had trouble connecting to all my devices using a brand new Raspbee II. Nothing worked to the point I was wondering if the Raspbee II was DOA.
After moving the Raspbee II away from other sources of possible radio interference (my Ubiquiti switch), and moving closer to the device I was attempting to pair with, all the problems went away. It appears once paired, the connection is much more tolerant to a noisy radio environment - possible the pairing process is just more dependent on a strong connection (i.e, timing related?)