dresden-elektronik/deconz-rest-plugin #4365: Two instances of deCONZ with the same bridge id?

sannc, 2021-02-07:
## Describe the question or issue you are having
I am failing at adding two instances of deCONZ on two separate Raspberry Pis with two separate Conbee II sticks to Homebridge. It seems that HB uses the same user name for both instances in the JSON config file which does not work because the information there is stored as an array and this would lead to a duplicate key in the array.

After a lot of investigation I came to the conclusion that the cause for this problem is, that both instances use the same bridge id (as returned by https://phoscon.de/discover - 00212EFFFF060E89). This is also the user name, HB wants to use.

I set up both instances manually - there was no backup/restore involved.

Question: How can it be that two completely separate instances on two different computers use the same id? How can I change the id? Will this actually solve my HB problem?

Thanks in advance!

## Screenshots

## Environment
Both instances are running identical configurations

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26660700
 - deCONZ version: 2.07.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? USB memory stick

## deCONZ Logs

## Additional context

Smanar, 2021-02-07:
It  s not the gateway MAC adress ?
You can chnage it in deconz config, but you will probably need to re - include all devices.

SwoopX, 2021-02-07:
When you query the `/config` resource of the REST API, it should give you the current bridgeID of each installation. Are those identical as well (that would be odd). Iirc, you cannot change that ID, as it gets set automatically, based on the Conbee you're using.

There's only one condition I can think of where both Conbees would assume the exact same mac address, which is when you biuld up your network with stick A on a system and then later on swap Sticks to stick B to continue operation on the very same installation...

sannc, 2021-02-07:
> When you query the `/config` resource of the REST API, it should give you the current bridgeID of each installation. Are those identical as well (that would be odd). Iirc, you cannot change that ID, as it gets set automatically, based on the Conbee you're using.
Yes. Both requests yield the same bridge id

> There's only one condition I can think of where both Conbees would assume the exact same mac address, which is when you biuld up your network with stick A on a system and then later on swap Sticks to stick B to continue operation on the very same installation...
I don't think I swapped the Conbees, but I am not 100 % sure … so, how can I fix it?

sannc, 2021-02-07:
> It s not the gateway MAC adress ?
> You can change it in deconz config, but you will probably need to re - include all devices.

Re-including would be no problem. How do I change it, given that I am running a headless system without X installed?

SwoopX, 2021-02-07:
> I don't think I swapped the Conbees, but I am not 100 % sure

Maybe you imported your configuration to the other system then?

Anyway, the only way I can currently think of to change it is to edit the database directly. Not sure though if this will be presistent due to the aforementioned mechanism. When you plan to edit the DB, be careful and make sure to have a backup before starting.

sannc, 2021-02-07:
> Maybe you imported your configuration to the other system then?
Definitely not.
 
> Anyway, the only way I can currently think of to change it is to edit the database directly. Not sure though if this will be presistent due to the aforementioned mechanism. When you plan to edit the DB, be careful and make sure to have a backup before starting.
A backup goes without saying ;-) Where is the DB located? Is it a file? Or SQLite?

SwoopX, 2021-02-07:
It's a sqlite DB, `zll.db`. Usually in the home directory of the user running the deconz service.

sannc, 2021-02-08:
Hi, tried to change the id in the DB. The api/config shows the changed id, but for some friggin reason, it switches back to the duplicate id after I log into the web interface. 

I also tried to reset and even uninstall/install deCONZ with both Conbee II on my secondary raspi, but every time I end up with the same id 00212EFFFF060E89.

Is there a way to check the mac addresses of the Conbee II directly?

SwoopX, 2021-02-08:
The MAC is also used in various places in the network settings in deconz, but I dunno what possibly might happen if you fiddle with them.

@manup you probably know this best ;)

sannc, 2021-02-08:
I am quite desperate … next thing I will try is buy another Conbee II and set up a clean instance of deCONZ on my secondary raspi …

ebaauw, 2021-02-08:
You can change the MAC address of a RaspBee or ConBee, but this is **not** exposed over the REST API.  This is used when replacing a faulty device, and should probably not be used beyond that.  I think the REST API plugin does this when restoring a backup.  You can also do this manually from the GUI (under the _Network Settings_, after selecting _Custom MAC Address_).  Before attempting that, I would double check if indeed both devices have the same MAC address, in the _Node Info_ panel in the GUI.

If you don't run the GUI, you might try and check the `uniqueid` of the `/lights` resource created for the coordinator through the API.

sannc, 2021-02-08:
Thank you. I have no lights in my secondary installation, but _/sensors_ returns this:

{
  "1": {
    "config": {
      "configured": true,
      "on": true,
      "sunriseoffset": 30,
      "sunsetoffset": -30
    },
    "etag": "79d9a68bb43c94fc77041357718eb6a3",
    "manufacturername": "Philips",
    "modelid": "PHDL00",
    "name": "Daylight",
    "state": {
      "dark": false,
      "daylight": true,
      "lastupdated": "2021-02-08T11:17:43.302",
      "status": 160,
      "sunrise": "2021-02-08T06:47:28",
      "sunset": "2021-02-08T16:30:25"
    },
    "swversion": "1.0",
    "type": "Daylight",
    **"uniqueid": "00:21:2e:ff:ff:06:0e:89-01"**
  }
}

On the primary system _/lights_ returns this

{
  "1": {
    "etag": "b33169eb457e5dc6bf391b59e6504fea",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-02-08T11:15Z",
    "manufacturername": "dresden elektronik",
    "modelid": "ConBee II",
    "name": "Configuration tool 1",
    "state": {
      "reachable": true
    },
    "swversion": "0x26660700",
    "type": "Configuration tool",
    **"uniqueid": "00:21:2e:ff:ff:06:0e:89-01"**
  },

Looks as if both uniqueids are identical …

Mimiix, 2021-02-08:
Which is totally fine, as the daylight sensor is a virtual device.

ebaauw, 2021-02-08:
So it looks like the coordinator devices in both installations have the same MAC address.  Are they both ConBee II devices?  Do you still have the packaging with the factory-configured MAC address?  I don't know if there's an option to revert to that (to un-customise the MAC address so to say), you might just need to change it back manually.  I'm afraid you need to use the GUI for that, or write a programme using the serial protocol.

sannc, 2021-02-08:
I have the packagings not at hand but will check as soon as I can. Both are Conbee II dongles.

I definitely will not install X to reset the MAC address … and I am still puzzled why something like this could happen. 

The easiest thing for me to do is to order another Conbee II, swap them and sell the extra dongle on eBay. Still strange, though …

sannc, 2021-02-09:
So, the new Conbee II arrived today and I was able to set eveything up. Worked without any problems.

However, it stays a mystery how the two dogles ever came to have the same id …

Thanks everyone for helping.

closed 2021-02-09 by sannc