dresden-elektronik/deconz-rest-plugin #4367: Lidl wireless doorbell and phoscon

dodo124, 2021-02-07:
## Describe the question or issue you are having

## Screenshots

## Environment

  - Host system: (Raspberry Pi /  Conbee2)
 - Running method: (Raspbian)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.09.91)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes) 
 - Is there any other USB or serial devices connected to the host system? Yes: Touchdisplay of the RPi.

## deCONZ Logs

## Additional context

T0mWz, 2021-02-07:
Pllease, see; https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4260

SwoopX, 2021-02-07:
The device is not yet supported, hence it is not usable via REST API. Even it it would, that doesn't automatically mean it is visible in Phoscon, as it is a different software. You might want to consider raising a support request at the Phoscon beta repository upfront and reference the corresponding issue here.

Also, please use the search for issues next time to avoid any duplicates. The issue has been shared above.

closed 2021-02-07 by SwoopX

kejXero, 2021-07-10:
A litle bit off topic, but could you tell how you added the doorbell to Deconz? I have tried adding it a switch and a sensor without any luck.