dresden-elektronik/deconz-rest-plugin #4370: Update a Docker Deconz

marithpl, 2021-02-07:
## Describe the question or issue you are having

When docker version will be updated?
My Docker Deconz firmware is: 2.09.01 / 23.12.2020
On GitHub page there is Current Stable: v2.7.1

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

SwoopX, 2021-02-07:
Why do you want to downgrade? 2.09.01 > 2.07.01

closed 2021-02-08 by Mimiix

marithpl, 2021-02-08:
> Why do you want to downgrade? 2.09.01 > 2.07.01

2.**09**.01 is higher than 2.**7**.1 ?

Mimiix, 2021-02-08:
This has been discussed before :) The Phoscon app provides double digit. I'll put it on the internal to-do.

marithpl, 2021-02-08:
Sorry, I didn't know that. But honestly it's so confusing.