dresden-elektronik/deconz-rest-plugin #4374: Tracking down crashes -  what does this error mean? dev can't create file: /dev/ttyACM0 via mknod

kses, 2021-02-08:
## Describe the question or issue you are having
I am trying to track down deconz crashes I have been seeing and wifi crashes. A lot of strange behaivior. I found some references to ttyACM0 in the log, which seems strange since conbee use ttyUSBX. Does anyone know the reason for the error found in the log below? Could this caused deconz to crash?

## Screenshots
no

## Environment

 - Host system: NUC
 - Running method: Marthoc Docker container
 - Firmware version:  26390500
 - deCONZ version: 2.09.01 / 2020-12-23
 - Device: ConBee I
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

11:08:36:613 dev can't create file: /dev/ttyACM0 via mknod (missing permissions),
11:08:31:569 wait reconnect 6 seconds,
11:08:32:569 wait reconnect 5 seconds,
11:08:33:568 wait reconnect 4 seconds,
11:08:34:569 wait reconnect 3 seconds,
11:08:35:569 wait reconnect 2 seconds,
11:08:36:568 wait reconnect 1 seconds,
11:08:36:617 device disconnected reason: 4, index: 0,
11:08:37:569 wait reconnect 15 seconds,
11:08:37:614 dev can't create file: /dev/ttyACM0 via mknod (missing permissions),
11:08:38:569 wait reconnect 14 seconds,
11:08:40:569 wait reconnect 12 seconds,
11:08:41:569 wait reconnect 11 seconds,
11:08:43:569 wait reconnect 9 seconds,
11:08:44:569 wait reconnect 8 seconds,
11:08:45:569 wait reconnect 7 seconds,
11:08:42:568 wait reconnect 10 seconds,
11:08:39:569 wait reconnect 13 seconds

## Additional context
Mixed of mainly Trådfri and Xiaomi devices

Mimiix, 2021-02-08:
`I am trying to track down deconz crashes I have been seeing and wifi crashes.`
 
 So you've changed something, and now it started doing this?
 
 From what i can see, it needs permissions.  No sudo? Would help if you shared some OS details and other things you tried :)

Mimiix, 2021-02-08:
A quick google found me this: https://github.com/marthoc/docker-deconz/issues/327

kses, 2021-02-08:
It is running in a docker container... (The host is Ubuntu Server 20.04 LTS.)
The container has the Conbee I forwarded to /dev/ttyUSB0. Deconz can and does talk to the Conbee I.
There is nothing, on ttyACM0 that deconz should talk...

Docker compose file:
```
deconz:
    image: marthoc/deconz:latest
    container_name: deconz
    network_mode: host
    restart: always
    volumes:
      - /opt/deconz:/root/.local/share/dresden-elektronik/deCONZ
    devices:
      - /dev/serial/by-id/usb-FTDI_FT230X_Basic_UART_DO00K6MH-if00-port0:/dev/ttyUSB0
    environment:
      - DECONZ_WEB_PORT=10080
      - DECONZ_WS_PORT=10443
      - DEBUG_INFO=1
      - DEBUG_APS=0
      - DEBUG_ZCL=0
      - DEBUG_ZDP=0
      - DEBUG_OTAU=0
      - DECONZ_VNC_MODE=1
      - DECONZ_VNC_PORT=5900

```

Mimiix, 2021-02-08:
I think you should ask this in the Docker repository.

kses, 2021-02-16:
I will, are there any confirmed reports of hardware failure? How do they behave? I run into all kinds of strange errors. Not the normal kind related to EMC issues. The strange thing is,  the issue daily takes down the wifi and zigbee network, and require reboot to fix either.

closed 2021-03-17 by github-actions[bot]

siaox, 2021-11-04:
> It is running in a docker container... (The host is Ubuntu Server 20.04 LTS.) The container has the Conbee I forwarded to /dev/ttyUSB0. Deconz can and does talk to the Conbee I. There is nothing, on [...]

hey man did you manage to figure this out?

0xSh1v4, 2021-12-01:
> hey man did you manage to figure this out?

Did you find any solution for this. I am struck with the same thing

LyndseyPaxton, 2021-12-15:
Same issue here too!

Vligsu, 2022-01-07:
Same issue as well!

adocampo, 2022-01-20:
I'm on the same boat with the ConBee II running on docker with an arch host.
If I pass the `--device=/dev/ttyACM0:/deb/ttyUSB0` then it complains 
```
dev can't create file: /dev/ttyACM0 via mknod (missing permissions),
```
If I pass the `--device=/dev/ttyACM0` then it complains 
```
dev can't create file: /dev/ttyUSB0 via mknod (missing permissions),
```
I'm running docker as root, but anyway id added root to `uucp` as there are no `dialout` group on arch

Any clue?

gonzo09, 2022-03-24:
> It is running in a docker container... (The host is Ubuntu Server 20.04 LTS.) The container has the Conbee I forwarded to /dev/ttyUSB0. Deconz can and does talk to the Conbee I. There is nothing, on [...]

I think deconz search for a Conbee II.
See device option on https://github.com/deconz-community/deconz-docker#command-line-options

malekcellier, 2022-04-28:
I fixed this using elevated persmissions to run the container..

Sacerdoss, 2022-05-01:
Fixed it by using the correct device file ("/dev/ttyACM0" instead of /dev/serial/by-id) when passing it to the docker container..

> Readme.md:
> by default ConBee = /dev/ttyUSB0, Conbee II = /dev/ttyACM0, RaspBee = /dev/ttyAMA0 or /dev/ttyS0