dresden-elektronik/deconz-rest-plugin #4376: Some Devices are not using routing devices to connect. Why not?

easybeat, 2021-02-08:
## Describe the question or issue you are having
I have several devices like Bitron 902010/32 Thermostat or TRV TS0601 that do not connect to router devices like lights. They only connect sometimes to an Ikea range extender but then lose the connection after a while.

As I understand the zigbee protocoll defines that powered devices can act as router devices to extend the network range.

Do these devices not support the zigbee standard properly? And is there a way to find out which devices do support the standard?

## Screenshots
[image]

## Environment

 - Host system: NUC with Proxmox with Ubuntu OS
 - Running method: Proxmox VM Ubuntu
 - Firmware version: 26680700
 - deCONZ version: 2.08.00
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Aeotec Z-Wave Stick

Mimiix, 2021-02-09:
I've forwarded this to @manup  to get you a proper answer 😅.

manup, 2021-02-10:
The lines are only a visual representation of what deCONZ knows about nodes and their neighbor / child tables. This data can be incomplete as all routers are queried in a lazy manner. So even if no line is shown the underlying communication usually works fine as long as data is received from the device, in which case it blinks blue.

End devices might select new parent nodes, for example when the parent is powered off, or communication errors are detected.

easybeat, 2021-02-10:
> The lines are only a visual representation of what deCONZ knows about nodes and their neighbor / child tables. This data can be incomplete as all routers are queried in a lazy manner. So even if no [...]

Thank you Manuel for the explanation.

But what can I do if I have a bunch of devices which losing connection to the system even with an Ikea range extender within 1m distance but away from deconz hub? 

Also to say that there are at least 10 other devices like lights around which work without a problem since months/years...

Thank you for your help.

easybeat, 2021-02-16:
It looks like somehow there is some improvement since I have updated to 2.9.3. Almost every device now showing a connection to a router and not only to Ikea range extender's also to lights or other devices. And this is working for days now without an outage.

Could this be related to a change in 2.9.3?

Smanar, 2021-02-16:
No, not from my memory, perhaps the network need some time to be optimised ..

Mimiix, 2021-02-17:
As smanar mentioned , it can take a while. As long as it all works, it should be fine.

Case closed?

closed 2021-03-18 by github-actions[bot]

easybeat, 2021-03-18:
I still have a lot of devices which don't pick up a router and there are plenty routers around like Ikea range extender, HUE GU10 lamps etc.

[image]

Is there nothing we can do about this?

Smanar, 2021-03-18:
Not realy but thoses one are not xiaomi.
What is the problem exactly, compatibilty between brand ?

They always choose direct connexion ? Can be normal, but less if they loose the connexion just after.

easybeat, 2021-03-18:
The problem is that at the moment every day 1 device losing connection and I have to re-join it. It happens completely random and also with devices which are only 4 meters from ConbeeII away!

And this is more than annoying!

easybeat, 2021-03-18:
And these are also different devices. 1 day an Aqara Opple Switch, the other day a Dresden RGBW-Zigbee Controller, then a Zemismart Actuator...just random!

Smanar, 2021-03-18:
And they use all direct connexion ? no router ?
If they use direct connexion, take a look on logs if you don't see connexion issue, are you using the USB extension ?

easybeat, 2021-03-19:
Hi
yes I'm using a USB extension. It is very hard to find them connection problems and what should I do if there are some?

I started yesterday using also zigbee2mqtt. And I have now joned on there a TRV and 2 Opple switches and placed them at the same positions where I'm losing connection all the time with deconz. It will be interesting to see if there is a difference.

Interesting to see that also here the TRV is only using direct connection and not the IKEA range extender
[image]

But so far the connection to the TRV is stable but I give it another 72 hours to see.

What I realised yesterday when I started with the zigbee2mqtt that there could be a USB power problem on my NUC. I struggled first to join any new devices and only after I took the Aeotec Z-Wave Stick out it started working. I have now connected the Z-Wave stick in a powered USB-Hub and the conbee II stick direct in the box. Is there a way I can check if the conbee II get's "enough" power or is sending at full power?

Smanar, 2021-03-19:
I know on raspberry power undervoltage produce error message, you can too see connexion error on deconz logs.

The ikea range extender is router
The TRV is end device.